espressif/esp-idf #8401: Incorrect handling of key RSC in wpa_supplicant allows replay attack

ivq, 2022-02-13:
wpa_sm_set_seq() in components/wpa_supplicant/src/rsn_supp/wpa.c is handling key RSC incorrectly.
Whether to use Null RSC should be based on protocol and whether GTK or PTK is to be installed.
Current implementation ignores key RSC provided in EAPOL key for RSN GTK, which allows replaying previous group addressed packets.

AxelLin, 2022-05-11:
Just remind stable branches also need this fix.

ivq, 2022-05-28:
This vulnerability was assigned CVE ID CVE-2021-43134.

AxelLin, 2022-05-28:
> This vulnerability was assigned CVE ID CVE-2021-43134.

@Alvin1Zhang 
This is not a complex fix, no idea why stable branches do not include this fix.
Should backport the fix ASAP.

jack0c, 2022-05-30:
@ivq @AxelLin Thanks for the reminder, will do it soon.

AxelLin, 2022-06-18:
> @ivq @AxelLin Thanks for the reminder, will do it soon.

Unfortunately, v4.3.3 does not include this fix.

AxelLin, 2022-07-12:
@jack0c 
v4.3 branch still does not include the fix, it's quite late for a CVS fix.

AxelLin, 2022-12-18:
@jack0c If the fix is available in all impacted branches, you can close it.

BTW, this page https://github.com/espressif/esp-idf/security/advisories/GHSA-7f7f-jj2q-28wm  needs update since new releases are available. @mahavirj

jack0c, 2023-04-24:
master/v5.0/v5.1: 079b5b1857242d6288478c62dc36a843718882e9
v4.4: e41a7d64b566eee2ec5e1c2d4d5af02fbf19cf21
v4.3:1970d887a39f4150f3621022418b5e4cd7a73e3b
v4.2:25253b22ccc70a42061d87f1b74af126905935cb
v4.1:3d88183b23864d7a9374ca5dfaf00acba7871a5a