espressif/esp-idf #8404: Inconsistent naming for compiler optimisation

mbrunnen, 2022-02-14:
The naming scheme for the debug compiler optimisation is inconsistent:
- [`COMPILER_OPTIMIZATION_DEFAULT`](https://github.com/espressif/esp-idf/blob/20847eeb96c486629a678d6ffbe73670013fb771/Kconfig#L229)
- [`BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG`](https://github.com/espressif/esp-idf/blob/20847eeb96c486629a678d6ffbe73670013fb771/components/bootloader/Kconfig.projbuild#L28)

The one ends with `_DEFAULT` and the other one ends with `_DEBUG`. To complete the confusion there is even an alias with the name [`COMPILER_OPTIMIZATION_LEVEL_DEBUG`](https://github.com/espressif/esp-idf/blob/20847eeb96c486629a678d6ffbe73670013fb771/components/mdns/test_afl_fuzz_host/sdkconfig.h#L347).

0xjakob, 2022-02-16:
@mbrunnen Thanks for this hint. We'll take a look. But from the first sight I'm not sure if this can be fixed easily in older versions due to compatibility issues. Someone could have created an `sdkconfig.defaults` with these values which would quite silently fail if the their names get changed.

mbrunnen, 2022-02-16:
Changing it in maintenance releases for older versions would not be a great idea, of course. But maybe in a coming release? BTW, it could be done backwards compatible by switching `CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG` and `CONFIG_COMPILER_OPTIMIZATION_DEFAULT`

mbrunnen, 2022-02-18:
> We could probably use [`sdkconfig.rename` mechanism](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#backward-compatibility-of-kconfig-options) to provide compatibility with the old name.

Interesting, there is already a line, `CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG      CONFIG_COMPILER_OPTIMIZATION_DEFAULT` in `sdkconfig.rename`. But in my understanding it does the exact opposite.

igrr, 2022-02-20:
Indeed, CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG was renamed to CONFIG_COMPILER_OPTIMIZATION_DEFAULT in https://github.com/espressif/esp-idf/commit/4fdaeb6b6e22daea72dea47fd7e271120af0935a.

AxelLin, 2022-09-25:
> Indeed, CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG was renamed to CONFIG_COMPILER_OPTIMIZATION_DEFAULT in [4fdaeb6](https://github.com/espressif/esp-idf/commit/4fdaeb6b6e22daea72dea47fd7e271120af0935a).

This issue is stiill in OPEN status, so I'm wondering if your conclusion is to keep it as is (CONFIG_COMPILER_OPTIMIZATION_DEFAULT) or change back to CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG ?

ESP-Marius, 2023-06-06:
We agree that `DEBUG` was probably a more descriptive name as well as being consistent with the bootloader config. It has been renamed back now and the issue will be updated when the commit is synced to github.

closed 2023-06-08 by espressif-bot