espressif/esp-idf #8406: Rate is ignored for esp_wifi_80211_tx

Jeija, 2022-02-14:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version: v5.0-dev-957-g6cb6087dd0
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I want to transmit a custom data frame using `esp_wifi_80211_tx` with OFDM modulation.
I am able to transmit arbitrary data with 1 Mbps DSSS modulation, but calling `esp_wifi_internal_set_fix_rate` with e.g. rate `WIFI_PHY_RATE_MCS7_LGI` does not change this behaviour.

### Expected Behavior
The modulation of the frame transmitted using `esp_wifi_80211_tx` changes depending on the parameter that was previously provided to `esp_wifi_internal_set_fix_rate`.

### Actual Behavior
The ESP32 always transmits a 1Mbps DSSS-modulated frame.
The call to `esp_wifi_internal_set_fix_rate` returns `ESP_OK`, but is otherwise completely ignored as far as I can tell.

### Steps to reproduce

1. Flash code (provided below) to ESP32
2. Monitor WiFi channel 13 with e.g. a WiFi USB dongle in promiscuous mode or an SDR and use a suitable tool to display modulation types

### Code to reproduce this issue

```cpp
#include "freertos/FreeRTOS.h"

#include "esp_system.h"
#include "esp_event.h"
#include "esp_event.h"
#include "esp_wifi.h"
#include "esp_private/wifi.h"

#include "nvs_flash.h"
#include "string.h"

uint8_t data_raw[] = {
	0x08, 0x00,							// 0-1: Frame Control
	0x00, 0x00,							// 2-3: Duration
	0x20, 0x39, 0x56, 0x2a, 0xcb, 0xaf,				// 4-9: Destination address (broadcast)
	0xf4, 0x92, 0xbf, 0x14, 0xdb, 0xa9,				// 10-15: Source address
	0xba, 0xde, 0xaf, 0xfe, 0x00, 0x06,				// 16-21: BSSID
	0x00, 0x00,							// 22-23: Sequence / fragment number	
};

void tx_task(void *pvParameter) {
	for (;;) {
		vTaskDelay(100 / portTICK_PERIOD_MS);
		esp_wifi_80211_tx(WIFI_IF_STA, data_raw, sizeof(data_raw), false);
	}
}

void app_main(void) {
	nvs_flash_init();
	esp_netif_init();

	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	cfg.ampdu_tx_enable = 0;

	ESP_ERROR_CHECK(esp_event_loop_create_default());
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));

	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
	wifi_config_t ap_config = { .sta = { .ssid = "dummy", .password = "dummy" }	};

	ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &ap_config));
	ESP_ERROR_CHECK(esp_wifi_start());
	ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));
	ESP_ERROR_CHECK(esp_wifi_set_bandwidth( ESP_IF_WIFI_STA, WIFI_BW_HT20));
	ESP_ERROR_CHECK(esp_wifi_set_protocol(ESP_IF_WIFI_STA, WIFI_PROTOCOL_11B|WIFI_PROTOCOL_11G|WIFI_PROTOCOL_11N));
	ESP_ERROR_CHECK(esp_wifi_set_channel(13, WIFI_SECOND_CHAN_BELOW));

	/* The following line is apparently completely ignored */
	ESP_ERROR_CHECK(esp_wifi_internal_set_fix_rate(ESP_IF_WIFI_STA, true, WIFI_PHY_RATE_MCS7_LGI));

	xTaskCreate(&tx_task, "tx_task", 2048, NULL, 5, NULL);
}
```

MaxwellAlan, 2022-02-15:
Hi @Jeija , thanks for reporting this issue, may be we need to update the docs.
Actually esp_wifi_internal_set_fix_rate api will affect all pkts rate, it's not a good way to modify the 80211_tx rate, so we add a new feature to config 80211_tx rate, which you can call this api: [esp_err_t esp_wifi_config_80211_tx_rate(wifi_interface_t ifx, wifi_phy_rate_t rate);](https://github.com/espressif/esp-idf/blob/1d77b5b6fddb87f064c6fa486714eaaad43cd915/components/esp_wifi/include/esp_wifi.h#L1296) to modify 80211_tx rate now.

Jeija, 2022-02-15:
Hello @MaxwellAlan, thanks for your swift and helpful response!
I can confirm that this solution works, all I had to to was change the line
```
ESP_ERROR_CHECK(esp_wifi_internal_set_fix_rate(ESP_IF_WIFI_STA, true, WIFI_PHY_RATE_MCS7_LGI));
```
to
```
ESP_ERROR_CHECK(esp_wifi_config_80211_tx_rate(ESP_IF_WIFI_STA, WIFI_PHY_RATE_MCS7_LGI));
```

closed 2022-02-15 by Jeija

jack0c, 2022-04-25:
The code have been merged to master, you can find the fix in 68d4c47b7ef509ce1ae66714d2fe0d741e49d188