espressif/esp-idf #8407: I2C-tools crashes on ESP32-S3

Derek-RPB, 2022-02-14:
## Environment

- Development Kit:      ESP32-S3-DevKitC-1
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version (run ``git describe --tags`` to find it): v4.4
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes, Espressif IDE
- Power Supply:         USB

## Problem Description

I have an ESP32s3 running the I2C-tools example from ESP-IDF I can run some of the commands such as "i2cconfig" and such, but when I run "i2cdetect" it crashes after printing the first line with a "0 panic'd (Interrupt wdt timeout on CPU0)"

I have tried the other I2C examples and some of them will sometimes product expected results, and sometimes not. The bus has a single I2C device on it that has 3.3V 10k pull up resistors on the SDA and SCK line. I have tried this with the ESP32 internal pull up resistors enabled and disabled, and it fails in both scenarios.

### Expected Behavior

Scan the I2C bus for devices, report device ids.

### Actual Behavior

Crashed

### Steps to reproduce

See above

### Code to reproduce this issue

I2C-tool example code 

## Debug Logs

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1160
load:0x403b6000,len:0xa30
load:0x403ba000,len:0x2d48
entry 0x403b61ec
I (91) cpu_start: Pro cpu up.
I (91) cpu_start: Starting app cpu, entry point is 0x40375350
I (0) cpu_start: App cpu up.
I (105) cpu_start: Pro cpu start user code
I (105) cpu_start: cpu freq: 160000000
I (105) cpu_start: Application information:
I (108) cpu_start: Project name:     i2c_tools
I (113) cpu_start: App version:      v4.4-dirty
I (118) cpu_start: Compile time:     Feb 14 2022 17:02:09
I (124) cpu_start: ELF file SHA256:  e1d6c35f44dfbb81...
I (130) cpu_start: ESP-IDF:          v4.4-dirty
I (135) heap_init: Initializing. RAM available for dynamic allocation:
I (143) heap_init: At 3FC97468 len 00048B98 (290 KiB): D/IRAM
I (149) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (156) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (162) heap_init: At 600FE028 len 00001FD8 (7 KiB): RTCRAM
I (169) spi_flash: detected chip: generic
I (173) spi_flash: flash io: dio
W (177) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (190) sleep: Configure to isolate all GPIO pins in sleep state
I (197) sleep: Enable automatic switching of GPIO sleep configuration
I (204) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

 ==============================================================
 |             Steps to Use i2c-tools                         |
 |                                                            |
 |  1. Try 'help', check all supported commands               |
 |  2. Try 'i2cconfig' to configure your I2C bus              |
 |  3. Try 'i2cdetect' to scan devices on the bus             |
 |  4. Try 'i2cget' to get the content of specific register   |
 |  5. Try 'i2cset' to set the value of specific register     |
 |  6. Try 'i2cdump' to dump all the register (Experiment)    |
 |                                                            |
 ==============================================================

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
i2c-tools> i2cconfig --port=1 --freq=400000 --sda=1 --scl=2
i2c-tools> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- UU Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x40001c3b  PS      : 0x00060034  A0      : 0x8037ccec  A1      : 0x3fc94cb0
A2      : 0x3fc96b7c  A3      : 0x3fc96b7c  A4      : 0x00060021  A5      : 0x3fce9210
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x8037f4a9  A9      : 0x400559cc
A10     : 0x00060023  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00060023
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff
Core  0 was running in ISR context:
EPC1    : 0x4202d96b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40001c3b

Backtrace:0x40001c38:0x3fc94cb00x4037cce9:0x3fc94cd0 0x403775cd:0x3fc94d00 0x40376a5e:0x3fc94d40 0x400559dd:0x3fce7a50  |<-CORRUPTED

Core  1 register dump:
PC      : 0x4202dd56  PS      : 0x00060f34  A0      : 0x8200297a  A1      : 0x3fcf65b0
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001
A6      : 0x3fc96b7c  A7      : 0x3fc96b7c  A8      : 0x82012d96  A9      : 0x3fcf6580
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00060620  A13     : 0x00060623
A14     : 0x00060023  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x4202dd53:0x3fcf65b00x42002977:0x3fcf65d0 0x4037d779:0x3fcf65f0 0x4037f1f1:0x3fcf6610

ELF file SHA256: e1d6c35f44dfbb81

Rebooting...

```

o-marshmallow, 2022-02-15:
Hello @Derek-RPB ,

I tried to reproduce this issue but was unable to. I am also using an ESP32-S3-DevKitC-1, on ESP-IDF v4.4 (commit `8153bfe4125e6a608abccf1561fd10285016c90a`), `i2c_tools` example and executing the commands:
```
i2cconfig --port=1 --freq=400000 --sda=1 --scl=2
i2cdetect
```
after flashing. All I get is:
```
i2c-tools> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
```
when I don't plug any device on the bus (internal pull-up enabled).
When I plug a device on the bus:
```
i2c-tools> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: 20 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
```
Here, I plugged an I/O expander MCP23017. I tried with internal pull-ups first and then with external pull-ups (20kOhms). Same result.

The first questions that come to my mind are:
- Can you check which C line corresponds the crash address `0x40001c3b`? `idf.py monitor` should decode all these addresses when a crash occurs.
- What happens if you totally unplug the device (no device on bus)? Do you still get this exception?
- Do you have a logic analyzer, or scope, to check what happens on the bus before crashing?
- Which device are you connecting to the bus? Is it possible that it does clock stretching?

Derek-RPB, 2022-02-15:
Apologies for the issues. I took a fresh look at it today and was able to solve my problems.

At 100khz:
With the device connected to nothing and the pullups enabled, it worked correctly. With the device connected to the target and either the target or the device having the pullups enabled, it worked correctly. The issues came from both devices having pullups enabled.

At 400khz:
This is really hit and miss. This is probably due to my signal lengths on my prototyping board right now. This is what I was testing at yesterday when I would constantly get issues.

So for now, I will stick with 100khz and pay attention to my pullups. Thank you!

closed 2022-02-15 by Derek-RPB

o-marshmallow, 2022-02-15:
@Derek-RPB Thanks for the follow-up!
Indeed, 10k for 400KHz is a bit too high for pull-up resistors (but ok for 100Khz)
4.7KOhm should be ok for 400KHz

Derek-RPB, 2022-02-15:
Understood, thank you!