espressif/esp-idf #8408: SDK Configuration editor

flair0626, 2022-02-15:
Turning off both Enable Modbus stack support for RTU mode and Enable Modbus stack support for ASCII mode in the SDK Configuration editor will result in an error compiling.

alisitsyn, 2022-02-17:
Hi @flair0626,

Thank you for this issue report. Could you be more specific and specify the ESP-IDF version (commit) you are using, the example and sdkconfig.h file you are using? The ESP-Modbus component requires at least one communication option to be selected. If not, it causes the compilation error (intentional behavior). So, if you selected at least one of the options `Enable Modbus stack support for TCP comm mode` and disabled the `Enable Modbus Stack support for RTU mode`, `Enable Modbus Stack support for ASCII mode` there should not be any error during compiling. However, I have checked this on master and maybe some backport fixes are not merged.

Thanks for the information.

Alvin1Zhang, 2022-03-04:
@flair0626 Thanks for reporting, would you please help share if any more updates for https://github.com/espressif/esp-idf/issues/8408#issuecomment-1042769008? Thanks.

kriegste, 2022-11-10:
This was the first problem I stumbled across when I tried to optimize the sdkconfig.
A user that does not know anything about modbus and does not want to use it, will always try to disable all those options. Because, "I do not use it, I do not need it, it will save some space when I disable it".

alisitsyn, 2023-02-20:
@kriegste , @flair0626 , 
Sorry for late response. Unfortunately this issue will not going to be fixed in the esp-idf. The old freemodbus component is migrated to separate [esp-modbus component](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/protocols.html#id9).

mahavirj, 2023-04-03:
Closing. Please feel free to raise a issue on https://github.com/espressif/esp-modbus (if any)

closed 2023-04-03 by mahavirj