espressif/esp-idf #8409: ESP32-S3R8V issue: sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107

W00ng, 2022-02-15:
## Environment

- Development Kit:      ESP32-S3 Devkit
- Module or chip used:  ESP32-S3R8V
- IDF version (run ``git describe --tags`` to find it):  v0.9.3-145-g3f9d35f
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
- Operating System:     Windows

## Problem Description

SD card initialization failed when i use ESP32-S3R8V. then i change chip to ESP32-S3R8,  it works fine. 

### Expected Behavior

### Actual Behavior

## Debug Logs
**NG log:**
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048d06
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x17fc
load:0x403b6000,len:0xb58
load:0x403ba000,len:0x2fb8
entry 0x403b61e4
I (24) boot: ESP-IDF v4.4-beta1-183-gf23dcd3555-dirt 2nd stage bootloader
I (25) boot: compile time 22:36:31
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling QIO for flash chip GD
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
W (53) bootloader_random: RNG for ESP32-S3 not currently supported
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003c0000
I (93) boot:  3 storage          Unknown data     01 82 003d0000 00100000
I (101) boot: End of partition table
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0c23ch ( 49724) map
I (121) esp_image: segment 1: paddr=0001c264 vaddr=3fc93960 size=02af4h ( 10996) load
I (124) esp_image: segment 2: paddr=0001ed60 vaddr=40374000 size=012b8h (  4792) load
I (132) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=268b8h (157880) map
I (163) esp_image: segment 4: paddr=000468e0 vaddr=403752b8 size=0e6a8h ( 59048) load
I (175) esp_image: segment 5: paddr=00054f90 vaddr=50000000 size=00010h (    16) load
I (181) boot: Loaded app from partition at offset 0x10000
I (181) boot: Disabling RNG early entropy source...
W (184) bootloader_random: RNG for ESP32-S3 not currently supported
I (191) opi psram: vendor id : 0x0d (AP)
I (195) opi psram: dev id    : 0x02 (generation 3)
I (201) opi psram: density   : 0x03 (64 Mbit)
I (206) opi psram: good-die  : 0x01 (Pass)
I (210) opi psram: Latency   : 0x01 (Fixed)
I (215) opi psram: VCC       : 0x00 (1.8V)
I (220) opi psram: SRF       : 0x01 (Fast Refresh)
I (226) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (231) opi psram: BurstLen  : 0x01 (32 Byte)
I (236) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (242) opi psram: DriveStrength: 0x00 (1/1)
W (247) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (258) spiram: Found 64MBit SPI RAM device
I (262) spiram: SPI RAM mode: sram 80m
I (266) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (273) cpu_start: Pro cpu up.
I (277) cpu_start: Starting app cpu, entry point is 0x40375338
0x40375338: call_start_cpu1 at D:/esp-idf/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (572) spiram: SPI SRAM memory test OK
I (581) cpu_start: Pro cpu start user code
I (581) cpu_start: cpu freq: 240000000
I (581) cpu_start: Application information:
I (581) cpu_start: Project name:     mp3_demo
I (581) cpu_start: App version:      v0.9.3-145-g3f9d35f-dirty
I (582) cpu_start: Compile time:     Feb 14 2022 22:36:18
I (582) cpu_start: ELF file SHA256:  466e4755feb88558...
I (582) cpu_start: ESP-IDF:          v4.4-beta1-183-gf23dcd3555-dirt
I (582) heap_init: Initializing. RAM available for dynamic allocation:
I (583) heap_init: At 3FC96F90 len 00049070 (292 KiB): D/IRAM
I (583) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (583) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (583) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (584) spi_flash: detected chip: gd
I (584) spi_flash: flash io: qio
I (585) sleep: Configure to isolate all GPIO pins in sleep state
I (585) sleep: Enable automatic switching of GPIO sleep configuration
I (586) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (586) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (586) i2c_bus: i2c0 bus inited
I (586) gpio: GPIO[47]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (586) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (586) gpio: GPIO[48]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (606) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (606) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
E (606) bsp_sdcard: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
I (606) main: init done

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2022-02-15:
Gpio 47-48 are in vdd_spi power domain and on S3R8V the voltage is 1.8v

W00ng, 2022-02-15:
> Gpio 47-48 are in vdd_spi power domain and on S3R8V the voltage is 1.8v

ok, i got it，thank you very much