espressif/esp-idf #8410: The advanced_https_ota example run faild，esp_https_ota: Failed to get image length

yan-qi-2009, 2022-02-15:
1、在Amazon S3中上传用于升级的BIN文件，并复制了URL
2、在示例程序advanced_https_ota中，配置好BIN文件URL，并且替换连接aws用的PEM文件
3、配置好WiFi
运行程序
出现以下日志：
[0;32mI (9637) advanced_https_ota_exampl
e: Starting Advanced OTA example[0m
[0;31mE (13217) esp_https_ota: Failed to get image length[0m
[0;31mE (13217) esp_https_ota: esp_https_ota_read_img_desc: Invalid argument[0m
[0;31mE (13217) advanced_https_ota_example: esp_https_ota_read_img_desc failed[0m
[0;31mE (13227) advanced_https_ota_example: ESP_HTTPS_OTA upgrade failed[0m
是哪里出了问题？

ESP-YJM, 2022-02-16:
@yan-qi-2009 能否提供下 URL 地址。

Alvin1Zhang, 2022-03-04:
@yan-qi-2009 您好，烦请能够提供下 URL 地址？谢谢。

cvonk, 2022-03-04:
In my case `esp_https_ota_begin` returned ESP_OK, but left `https_ota_handle` NULL.
As a consequence, `esp_https_ota_get_img_desc` returns ESP_ERR_INVALID_ARG.

cvonk, 2022-03-04:
Line 217-220 in esp_https_ota.c read
```
        if (err != ESP_OK || esp_http_client_get_status_code(https_ota_handle->http_client) != HttpStatus_Ok) {
            ESP_LOGE(TAG, "Failed to get image length");
            goto http_cleanup;
        }
```
So it would return ESP_OK even though `esp_http_client_get_status_code` indicated an error.

AxelLin, 2022-03-05:
> So it would return ESP_OK even though `esp_http_client_get_status_code` indicated an error. In my case `esp_http_client_get_status_code` returned 404 (not found), because I forgot to upload the file to the server.

@cvonk 
Above issue is already fixed: https://github.com/espressif/esp-idf/issues/8195, try update your esp-idf.

yan-qi-2009, 2022-03-07:
@ESP-YJM @Alvin1Zhang  是aws s3设置问题，已解决

Alvin1Zhang, 2022-03-07:
@cvonk 感谢反馈，谢谢。

closed 2022-03-07 by Alvin1Zhang

cvonk, 2022-03-08:
Graag gedaan (Dutch)