espressif/esp-idf #8412: Ethernet doesn't work with IPv6

copercini, 2022-02-15:
## Environment

- Development Kit:      Custom ESP32 board + LAN8720A
- Module or chip used:  ESP32 chip
- IDF version: v4.4 (release)
- Build System:  idf.py
- Compiler version: 8.4.0
- Operating System:     Linux (Ubuntu 20.04.3 LTS)
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I'm trying to access a webserver running on ESP32 using IPv6 from a computer in local network. Via Wi-Fi it works fine, but via Ethernet (LAN8720A), only IPv4 works

### Expected Behavior

Be able to access ESP32 webserver from Ethernet IPv6 

### Actual Behavior

Only IPV4 seems to work in Ethernet, a local IPv6 is printed in UART, but webpages don't load
Using Wi-Fi interface, both work, IPV4 and IPv6

### Steps to reproduce

Build examples/protocols/http_server/simple with these options enabled:
[x] connect using Ethernet interface. I'm using Internal EMAC and LAN87xx phy with the proper pins configuration for my board
[x] Obtain IPv6 address

Access it from a web browser

### Code to reproduce this issue

It's possible to reproduce using http_server/simple example described above

I also have tried enabling `SLAAC` and `DHCPv6` options in menuconfig, it gets a public IPv6, but also doesn't work

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/8073418/sdkconfig.zip)

copercini, 2022-02-16:
A possible fix for it: https://github.com/espressif/arduino-esp32/issues/6283

david-cermak, 2022-02-16:
@copercini Note, that you can also enable the MLD6 flag directly in the user code with

```cpp
netif_set_flags(lwip_netif, NETIF_FLAG_MLD6);
```

(see [this forum post](https://www.esp32.com/viewtopic.php?t=23308#p84188))

copercini, 2022-02-16:
@david-cermak Nice, it also worked with MLD6 flag instead patching lwip, but I think it should be a default thing, otherwise IPv6 simply doesn't work in ethernet interface. At least here, by default it shows the local ethernet IPv6 in the UART logs, but it's not accessible without this flag or patch

AxelLin, 2022-08-10:
fixed in v4.4: 8215bd6670e545e08bbfcf84a91702e16f3d19fd

Alvin1Zhang, 2022-08-19:
Thanks for reporting, this seems to duplicate https://github.com/espressif/esp-idf/issues/8602, and the fix already back ported to release/4.4, feel free to reopen. Thanks.

closed 2022-08-19 by Alvin1Zhang