espressif/esp-idf #8413: About EventBits_t of xEventGroupWaitBits

nopnop2002, 2022-02-15:
About EventBits_t of xEventGroupWaitBits

## Environment
```
$ idf.py --version
ESP-IDF v4.4-dev-3573-gfb24a2941c-dirty
```

## Problem Description
In xEventGroupWaitBits, panic occurs when 0xFFFFFFFF is specified for EventBits_t.
sizeof(EventBits_t) is 4(=32Bits).

This is OK.
```
EventBits_t bits = xEventGroupWaitBits(gpio_event_group, 0xFFFFFF, pdTRUE, pdFALSE, portMAX_DELAY);
```

This is FAIL.
```
EventBits_t bits = xEventGroupWaitBits(gpio_event_group, 0xFFFFFFFF, pdTRUE, pdFALSE, portMAX_DELAY);
```

# STDOUT
```
I (0) cpu_start: Starting scheduler on APP CPU.
I (324) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (334) main: sizeof(io_conf.pin_bit_mask)=8
I (334) gpio: GPIO[32]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (344) main: sizeof(EventBits_t)=4

assert failed: xEventGroupWaitBits event_groups.c:370 (( uxBitsToWaitFor & 0xff000000UL ) == 0)

Backtrace:0x400818da:0x3ffb86600x4008523d:0x3ffb8680 0x4008aac5:0x3ffb86a0 0x40087f0a:0x3ffb87c0 0x400d4a03:0x3ffb87f0 0x40088601:0x3ffb8810
0x400818da: panic_abort at /home/nop/esp-idf/components/esp_system/panic.c:402

0x4008523d: esp_system_abort at /home/nop/esp-idf/components/esp_system/esp_system.c:129

0x4008aac5: __assert_func at /home/nop/esp-idf/components/newlib/assert.c:85

0x40087f0a: xEventGroupWaitBits at /home/nop/esp-idf/components/freertos/event_groups.c:370 (discriminator 1)

0x400d4a03: gpio_task2 at /home/nop/rtos/gpio_isr/build/../main/main.c:62

0x40088601: vPortTaskWrapper at /home/nop/esp-idf/components/freertos/port/xtensa/port.c:130

ELF file SHA256: 804be893aadbdf66

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6832
load:0x40078000,len:14836
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?
```

KaeLL, 2022-02-16:
If I recall correctly, the 8 MSBits of the EventGroup are reserved, so you can only the 24 LSBits.

nopnop2002, 2022-02-16:
@KaeLL 

My results and your point are in agreement.

However, I can't find any documentation on the number of valid bits in EventBits_t.

AxelLin, 2022-02-16:
https://github.com/espressif/esp-idf/blob/release/v4.4/components/freertos/event_groups.c#L59-L72

nopnop2002, 2022-02-16:
I found documentation about the number of valid bits in EventBits_t.

https://www.freertos.org/FreeRTOS-Event-Groups.html

The number of bits (or flags) stored within an event group is 8 if configUSE_16_BIT_TICKS is set to 1, or 24 if configUSE_16_BIT_TICKS is set to 0. The dependency on configUSE_16_BIT_TICKS results from the data type used for thread local storage in the internal implementation of tasks.

closed 2022-02-16 by nopnop2002