espressif/esp-idf #8414: Secure Boot V2 failure: Sig block 0 invalid: Image digest does not match

badevos, 2022-02-16:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2022-02-16:
@badevos Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

badevos, 2022-02-16:
Sorry, my fault: I did prepare this, but my details are gone ...

## Environment

- Development Kit:      [ESP32-DevKitC vs. propriatary board]
- Kit version [v4]
- Module or chip used:  [ESP32-WROVER-IE on devkit, ESP32-S0WD on propriatary board]
- IDF version (run ``git describe --tags`` to find it): v4.4
- Build System:         [idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB for devkit, propriatary on our own board]

## Problem Description
I have a configuration and some scripts to activate Secure Boot and Flash Encryption and burn the relevant fuses from the bootloader (first run). This does work on the devkit, but does not on our own board. The digest calculation of the app image does not match the one from the bootloader on our own board.

### Expected Behavior

Bootloader to run and execute exactly the same on both boards. 

### Actual Behavior

The digest calculation of the app image does not match the one from the bootloader on our own board.

### Steps to reproduce

I have made some scripts (see attachements). Those are shell scripts, but I renamed them to .txt to get them uploaded here. They should be ran in this order:
1. build.txt
2. nvs.txt
3. flash.txt

[build.txt](https://github.com/espressif/esp-idf/files/8078514/build.txt)
[nvs.txt](https://github.com/espressif/esp-idf/files/8078516/nvs.txt)
[flash.txt](https://github.com/espressif/esp-idf/files/8078515/flash.txt)

The flash memory is connected as follows:
[image]

### Code to reproduce this issue
I have no specific code, as I rely on the bootloader code.

## Debug Logs

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1e (SPets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0038,len:10140
ho 0 tail 12 room 4
load:0x40078000,len:22792
ho 0 tail 12 room 4
load:0x40080400,len:3464
0x40080400: _init at ??:?

entry 0x40080640
I (31) boot: ESP-IDF v4.4 2nd stage bootloader
I (31) boot: compile time 14:43:01
I (31) boot: chip revision: 3
I (31) boot.esp32: SPI Speed      : 40MHz
I (35) boot.esp32: SPI Mode       : DIO
I (38) boot.esp32: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
[... 57 lines cut ...]
I (832) esp_image: segment 2: paddr=000090d4 vaddr=40080400 size=00d88h (  3464) 
I (834) esp_image: Verifying image signature...
I (836) secure_boot_v2: Secure boot V2 is not enabled yet and eFuse digest keys are not set
I (844) secure_boot_v2: Verifying with RSA-PSS...
Sig block 0 invalid: Image digest does not match
E (852) secure_boot_v2: Secure Boot V2 verification failed.
E (857) esp_image: Secure boot signature verification failed
I (863) esp_image: Calculating simple hash to check for corruption...
E (879) esp_image: Image hash failed - image is corrupt
W (879) esp_image: image corrupted on flash
E (879) secure_boot_v2: bootloader image appears invalid! error 8194
I (884) efuse: Batch mode of writing fields is cancelled
E (889) boot: Secure Boot v2 failed (8194)
E (893) boot: OTA app partition slot 1 is not bootable
E (898) boot: No bootable app partitions in the partition table

```

## Other items if possible
my `sdkconfig`:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8078538/sdkconfig.txt)
the generated booloader (signed):
[bootloader.bin.gz](https://github.com/espressif/esp-idf/files/8078540/bootloader.bin.gz)
my partytion table:
[partitions.csv](https://github.com/espressif/esp-idf/files/8078544/partitions.csv)

badevos, 2022-02-16:
Some more info:
The application works without secure boot, with exactly the same partition table.

I tried by manually burning the digest on both devkit and proprietary board:
It works on devkit
It fails on proprietary board.
key was burned as follows:

`espefuse.py burn_key_digest ./secure_boot_signing_key.pem`

This is a bootlog with debug info for the failing case:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0038,len:12820
load:0x40078000,len:24040
load:0x40080400,len:3844
0x40080400: _init at ??:?

entry 0x4008067c
I (30) boot: ESP-IDF v4.4 2nd stage bootloader
I (30) boot: compile time 14:49:09
D (30) bootloader_flash: non-XMC chip detected by SFDP Read (FF), skip.
D (34) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (40) boot: chip revision: 3
D (43) boot.esp32: magic e9
D (45) boot.esp32: segments 03
D (48) boot.esp32: spi_mode 02
D (51) boot.esp32: spi_speed 00
D (54) boot.esp32: spi_size 02
I (57) boot.esp32: SPI Speed      : 40MHz
I (60) boot.esp32: SPI Mode       : DIO
I (64) boot.esp32: SPI Flash Size : 4MB
D (67) boot: Enabling RTCWDT(9000 ms)
I (71) boot: Enabling RNG early entropy source...
D (75) bootloader_flash: mmu set paddr=00010000 count=1 size=c00 src_addr=10000 src_addr_aligned=10000
D (84) boot: mapped partition table 0x10000 at 0x3f400000
D (90) flash_parts: partition table verified, 8 entries
I (94) boot: Partition Table:
I (97) boot: ## Label            Usage          Type ST Offset   Length
D (103) boot: load partition table entry 0x3f400000
D (108) boot: type=1 subtype=4
I (111) boot:  0 nvs_key          NVS keys         01 04 00011000 00001000
D (117) boot: load partition table entry 0x3f400020
D (122) boot: type=1 subtype=2
I (125) boot:  1 nvs              WiFi data        01 02 00012000 00020000
D (131) boot: load partition table entry 0x3f400040
D (136) boot: type=1 subtype=0
I (139) boot:  2 otadata          OTA data         01 00 00032000 00002000
D (145) boot: load partition table entry 0x3f400060
D (150) boot: type=1 subtype=1
I (153) boot:  3 phy_init         RF data          01 01 00034000 00001000
D (159) boot: load partition table entry 0x3f400080
D (164) boot: type=1 subtype=3
I (167) boot:  4 coredump         Unknown data     01 03 00035000 00020000
D (173) boot: load partition table entry 0x3f4000a0
D (178) boot: type=0 subtype=10
I (181) boot:  5 ota_0            OTA app          00 10 00060000 001a0000
D (187) boot: load partition table entry 0x3f4000c0
D (192) boot: type=0 subtype=11
I (195) boot:  6 ota_1            OTA app          00 11 00210000 001a0000
I (201) boot: End of partition table
D (205) boot: OTA data offset 0x32000
D (208) bootloader_flash: mmu set paddr=00030000 count=1 size=2000 src_addr=32000 src_addr_aligned=30000
D (217) boot: otadata[0]: sequence values 0xffffffff
D (222) boot: otadata[1]: sequence values 0xffffffff
[... 164 lines cut ...]
D (1737) efuse: In EFUSE_BLK2__DATA0_REG is used 32 bits starting with 0 bit
D (1744) efuse: In EFUSE_BLK2__DATA1_REG is used 32 bits starting with 0 bit
D (1751) efuse: In EFUSE_BLK2__DATA2_REG is used 32 bits starting with 0 bit
D (1758) efuse: In EFUSE_BLK2__DATA3_REG is used 32 bits starting with 0 bit
D (1764) efuse: In EFUSE_BLK2__DATA4_REG is used 32 bits starting with 0 bit
D (1771) efuse: In EFUSE_BLK2__DATA5_REG is used 32 bits starting with 0 bit
D (1778) efuse: In EFUSE_BLK2__DATA6_REG is used 32 bits starting with 0 bit
D (1785) efuse: In EFUSE_BLK2__DATA7_REG is used 32 bits starting with 0 bit
I (1792) secure_boot_v2: Verifying with RSA-PSS...
Sig block 0 invalid: Image digest does not match
E (1800) secure_boot_v2: Secure Boot V2 verification failed.
E (1806) esp_image: Secure boot signature verification failed
I (1811) esp_image: Calculating simple hash to check for corruption...
D (1817) bootloader_flash: mmu set paddr=00000000 count=1 size=9f40 src_addr=1000 src_addr_aligned=0
D (1837) boot: Calculated hash: f20a58a787f66d9eb4dc0ea46c5be9bd9150de836b73a8d30309bcb9e020d045
E (1837) esp_image: Image hash failed - image is corrupt
D (1840) boot: Expected hash: e843adf9931a550bf592be648fbe5620f7bdfe639e5c5806ceb805fda0fe67fd
W (1848) esp_image: image corrupted on flash
E (1852) secure_boot_v2: bootloader image appears invalid! error 8194
I (1858) efuse: Batch mode of writing fields is cancelled
E (1863) boot: Secure Boot v2 failed (8194)
E (1867) boot: OTA app partition slot 1 is not bootable
D (1872) boot: Can't boot from zero-length partition
E (1877) boot: No bootable app partitions in the partition table
```

gb-123-git, 2022-02-17:
+1
I am also having the same problem.

I am using ESP32-DevkitC-VE (Wrover Module) with 8MB Flash.

If I burn the digest using : espefuse.py --port COM6 burn_key_digest X:\secure_boot_signing_key.pem,
I get

"Sig block 0 invalid: Image digest does not match"

If I don't burn the digest manually, I get similar message as you are getting in your first post.
I also have encryption enabled.
The app works perfectly if SecurebootV2 is not enabled.

gb-123-git, 2022-02-17:
@badevos - Did you sign your Partition-table.bin also ? 
Reason I am asking is that closely looking at your logs (and mine) it seems that the Bootlader is able to verify the signature of the partition table but not the application (ota) which is strange... I may be wrong though...

badevos, 2022-02-17:
Hi,
I did not 'sign' the partition table. 
As far as I can tell from the logs, the digest from the bootloader image is correct, but the one calculated from the application (ota, either 1 or 2) does not match. The partition table however is parsed correctly, as I do indeed flash the application in both OTA slots and both images are tried from the correct addresses, i.e. the addresses specified in the partition table. 
The signatures are appended (with some padding/alignment) _after_ the actual binary file and there is no overlap in my case.

gb-123-git, 2022-02-17:
Mahavir already confirmed that partition does not need to be signed in Secure Boot V2. (https://esp32.com/viewtopic.php?f=13&t=26199).
I was merely trying to understand why line is showing (after partition table)
`I (351) secure_boot_v2: Signature verified successfully!`
and then before loading app :
`I (451) secure_boot_v2: Verifying with RSA-PSS...
Sig block 0 invalid: Image digest does not match
E (459) secure_boot_v2: Secure Boot V2 verification failed.`

I am also getting similar result on my end.

badevos, 2022-02-17:
Meanwhile I tried to read back the content of the memory and dump it to a file. This fails as well.

For example to read back the bootloader, I tried:
```shell
esptool.py dump_mem 0x1000 0xd000 ./bootloader_dump.bin
esptool.py v3.2-dev
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 40:91:51:20:8d:c8
Uploading stub...
Running stub...
Stub running...

A fatal error occurred: Invalid head of packet (0x46): Possible serial noise or corruption.
```
I also tried with `-b 460800`, as this is the baudrate used for flashing.
Just reading (`esptool .py read_mem 0x1000` resp) did not work either.

This error I do get on both devkit and proprietary board.
Although the fuses burned on the boards should still allow to read/write to the flash.

Attached is the output of `espefuse.py summary` for both boards:
[fuses_proprietary.txt](https://github.com/espressif/esp-idf/files/8092229/fuses_proprietary.txt)
[fuses_devkit.txt](https://github.com/espressif/esp-idf/files/8092230/fuses_devkit.txt)

As mentioned, the system does work on the devkit in my case, hence the secure boot key is burned as well as the flash encryption key on this board.
On our proprietary board, no fuses have been burned yet, I hope to find a fix, since I only have 2 boards available ...

gb-123-git, 2022-02-17:
In my case even the devkit from espressif does not work.
Dump of my Efuse gives me :
`BLOCK2          (secure_boot_v1 s) [2 ] read_regs: b9ce6f61 b758814c 47d82945 e856d4c6 4ca3e63a 6566da9b 155ce3f4 8a35d1bc`
If I read Block 2, it gives me : 
`BLOCK2 (BLOCK2):                                   Secure boot key
   = 61 6f ce b9 4c 81 58 b7 45 29 d8 47 c6 d4 56 e8 3a e6 a3 4c 9b da 66 65 f4 e3 5c 15 bc d1 35 8a R/-`

I noticed that the dump has a block in reverse order.

gb-123-git, 2022-02-17:
I noticed ABS_DONE_1 is not burnt in your proprietary board. But be careful, if you burn it as 1 manually, you might not be able to go back!

badevos, 2022-02-17:
Block 2 is burned automatically in the devkit and manually in the proprietary board. Automatic burning in the bootloader occurs _after_ successfull verification of the application signature, which is a phase I never reached on proprietary board off course.

KonstantinKondrashov, 2022-02-18:
Hi @badevos!
You tried to read out the flash content of bootloader using cmd `esptool.py dump_mem 0x1000 0xd000 ./bootloader_dump.bin`, but actually it reads memory area not flash content. The right cmd for this `read_flash `.
`esptool.py read_flash 0x1000 0xd000 bootloader_dump.bin`
Could you try to read out the app.bin from flash and check it with the `verify_signature` cmd?
`espsecure.py verify_signature   build/bootloader/bootloader.bin -v 2  --keyfile secure_boot_signing_key.pem `
I suspect that that app on the flash is not correct. (maybe using script hids some error messages during flashing).

gb-123-git, 2022-02-18:
Here is the method I tried seeing your post :
`esptool.py read_flash 0x1000 0xd000 bootloader_dump.bin`
`espsecure.py decrypt_flash_data --keyfile X:\Key.bin --output bootloader_dec.bin --address 0x1000 --flash_crypt_conf 0xf bootloader_dump.bin`
'espsecure.py verify_signature bootloader_dec.bin -v 2 --keyfile X:\secure_boot_signing_key.pem'

Error I get : 
`A fatal error occurred: Signature block 0 invalid. Signature could not be verified with the provided key.`

I am using the following command to sign :  
`espsecure.py sign_data --version 2 --keyfile X:\secure_boot_signing_key.pem --output bootloader_signed.bin bootloader.bin`

mahavirj, 2022-02-18:
@gb-123-git 

As I understand from summary and information shared by @badevos, this issue is happening only on custom hardware. Is that case with you as well?

Note: I am going through all logs, I will share my findings soon.

gb-123-git, 2022-02-18:
@mahavirj  No. I am using ESP32-DevkitC-VE V4 (WROVER-E with 8 MB Flash) directly made by Espressif.

I am manually signing the bootloader.bin after the build succeeds by using : 
`espsecure.py sign_data --version 2 --keyfile X:\secure_boot_signing_key.pem --output bootloader_signed.bin bootloader.bin`
I am also signing the partition.bin, firmware.bin 

Then I am merging the bin using : 
esptool.py --chip esp32 merge_bin -o merged-firmware.bin --flash_mode qout --flash_freq 80m 0x1000 bootloader_signed.bin 0xD000 partition_signed.bin 0x30000 app_signed.bin

Then I am encrypting the data:
`espsecure.py encrypt_flash_data --keyfile X:\Key.bin --address 0x0 --output firmware_enc.bin merged-firmware.bin`

Then flashing : 
`esptool.py write_flash 0x0 firmware_enc.bin`

Just to further clarify, I have unchecked the option of 'Sign Binaries during build'

mahavirj, 2022-02-18:
@gb-123-git 

>  I am using ESP32-DevkitC-VE V4 (WROVER-E with 8 MB Flash) directly made by Espressif.

I would request to file new issue in that case. Original report in this issue is specific to custom hardware only, as I understand it.

> Does it matter if I am using --flash_mode qout --flash_freq 80m ?

It is possible that `esptool` is re-writing bootloader header here, if settings in flashing command and actual build configuration of bootloader do not match. This can cause digest mismatch issue, because bootloader image on host and on device are now different. You may read out bootloader from flash again and compare it against original image to confirm this.

Have you set same flash configuration during build as well? Can you please share your log during `write_flash` command?

gb-123-git, 2022-02-18:
I get the following on the cmd prompt:
`esptool.py v3.2
Found 3 serial ports
Serial port COM6
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.......
Detecting chip type... ESP32
Chip is ESP32-D0WDQ5-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 7c:87:ce:e3:55:6c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x001e6fff...
Compressed 1994752 bytes to 1915940...
Wrote 1994752 bytes (1915940 compressed) at 0x00000000 in 169.7 seconds (effective 94.0 kbit/s)...
Hash of data verified.`

gb-123-git, 2022-02-18:
@badevos  
let me disable the 'ESPTOOLPY_FLASHSIZE_DETECT' and then test it out. Should be able to do this in the next 15-20 mins

KonstantinKondrashov, 2022-02-18:
@badevos 

>A side question I have, I did not know the exact size of the bootloader, so I did read the complete bootloader partition, which is bigger than what was originally flashed. I can imagine that the size of the file provided to espsecure verify_signature ... is important. Can this be confirmed please?

The file size for this `verify_signature` command should not be smaller than the original file, if you read out more than necessary, then anyway it will verify the signature correctly.
 
> The fourth byte is 0x30 vs. 0x20, meaning that they are not equal, and signature verification should fail for that reason.

Yes, Hash is calculated over the whole image including this header up to end including padding + checksum. 

> I assume that disabling ESPTOOLPY_FLASHSIZE_DETECT and applying the correct size information can solve this,

Yes, you need to specify the certain flash size `ESPTOOLPY_FLASHSIZE` instead of detection option.

mahavirj, 2022-02-18:
@badevos 

> The comparison reveals that the binary images are actually equal, except for a single byte, in the very first line.
for the bootloader.bin, as produced by the build system: this is the first line:
000000 e9 03 02 30 7c 06 08 40 ee 00 00 00 00 00 03 00 00 00 00 00 00 00 00 01 38 00 ff 3f 14 32 00 00
for the bootloader_dump.bin, as read back from the flash, this is the first line:
000000 e9 03 02 20 7c 06 08 40 ee 00 00 00 00 00 03 00 00 00 00 00 00 00 00 01 38 00 ff 3f 14 32 00 00
(The first six digits is the offset, the rest is the actual data).

> The fourth byte is 0x30 vs. 0x20, meaning that they are not equal, and signature verification should fail for that reason.

This makes sense now. 

If we compare this data against [image header](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/app_image_format.html#_CPPv418esp_image_header_t) then this field corresponds to flash size (4-bit field) and this has been updated (see `esp_image_flash_size_t` for details on flash size) in header of bootloader image on device.

I would recommend using `--flash-size keep` to `esptool` which would ensure to keep flash size in header intact.

gb-123-git, 2022-02-18:
@mahavirj 

I have already tried specifying the --flash-size as 8MB in the merge_bin command and the same size in the sdkconfig with ESPTOOLPY_FLASHSIZE_DETECT disabled.

It still doesn't work.

gb-123-git, 2022-02-18:
One possibility : `esptool.py merge_bin` pads the data till the next partition offset rather than the actual signed size. and the size header gets 'updated' with the new size breaking the signatures ?

gb-123-git, 2022-02-18:
Update on comparing my bootloader_signed.bin with the decoded bootloder.dump, I am also getting a difference of 01

Signed:
`E9 XX X0 XX XX XX XX XX XX XX XX XX XX XX XX XX`
Read from Flash:
`E9 XX X1 XX XX XX XX XX XX XX XX XX XX XX XX XX`

This is just as @badevos reported.

gb-123-git, 2022-02-19:
@mahavirj @KonstantinKondrashov @badevos 

I did some more tests. Here is what I found out :

If I flashed merged.bin after encryption, the third bit is always wrong even when `ESPTOOLPY_FLASHSIZE_DETECT` is set as disabled and `flash_size` is explicitly set as 8MB.
If I flash the individual bootloader.bin after I flash merged.bin (just the bootloader at address 0x1000)
I see the following in console: 
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
**mode:QIO**, clock div:1
load:0x3fff0038,len:11660
ho 0 tail 12 room 4
load:0x9d7c1054,len:776671543
1150 mmu set 00010000, pos 00010000
1150 mmu set 00020000, pos 00020000

This is when I have set mode as "DIO" in sdkconfig and also explicitly passing `--flash_mode dio`

gb-123-git, 2022-02-19:
@badevos
Were you flashing the files after merging were you flashing separately ?

gb-123-git, 2022-02-19:
I can confirm my ESP32 has started to work.

The header size seemed to be an issue as it was getting over-riden by a different value while flashing bootloader.

@badevos - You can safely try flashing your board. In case you are flashing/building using PlatformIO, your value in menuconfig / sdkconfig will NOT work. It will be over-ridden by platform.ini or your boardconfig.json

gb-123-git, 2022-02-19:
@mahavirj @KonstantinKondrashov 

Thanks for your support!

@badevos 

Thanks for posting this issue. It has given me a lot of insight on how this works.. I am still here to help you out in case you need it.

badevos, 2022-02-21:
@gb-123-git: thanks for your confirmation. 

I don't use platform-io, yet I understand the confusion due to `platform.ini` and/or `boardconfig.json`.
I did not merge any binaries together as I rely on the bootloader for the encryption and fuse burning at runtime (current situation). This allows me to verify the fuses and then finally burn the latest fuses. 

Your remarks regarding 'qio'/'dio' doesn't ring a bell with me.

Unfortunately I have a priority change and I can't test this further on short term. Nevertheless within a few weeks I have to prepare production scripts and pick this up again then. Sorry for the inconvenience.

gb-123-git, 2022-02-21:
@badevos 

My job is already done and I have successfully implemented the secure rom v2 at my end.
I was just thanking you for posting this issue as it gave me a deeper insight and helped me fix my problem.

I just wanted to help you fix your board as well. :)

mahavirj, 2022-03-22:
@badevos @gb-123-git 

Further update on this issue:

- We had changed default `esptool` behavior for `--flash-size` to `keep` sometimes back. Please find relevant commit at https://github.com/espressif/esptool/commit/15f791b0. In this particular case flashing scripts used `detect` strategy and hence the issue.
- We have filed an internal enhancement tracker in `esptool` project to not re-write flash settings header in bootloader in case secure boot is enabled. This will be addressed soon in subsequent `esptool` release.

I will close this issue for now. Please feel free to comment here in case you would like to keep this issue open or any further concerns.

closed 2022-03-22 by mahavirj

martinjaeger, 2024-07-26:
@mahavirj Has this issue been fixed in the meantime? I still can't flash using `idf.py flash` with Secure Boot enabled, even with newest 4.7 release of `esptool.py`.