espressif/esp-idf #8416: wrong description for esp_transport_connect, and no way to get socket

maa1370, 2022-02-16:
## Environment

- IDF version (run ``git describe --tags`` to find it):
    // v4.4 and master

## Problem Description

in tcp_transport component, in esp_transport.h return value for int esp_transport_connect() is described as:
- socket for will use by this transport
- (-1) if there are any errors, should check errno
but check last lines of ssl_connect in transport_ssl.c :
    ssl->sockfd = ssl->tls->sockfd;
    return 0;
so at least for ssl, the description is wrong, and there is no public method to get socket number.

### Solution

either
-correct the bug in any transport type with this bug to return socket number when esp_transport_connect() succeeds
-correct the description, and move esp_transport_get_socket() from private_include/esp_transport_internal.h to include/esp_transport.h

euripedesrocha, 2022-02-21:
Hi @maa1370, thank you for reporting this. 
We are working to fix the docs. 

I would like to understand your use case for getting the socket number.

maa1370, 2022-02-21:
thanks
I need to know socket number used for transport layer since i need to change the socket receive timeout using `setsockopt( socket, SOL_SOCKET, SO_RCVTIMEO, ...)` somewhere in the code. do we have any other solution for this without directly using the socket number?

euripedesrocha, 2022-02-21:
Does `esp_transport_read` solves your issue? You can use the timeout parameter. 

And if it's possible, we would like to hear about your usage of `tcp_transport`, why are you using it directly?

maa1370, 2022-02-21:
I'm trying to develop some code using [esp-aws-iot](https://github.com/espressif/esp-aws-iot/tree/release/beta), see [tls_freertos.c](https://github.com/espressif/esp-aws-iot/blob/release/beta/port/network_transport/tls_freertos.c). 
so it seems that i should change parameters passed to TLS_FreeRTOS_recv() since it is using esp_transport_read with that parameter internally? And even if i changed SO_RCVTIMEO directly it had no effect since timeout for each esp_transport_read () is configured as one of it's parameters?
(and as described, i'm not using tcp_transport directly)

maa1370, 2022-02-21:
and another question, do we have some type of receive callback in tcp_transport? same as what described [here](https://github.com/aws/amazon-freertos/issues/1646#issuecomment-567758726).

euripedesrocha, 2022-02-22:
You are correct that SO_RCVTIMEO will have no effect in this case. `esp_transport` uses a poll based strategy to handle timeout. Does using esp-aws-iot `NetworkContext` timeouts solve your issue?

Regarding your second question, we don't have this already but I believe that it's possible using vfs component.

maa1370, 2022-02-23:
yes, it seems using `NetworkContext->receiveTimeoutMs` is the solution for timeout.

thanks for noting vfs, i will check it.

I leave the issue open for wrong description (any maybe the checking against 0 or >=0) i described, feel free to close it when you want.

Regards

closed 2022-05-15 by espressif-bot