espressif/esp-idf #8417: CMake: Source Code Generation fails with "add_dependencies called with incorrect number of arguments"

CvRXX, 2022-02-16:
As described in #4540 I'm using the if statement to prevent the not scriptable error. However, now make returns that add_dependencies is called with the incorrect number of arguments. If I pull the call out of the if statements cmake returns that add_dependecies is not scribtable.

## add_dependecies inside the if
CMakeLists.txt component file (below this is the idf_component_register):
```
if(NOT CMAKE_BUILD_EARLY_EXPANSION)
  add_custom_command(OUTPUT font_main_small.c font_main.c font_numbers_large.c font_numbers_medium.c
      COMMAND python3 generate_fonts.py
      WORKING_DIRECTORY ${CMAKE_CURRENT_LIST_DIR}/fonts
      VERBATIM)

  add_custom_target(fonts DEPENDS font_main_small.c font_main.c font_numbers_large.c font_numbers_medium.c)
  add_dependencies(${COMPONENT_LIB} fonts)
endif()
```
Error:
```
CMake Error at .../graphics/CMakeLists.txt:8 (add_dependencies):
  add_dependencies called with incorrect number of arguments
```

## add_dependecies inside the if
CMakeLists.txt component file (below this is the idf_component_register):
```
if(NOT CMAKE_BUILD_EARLY_EXPANSION)
  add_custom_command(OUTPUT font_main_small.c font_main.c font_numbers_large.c font_numbers_medium.c
      COMMAND python3 generate_fonts.py
      WORKING_DIRECTORY ${CMAKE_CURRENT_LIST_DIR}/fonts
      VERBATIM)

  add_custom_target(fonts DEPENDS font_main_small.c font_main.c font_numbers_large.c font_numbers_medium.c)
endif()
add_dependencies(${COMPONENT_LIB} fonts)
```
Error:
```
  CMake Error at
  .../graphics/CMakeLists.txt:9
  (add_dependencies):

    add_dependencies command is not scriptable
```

ESP-Marius, 2022-02-17:
Hi,

Something like:

```
idf_component_register(SRCS "hello_world_main.c"
                    INCLUDE_DIRS "")

add_custom_command(OUTPUT font_main_small.c font_main.c font_numbers_large.c font_numbers_medium.c
                    COMMAND python3 generate_fonts.py
                    WORKING_DIRECTORY ${CMAKE_CURRENT_LIST_DIR}/fonts
                    VERBATIM)

add_custom_target(fonts DEPENDS font_main_small.c font_main.c font_numbers_large.c font_numbers_medium.c)
add_dependencies(${COMPONENT_LIB} fonts)
target_sources(${COMPONENT_LIB} PRIVATE fonts/font_main_small.c)
```
seems to work for me. (`font_main_small.c` is generated and compiled at least).

Alvin1Zhang, 2022-04-19:
Thanks for reporting.

ESP-Marius, 2022-05-12:
Marking this as resolved then, feel free to reopen if the proposed solution did not work for you.

closed 2022-05-12 by ESP-Marius