espressif/esp-idf #8420: espefuse.py options?

fpgamaster, 2022-02-16:
- IDF version: v5.0-dev-1426-ge899edd793
 - Operating System: Linux

## Problem Description
espefuse.py execute_scripts ./my_efuses.py --do-not-confirm --chip esp32 --force-write-always
Connecting....
usage: espefuse.py [-h] [--chip {auto,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32h2beta1}] [--baud BAUD] [--port PORT]
                   [--before {default_reset,no_reset,esp32r1,no_reset_no_sync}] [--debug] [--virt]
                   [--path-efuse-file PATH_EFUSE_FILE] [--do-not-confirm]
                   {burn_efuse,read_protect_efuse,write_protect_efuse,burn_block_data,burn_bit,adc_info,dump,summary,execute_scripts,check_error,burn_key,burn_key_digest,set_flash_voltage,burn_custom_mac,get_custom_mac}
                   ...
espefuse.py: error: unrecognized arguments: --force-write-always

At the same time:
espefuse.py execute_scripts ./my_efuses.py --do-not-confirm --chip esp32
Connecting......
espefuse.py v3.3-dev
connected chip: ESP32-D0WD-V3 (revision 3), coding scheme NONE (BLK1-3 len=256 bits)
Burn keys to blocks:
 - BLOCK2 -> [85 5e ea e1 3e 78 b5 68 a6 c7 14 31 04 5f 00 46 8e d6 25 63 55 8b 35 ab 81 76 69 1f c0 b4 7b 1d]
        Reversing the byte order

A fatal error occurred:         BLOCK2 is read-protected. The written value can not be read, the efuse/block looks as all 0.
        Burn in this case may damage an already written value.(use '--force-write-always' option to ignore it)

KonstantinKondrashov, 2022-02-17:
Hi @fpgamaster!
The `execute_scripts` command does not have the `--force-write-always` flag. You can see what options are available for the command with `--help`. `espefuse.py -c esp32  execute_scripts   -h`.

Seems inside your `my_efuses.py` you use something `burn_key  flash_encryption  file_to_key.bin`. 
```
A fatal error occurred: BLOCK2 is read-protected. The written value can not be read, the efuse/block looks as all 0.
Burn in this case may damage an already written value.(use '--force-write-always' option to ignore it)
```
This message is related to the `burn_key` from the script. If you want to force it just change it to  `burn_key  flash_encryption file_to_key.bin  --force-write-always`.

Note: seems like you already run this script once, it is why you see the message, maybe you need to skip this command based on read-protection `efuses["BLOCK2"].is_readable()`.

fpgamaster, 2022-02-17:
Hi @KonstantinKondrashov,
Thanks for the reply!
Ok, the execute_scripts command does not have the force option but the error message is misleading.

Best regards

closed 2022-02-22 by espressif-bot