espressif/esp-idf #8421: mfg_gen.py does not handle blank lines correctly

armandr, 2022-02-16:
## Environment

- IDF version: 4.4
- Build System:         idf.py
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt

## Problem Description

//Detailed problem description goes here.

Line 284 increments an index that is used by line 288 without checking.  In my case (despite having a seemingly clean .csv file), "row" is `['']` .  This means that on the second iteration of the inner while loop, `row[index] = val` will access an index that does not exist.

In my case, modifying the line to: `while key_val_new and key_repeated and row[index]:` allowed the program to complete correctly.

### Expected Behavior
Not crash
### Actual Behavior
Crash
### Steps to reproduce
feed a csv resulting in a blank line

Original code:
```cpp
 while row:
            index = -1
            key_val_new = list(zip_longest(keys, row))
            key_val_pair = total_keys_values[:]
            key_repeated = total_keys_repeat[:]
            while key_val_new and key_repeated:
                index = index + 1
                #  if key has repeat tag, get its corresponding value, write to file
                if key_val_new[0][0] == key_repeated[0]:
                    val = key_val_pair[0][1]
                    row[index] = val
                    del key_repeated[0]
                del key_val_new[0]
                del key_val_pair[0]

```

PSONALl, 2022-02-17:
Hi @armandr 
Could you please share your csv file ?
If there is any extra spaces, the row will be taken as empty, so would be helpful to see the csv file.
Thanks

shivani-tipnis, 2022-02-17:
Hi @armandr 
Could you please share your csv file ?
If there is any extra spaces, the row will be taken as empty, so would be helpful to see the csv file.
Thanks

hmalpani, 2022-06-01:
@armandr,
Can you share the csv file you are using? It will be helpful to reproduce the issue.

closed 2022-07-19 by espressif-bot