espressif/esp-idf #8425: Core Dump when combining Bluetooth Serial and ESP-TLS

Sjurinho, 2022-02-17:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v1
- Module or chip used: ESP32-WROOM-32
- IDF version (run git describe --tags to find it): v4.2.2
- Build System: idf.py
- Compiler version (run xtensa-esp32-elf-gcc --version to find it): 8.4.0
- Operating System: Linux
- Using an IDE?: VSCode with the ESP-IDF extension
- Power Supply: USB

## Problem Description

When trying to connect to my esp32 via bluetooth while it is connecting to an mqtt server using tls I am getting a core dump thrown by the bluedroid component inside esp-idf.

### Expected Behavior
The esp32 manages to connect via bluetooth even though the esp-tls handshake is happening

### Actual Behavior
The esp32 has a panic.

### Steps to reproduce

1. Connect to an mqtt server with tls
2. While handshaking, try to connect to esp32 using bluetooth

// If possible, attach a picture of your setup/wiring here.

## Debug Logs
See attached for core dump

## Other items if possible

- [] sdkconfig file (attach the sdkconfig file from your project folder)
- [ x ] coredump (This provides stacks of tasks.)
[debug_logs.txt](https://github.com/espressif/esp-idf/files/8088629/debug_logs.txt)

Alvin1Zhang, 2022-02-18:
Thanks for reporting.

igrr, 2022-02-18:
From the logs it seems that you are running out of heap memory. Please check https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#reducing-heap-usage and see if you can apply any suggestions found there to free up some heap memory.

(At the same time, the code in components/bt/host/bluedroid/stack/btu/btu_task.c:423 should do a better job at checking the result of the call to `osi_alarm_new`, so I'll keep this issue open)

Alvin1Zhang, 2022-04-19:
@Sjurinho Thanks for reporting, would you please help share if any updates for the issue? Thanks.

Sjurinho, 2022-04-20:
@Alvin1Zhang I believe the issue was due to memory usage, as suggested by @igrr. I increased the tls buffer a bit in the sdkconfig and the issue has not arisen since. Thanks for the help

Alvin1Zhang, 2022-04-20:
@Sjurinho Thanks for sharing the updates.