espressif/esp-idf #8426: AP authmode returns as OPEN/insecure when WPA-PSK-SHA256 available

oopscentral, 2022-02-18:
## Environment

- Development Kit:      [esp32doit-devkit-v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    v5.0-dev-1565-g1d77b5b6fd
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

Attempting to enable PMF/11w on OpenWrt, by default, sets hostapd option `wpa_key_mgmt=WPA-PSK WPA-PSK-SHA256`. This appears to be misidentified as WIFI_AUTH_OPEN by my esp32 module. It is correctly identified in esp-idf v3.3.6-16-gcc5440f6a2.

I checked that the misidentification is present in v4.0.3-215-gbe7df8bce9, v4.4 and master.

The router I am using is an old Buffalo WZR-HP-G300NH running OpenWrt 21.02.1.

### Expected Behavior
esp-idf v3.3.6-16 wifi/scan example
```
I (2502) scan: SSID             example-1-wpa2-psk
I (2502) scan: RSSI             -69
I (2502) scan: Authmode         WIFI_AUTH_WPA2_PSK
I (2502) scan: Pairwise Cipher  WIFI_CIPHER_TYPE_CCMP
I (2512) scan: Group Cipher     WIFI_CIPHER_TYPE_CCMP
I (2512) scan: Channel          6

I (2522) scan: SSID             example-2-wpa2-psk-sha256
I (2522) scan: RSSI             -69
I (2532) scan: Authmode         >>WIFI_AUTH_WPA2_PSK<<
I (2532) scan: Pairwise Cipher  WIFI_CIPHER_TYPE_CCMP
I (2542) scan: Group Cipher     WIFI_CIPHER_TYPE_CCMP
I (2542) scan: Channel          6
```
### Actual Behavior
esp-idf v4.0.3-215, v4.4 and v5.0-dev-1565 wifi/scan example
```
I (2879) scan: SSID             example-1-wpa2-psk
I (2879) scan: RSSI             -64
I (2879) scan: Authmode         WIFI_AUTH_WPA2_PSK
I (2879) scan: Pairwise Cipher  WIFI_CIPHER_TYPE_CCMP
I (2889) scan: Group Cipher     WIFI_CIPHER_TYPE_CCMP
I (2889) scan: Channel          6

I (2899) scan: SSID             example-2-wpa2-psk-sha256
I (2899) scan: RSSI             -64
I (2909) scan: Authmode         >>WIFI_AUTH_OPEN<<
I (2909) scan: Pairwise Cipher  WIFI_CIPHER_TYPE_CCMP
I (2919) scan: Group Cipher     WIFI_CIPHER_TYPE_CCMP
I (2919) scan: Channel          6
```
### Steps to reproduce

Run the following on a machine or router capable of creating an AP with hostapd.
```
killall wpa_supplicant
iw dev wlan0 del
iw phy phy0 interface add wlan0 type managed
hostapd hostapd-phy0-wpa2-psk-sha256.conf
```
[hostapd-phy0-psk-sha256.conf.txt](https://github.com/espressif/esp-idf/files/8093447/hostapd-phy0-psk-sha256.conf.txt)
This configuration was based on an OpenWrt 21.02 generated configuration where the `example-2-wpa2-psk-sha256` network originally had PMF enabled.

The only difference between the networks is the addition of WPA-PSK-SHA256 to the `example-2-wpa2-psk-sha256` network.
```
# iw dev wlan0 scan|egrep "^BSS|SSID:|* (Group cipher|Pairwise ciphers|Authentication suites|Capabilities):"
BSS 00:1d:73:aa:bb:cc(on wlan0)
        SSID: example-1-wpa2-psk
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 16-PTKSA-RC 1-GTKSA-RC (0x000c)
BSS 02:1d:73:aa:bb:cc(on wlan0)
        SSID: example-2-wpa2-psk-sha256
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK PSK/SHA-256
                 * Capabilities: 16-PTKSA-RC 1-GTKSA-RC (0x000c)
```

### Code to reproduce this issue

The examples/wifi/scan code is sufficient to observe the behaviour.

closed 2022-03-22 by espressif-bot

AxelLin, 2022-04-11:
@nachiketkukade 
Is this fix required for stable branches? (especially for the fix: In scan, correctly identify AP's with mixed key mgmt)

nachiketkukade, 2022-04-13:
this fix will be available on IDF v4.3 and later releases.