espressif/esp-idf #8427: ESP32-C3 : Unable to use coredump tool, crc error

HiFiPhile, 2022-02-18:
## Environment

- Development Kit:      [[ESP32-C3-DevKitM-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html)]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- IDF version (run ``git describe --tags`` to find it):
    v5.0-dev-1599-gb66cc63c41
- Build System:         [df.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
     (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

Core panic inside the tick hook callback with following code, want to diagnose the cause with core dump, but failed with checksum error.
```
// in app_main()
esp_register_freertos_tick_hook_for_cpu(cbTick, 0);

static void cbTick(void)
{
    static int lvl = 0;
    gpio_set_level(8, lvl);
    lvl = !lvl;
}
```
* In **espcoredump\corefile\loader.py", line 223, `crc` should be replaced by `checksum`, otherwise this error occured.
```
Traceback (most recent call last):
  File "C:\Tool\esp-idf\python_env\idf5.0_py3.9_env\lib\site-packages\construct\lib\containers.py", line 91, in __getattr__
    return self[name]
KeyError: 'crc'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "e:\mcu\esp-idf\components\espcoredump\espcoredump.py", line 352, in <module>
    temp_core_files = dbg_corefile()
  File "e:\mcu\esp-idf\components\espcoredump\espcoredump.py", line 125, in dbg_corefile
    core_elf_path, target, temp_files = get_core_dump_elf(e_machine=exe_elf.e_machine)
  File "e:\mcu\esp-idf\components\espcoredump\espcoredump.py", line 75, in get_core_dump_elf
    loader.create_corefile(exe_name=args.prog, e_machine=e_machine)
  File "e:\mcu\esp-idf\components\espcoredump\corefile\loader.py", line 239, in create_corefile
    self._validate_dump_file()
  File "e:\mcu\esp-idf\components\espcoredump\corefile\loader.py", line 214, in _validate_dump_file
    self._crc_validate()
  File "e:\mcu\esp-idf\components\espcoredump\corefile\loader.py", line 223, in _crc_validate
    'Invalid core dump CRC %x, should be %x' % (data_crc, self.core_src.crc))  # type: ignore
  File "C:\Tool\esp-idf\python_env\idf5.0_py3.9_env\lib\site-packages\construct\lib\containers.py", line 93, in __getattr__
    raise AttributeError(name)
AttributeError: crc
```
* After the correction, for both UART & USB dump mode crc are incorrect:

**Flash mode**
```
I (343) sleep: Configure to isolate all GPIO pins in sleep state
I (350) sleep: Enable automatic switching of GPIO sleep configuration
I (368) pm: Frequency switching config: CPU_MAX: 80, APB_MAX: 80, APB_MIN: 40, Light sleep: DISABLED
I (368) esp_core_dump_flash: Init core dump to flash
I (372) esp_core_dump_flash: Found partition 'coredump' @ 110000 65536 bytes
E (387) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='1748bad0'
Image checksum='e3c5e008'
```
**UART mode**, dump file is attached
```
Initiating core dump!
[... 203 lines cut ...]
3fc8feb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc8fed0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc8fef0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: d41dc5b743c4bfb7
ELF file SHA256: d41dc5b743c4bfb7

Initiating core dump!
I (695) esp_core_dump_uart: Press Enter to print core dump to UART...
I (702) esp_core_dump_uart: Print core dump to uart...
I (708) esp_core_dump_common: Backing up stack @ 0x3fc8f950 and use core dump stack @ 0x3fc903d0
Core dump started (further output muted)
Received   8 kB...
Core dump finished!
Failed to run espcoredump script: Command '['C:\\Tool\\esp-idf\\python_env\\idf5.0_py3.9_env\\Scripts\\python.exe', 'E:\\mcu\\esp-idf\\tools\\idf_monitor_base\\..\\..\\components\\espcoredump\\espcoredump.py', 'info_corefile', '--core', 'C:\\Users\\Mengsk\\AppDa
[dump.txt](https://github.com/espressif/esp-idf/files/8096052/dump.txt)
ta\\Local\\Temp\\tmp635lox2o', '--core-format', 'b64', 'e:\\mcu\\esp32c3\\linkify_fw\\build\\linkify.elf']' returned non-zero exit status 1.
b'espcoredump.py v0.4-dev\r\nTraceback (most recent call last):\r\n  File "E:\\mcu\\esp-idf\\components\\espcoredump\\espcoredump.py", line 350, in <module>\r\n    temp_core_files = info_corefile()\r\n  File "E:\\mcu\\esp-idf\\components\\espcoredump\\espcoredump.py", line 149, in info_corefile\r\n    core_elf_path, target, temp_files = get_core_dump_elf(e_machine=exe_elf.e_machine)\r\n  File "E:\\mcu\\esp-idf\\components\\espcoredump\\espcoredump.py", line 75, in get_core_dump_elf\r\n    loader.create_corefile(exe_name=args.prog, e_machine=e_machine)\r\n  File "E:\\mcu\\esp-idf\\components\\espcoredump\\corefile\\loader.py", line 239, in create_corefile\r\n    self._validate_dump_file()\r\n  File "E:\\mcu\\esp-idf\\components\\espcoredump\\corefile\\loader.py", line 214, in _validate_dump_file\r\n    self._crc_validate()\r\n  File "E:\\mcu\\esp-idf\\components\\espcoredump\\corefile\\loader.py", line 222, in _crc_validate\r\n    raise ESPCoreDumpLoaderError(\r\ncorefile.ESPCoreDumpLoaderError: Invalid core dump CRC b09438d2, should be cfa20ac4\r\n'
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [x] coredump (This provides stacks of tasks.)

erhankur, 2024-04-08:
@HiFiPhile Sorry for the very late response. Do you still have an issue? I gave a try with the master branch, it looks fine.

HiFiPhile, 2024-04-08:
I haven't working on C3 for a moment, if I meet it again I will create a new issue.

closed 2024-04-08 by HiFiPhile