espressif/esp-idf #8428: idf.py size-components: Unexpected section name

balix-microej, 2022-02-18:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-S2
- IDF version: v4.3.2
- Build System:    idf.py
- Compiler version : xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: No

## Problem Description
Trying to get size info on the firmware produced. `idf.py size` works as expected but not `idf.py size-components`.

### Expected Behavior
Print size info about components in an IDF project.

### Actual Behavior
Command failed to run with the following error:

```
  File "C:\Users\balix\git\cc0158_enkiplatform-esp32-s2\enki-platform-esp32-s2-bsp\Drivers\esp-idf\tools\idf_size.py", line 248, in get_display_name_order
    assert len(split_name) == 3 and split_name[0] == '', 'Unexpected section name'
AssertionError: Unexpected section name
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
Press any key to continue . . .
```

If I print the sections here is what I got:

```
['', 'dram0', 'data']
['', 'rtc', 'data']
['', 'dram0', 'bss']
['', 'noinit']
```

So I assume that this section `noinit` does not fit the naming scheme to be processed by this script. Is there any way I can work around this?

balix-microej, 2022-03-02:
Hi @dobairoland. No problem. I did as you advised and it worked. It seems that in fact this is normal in my case. I have a specific library included in my project that have section names that do not match the expected pattern. Thank you for your help!

closed 2022-03-02 by balix-microej

dobairoland, 2022-03-02:
Thanks for the feedback. The above fix will be probably included in ESP-IDF soon.