espressif/esp-idf #8431: ESP32-C3: smartconfig ESPTOUCH_V2 fail

HiFiPhile, 2022-02-18:
## Environment

- Development Kit:      [ESP32-C3-DevKitM-1]
- Kit version : [v1]
- IDF version (run ``git describe --tags`` to find it):
```
 v5.0-dev-1554-g20847eeb96
 f31694c9f1746ba189a4bcae2e34db15135ddb22 components/asio/asio (asio-1-12-0-201-gf31694c9)
 d037ec89546fad14b5c4d5456c2e23a71e554966 components/bootloader/subproject/components/micro-ecc/micro-ecc (v1.0)
 c0a5619d46e56cbf4602327c925fd14f04d125eb components/bt/controller/lib_esp32 (c0a5619)
 a3f7d7ffa2b261b1415042d8b7cd457cc2b4b1de components/bt/controller/lib_esp32c3_family (heads/master)
 554cd264cd028886411b183885bb21b47b8901a6 components/bt/host/nimble/nimble (heads/nimble-1.3.0-idf)
 eeecc49ce8af123cf8ad40efdb9673e37b56230f components/cmock/CMock (v2.5.2-2-geeecc49)
 3aa11612c143c9734d72022720f33e12506f7a2c components/coap/libcoap (v4.2.1-540-g3aa1161)
 fced1a554f1b870f5a4e336431ab9915c4d90326 components/esp_phy/lib (heads/master)
 66cc3ae82adce95f38d1f6738b00fca5bf09fda1 components/esp_wifi/lib (remotes/origin/feature/ftm_support_stage2-29-g66cc3ae)
 845252b0b2b768c7bea96f11af5439fbd12c73f5 components/esptool_py/esptool (v3.2-37-g845252b)
 5545eb14d1117a7865a05a6b4a0f19c62760f8c7 components/ieee802154/lib (heads/master)
 d348621ca93571343a56862df7de4ff3bc9b5667 components/json/cJSON (v1.7.15)
 2749568fe15df2003f6c3f37f0dfd44f8f01fcd6 components/lwip/lwip (remotes/origin/master-224-g2749568f)
 8b0e35f2ad477fcc2a267cf434528024b8499085 components/mbedtls/mbedtls (mbedtls-2.13.1-8771-g8b0e35f2a)
 985078affa8a2d2b56b87c8e6455252850f895c6 components/mqtt/esp-mqtt (ESP-MQTT_FOR_IDF_3.1-293-g985078a)
 901353a8c17d5c3040b3850d571550b0472b921b components/openthread/lib (heads/master)
 e736d2488d9d4f787650d81f2b84f59a194f2b41 components/openthread/openthread (thread-reference-20200818-1424-ge736d2488)
 dac1a65feac4ad72f612aab99f487056fbcf5c1a components/protobuf-c/protobuf-c (v1.3.0)
 f5e26c4e933189593a71c6b82cda381a7b21e41c components/spiffs/spiffs (0.2-221-gf5e26c4)
 c1986586098643edfa6b75a776299c576ad0ce8b components/tinyusb/tinyusb (remotes/origin/esp-based_on_9e992698)
 7d2bf62b7e6afaf38153041a9d53c21aeeca9a25 components/unity/unity (v2.4.3-51-g7d2bf62)
 7e8e249990ec491ec15990cf95b6d871a66cf64a examples/build_system/cmake/import_lib/main/lib/tinyxml2 (2.0.2-659-g7e8e249)
 077758db50d4b55addab56e723a5a379f09077d0 examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib (077758d)
```
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
(crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [USB]

- APP version: Android v2.3.0
## Problem Description

Smartconfig example failed, changed type to `SC_TYPE_ESPTOUCH_V2`
```
I (589) smartconfig: SC version: V3.0.1
I (4679) wifi:ic_enable_sniffer
I (4679) smartconfig: Start to find channel...
I (4679) smartconfig_example: Scan done
I (4819) smartconfig: lock channel: 1
I (4819) smartconfig: TYPE: ESPTOUCH_V2
I (4819) smartconfig: offset:52 tods: 0
I (4819) smartconfig_example: Found channel
I (8089) smartconfig: total_seq: 2
W (8419) smartconfig: found none, invald AP, tmp_crc: 38
W (8449) smartconfig: found none, invald AP, tmp_crc: 38
W (8479) smartconfig: found none, invald AP, tmp_crc: 38
W (8479) smartconfig: found none, invald AP, tmp_crc: 38
W (8499) smartconfig: found none, invald AP, tmp_crc: 38
W (8519) smartconfig: found none, invald AP, tmp_crc: 38
W (8529) smartconfig: found none, invald AP, tmp_crc: 38
```
With v1 it can get the credential successfully.

### Expected Behavior

smartconfig get credential.

### Actual Behavior

smartconfig can't get credential.
[... 99 lines cut ...]
I (4819) smartconfig_example: Found channel
I (8089) smartconfig: total_seq: 2
W (8419) smartconfig: found none, invald AP, tmp_crc: 38
W (8449) smartconfig: found none, invald AP, tmp_crc: 38
W (8479) smartconfig: found none, invald AP, tmp_crc: 38
W (8479) smartconfig: found none, invald AP, tmp_crc: 38
W (8499) smartconfig: found none, invald AP, tmp_crc: 38
W (8519) smartconfig: found none, invald AP, tmp_crc: 38
W (8529) smartconfig: found none, invald AP, tmp_crc: 38
W (8979) smartconfig: found none, invald AP, tmp_crc: 38
W (8999) smartconfig: found none, invald AP, tmp_crc: 38
W (9019) smartconfig: found none, invald AP, tmp_crc: 38
W (9029) smartconfig: found none, invald AP, tmp_crc: 38
W (9039) smartconfig: found none, invald AP, tmp_crc: 38
W (9059) smartconfig: found none, invald AP, tmp_crc: 38
W (9089) smartconfig: found none, invald AP, tmp_crc: 38
W (9359) smartconfig: found none, invald AP, tmp_crc: 38
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)

[template.zip](https://github.com/espressif/esp-idf/files/8098098/template.zip)

HiFiPhile, 2022-02-21:
I think the issue is related to the AP, while it doesn't work in my home it works in my office.
```
Mode: Master | SSID: Mengsk
BSSID: 00:25:9C:13:9E:88
Encryption: WPA2 PSK (CCMP)
Channel: 1 (2.412 GHz)
Mode: N
Channel: 1 (2412 Mhz)
Width: 40 MHz
```
Here is smartconfig log:
```
I (13500) smartconfig: total_seq: 4
I (13500) smartconfig: 0/0 8d9 1075 0
I (13510) smartconfig: 0/0 8db 1048 0
I (13530) smartconfig: 0/0 8dd 1075 0
I (13540) smartconfig: 0/1 [100] 8de 67 0
I (13560) smartconfig: 0/2 [100] 8df 228 0
I (13570) smartconfig: 0/3 [100] 8e0 357 0
I (13590) smartconfig: 0/4 [100] 8e1 464 0
I (13630) smartconfig: 0/5 [92] 8e6 577 0
I (13640) smartconfig: 0/6 [92] 8e7 709 0
I (13640) smartconfig: 0/7 [92] 8e8 832 0
I (13650) smartconfig: 0/8 [92] 8e9 961 0
I (13670) smartconfig: 1/0 8ea 128 0
I (13710) smartconfig: 1/0 8ed 128 0
I (13720) smartconfig: 1/1 [100] 8ee 93 0
I (13730) smartconfig: 1/2 [98] 8f0 232 0
I (13750) smartconfig: 1/3 [98] 8f1 356 0
I (13760) smartconfig: 1/4 [98] 8f2 456 0
I (13780) smartconfig: 1/5 [98] 8f3 578 0
I (13800) smartconfig: 1/6 [98] 8f4 767 0
I (13810) smartconfig: 1/7 [96] 8f6 895 0
I (13830) smartconfig: 1/8 [94] 8f8 1012 0
I (13840) smartconfig: 2/0 8f9 129 0
I (13860) smartconfig: 2/0 8fa 129 0
I (13880) smartconfig: 2/0 8fb 129 0
I (13890) smartconfig: 2/1 [100] 8fc 125 0
I (13910) smartconfig: 2/2 [98] 8fe 242 0
I (13930) smartconfig: 2/3 [98] 8ff 326 0
I (13940) smartconfig: 2/4 [96] 901 453 0
I (13960) smartconfig: 2/5 [96] 902 628 0
I (13970) smartconfig: 2/6 [96] 903 766 0
I (13990) smartconfig: 2/7 [96] 904 894 0
I (14010) smartconfig: 2/8 [94] 906 1006 0
I (14020) smartconfig: 3/0 907 130 0
I (14040) smartconfig: 3/0 909 130 0
I (14050) smartconfig: 3/0 90a 130 0
I (14070) smartconfig: 3/1 [100] 90b 119 0
I (14090) smartconfig: 3/2 [100] 90c 207 0
I (14100) smartconfig: 3/3 [100] 90d 380 0
I (14120) smartconfig: 3/4 [98] 90f 489 0
I (14130) smartconfig: 3/5 [98] 910 578 0
I (14150) smartconfig: 3/6 [96] 912 735 0
I (14170) smartconfig: 3/7 [96] 913 895 0
I (14180) smartconfig: 3/8 [96] 914 960 0
W (14180) smartconfig: found none, invald AP, tmp_crc: 38
I (14220) smartconfig: 0/0 916 1048 0
W (14220) smartconfig: found none, invald AP, tmp_crc: 38
I (14230) smartconfig: 0/0 917 1075 0
W (14230) smartconfig: found none, invald AP, tmp_crc: 38
I (14250) smartconfig: 0/0 919 1048 0
W (14250) smartconfig: found none, invald AP, tmp_crc: 38
I (14260) smartconfig: 0/0 91a 1075 0
W (14260) smartconfig: found none, invald AP, tmp_crc: 38
I (14280) smartconfig: 0/1 [100] 91b 67 0
W (14280) smartconfig: found none, invald AP, tmp_crc: 38
I (14310) smartconfig: 0/3 [92] 91d 357 0
W (14310) smartconfig: found none, invald AP, tmp_crc: 38
I (14330) smartconfig: 0/4 [90] 91f 464 0
[... 87 lines cut ...]
I (15710) smartconfig: 0/0 995 1075 0
I (15720) smartconfig: 0/0 996 1048 0
I (15740) smartconfig: 0/0 997 1075 0
I (15760) smartconfig: 0/1 [98] 999 67 0
I (15770) smartconfig: 0/2 [98] 99a 228 0
W (15770) smartconfig: found none, invald AP, tmp_crc: 38
I (15780) smartconfig: 0/3 [96] 99c 357 0
W (15780) smartconfig: found none, invald AP, tmp_crc: 38
I (15800) smartconfig: 0/4 [96] 99d 464 0
W (15800) smartconfig: found none, invald AP, tmp_crc: 38
I (15820) smartconfig: 0/5 [96] 99e 577 0
W (15820) smartconfig: found none, invald AP, tmp_crc: 38
I (15830) smartconfig: 0/6 [96] 99f 709 0
W (15830) smartconfig: found none, invald AP, tmp_crc: 38
I (15850) smartconfig: 0/7 [96] 9a0 832 0
W (15850) smartconfig: found none, invald AP, tmp_crc: 38
I (15860) smartconfig: 0/8 [94] 9a2 961 0
W (15860) smartconfig: found none, invald AP, tmp_crc: 38
I (15880) smartconfig: 1/0 9a4 128 0
I (15900) smartconfig: 1/0 9a5 128 0
I (15910) smartconfig: 1/0 9a6 128 0
I (15930) smartconfig: 1/1 [100] 9a7 93 0
I (15950) smartconfig: 1/2 [100] 9a8 232 0
I (15960) smartconfig: 1/3 [98] 9aa 356 0
```

HiFiPhile, 2022-02-22:
Here is the pcap from wifi interface.
[esp.zip](https://github.com/espressif/esp-idf/files/8116115/esp.zip)

HiFiPhile, 2022-02-26:
I've explored [lib-esptouch-v2-android](https://github.com/EspressifApp/lib-esptouch-v2-android) for the encoding, and calculated `tmp_crc: 38` is indeed the CRC of my AP's BSSID.

With following code the MCU is also capable of find my AP:
```
    esp_wifi_scan_start(NULL, true);
    ESP_ERROR_CHECK(esp_wifi_scan_get_ap_records(&ap_count, ap_info));
    ESP_LOGI(TAG, "Total APs scanned = %u", ap_count);
    for (int i = 0; (i < 32) && (i < ap_count); i++) {
        ESP_LOGI(TAG, "SSID \t\t%s", ap_info[i].ssid);
        ESP_LOGI(TAG, "BSSID \t\t%02x:%02x:%02x:%02x:%02x:%02x",
                    ap_info[i].bssid[0], ap_info[i].bssid[1], ap_info[i].bssid[2],
                    ap_info[i].bssid[3], ap_info[i].bssid[4], ap_info[i].bssid[5]);
        ESP_LOGI(TAG, "RSSI \t\t%d", ap_info[i].rssi);
        print_auth_mode(ap_info[i].authmode);
        if (ap_info[i].authmode != WIFI_AUTH_WEP) {
            print_cipher_type(ap_info[i].pairwise_cipher, ap_info[i].group_cipher);
        }
        ESP_LOGI(TAG, "Channel \t\t%d\n", ap_info[i].primary);
    }
```
Output:
```
I (2644) smartconfig_example: Total APs scanned = 27
I (2644) smartconfig_example: SSID              Mengsk
I (2644) smartconfig_example: BSSID             00:25:9c:13:9e:88
I (2644) smartconfig_example: RSSI              -32
I (2654) smartconfig_example: Authmode          WIFI_AUTH_WPA2_PSK
I (2654) smartconfig_example: Pairwise Cipher   WIFI_CIPHER_TYPE_CCMP
I (2664) smartconfig_example: Group Cipher      WIFI_CIPHER_TYPE_CCMP
I (2674) smartconfig_example: Channel           1

I (2674) smartconfig_example: SSID              Mengsk-Guest
I (2684) smartconfig_example: BSSID             02:25:9c:13:9e:88
I (2684) smartconfig_example: RSSI              -33
I (2694) smartconfig_example: Authmode          WIFI_AUTH_WPA2_PSK
I (2694) smartconfig_example: Pairwise Cipher   WIFI_CIPHER_TYPE_CCMP
I (2704) smartconfig_example: Group Cipher      WIFI_CIPHER_TYPE_CCMP
I (2714) smartconfig_example: Channel           1
```

mhdong, 2022-03-09:
hi @HiFiPhile  Could you provide the model of the router in issue? We will try to debug locally. 
"I (2644) smartconfig_example: BSSID             00:25:9c:13:9e:88" is this the bissid of  the router in question?
What is the authmode of the router in your home?

HiFiPhile, 2022-03-09:
Hi @mhdong ,

My router is Linksys WRT1200AC, firmware Openwrt 21.02.   00:25:9c:13:9e:88 is my router's BSSID.

More info are in the log:

```
I (2644) smartconfig_example: SSID              Mengsk
I (2644) smartconfig_example: BSSID             00:25:9c:13:9e:88
I (2644) smartconfig_example: RSSI              -32
I (2654) smartconfig_example: Authmode          WIFI_AUTH_WPA2_PSK
I (2654) smartconfig_example: Pairwise Cipher   WIFI_CIPHER_TYPE_CCMP
I (2664) smartconfig_example: Group Cipher      WIFI_CIPHER_TYPE_CCMP
I (2674) smartconfig_example: Channel           1

I (2674) smartconfig_example: SSID              Mengsk-Guest
I (2684) smartconfig_example: BSSID             02:25:9c:13:9e:88
I (2684) smartconfig_example: RSSI              -33
I (2694) smartconfig_example: Authmode          WIFI_AUTH_WPA2_PSK
I (2694) smartconfig_example: Pairwise Cipher   WIFI_CIPHER_TYPE_CCMP
I (2704) smartconfig_example: Group Cipher      WIFI_CIPHER_TYPE_CCMP
I (2714) smartconfig_example: Channel           1
```

HiFiPhile, 2022-04-22:
Hi @mhdong ,

Do you have any news regarding this issue ?

mhdong, 2022-04-22:
Hi @HiFiPhile ，Sorry for not replying in time. The problem has been fixed but the code has not been merged to the master yet. The main reason is that smartconfig code cannot properly handle MAC addresses starting with 0x00 bytes.

mhdong, 2022-04-25:
Hi @HiFiPhile , We have fix this issue in master, commit is with 8ef2ed8269e1f253b38734acc2e5bb17dabd65c9, please confirm.

HiFiPhile, 2022-04-25:
Hi, @mhdong,

I can't find the commit `8ef2ed8269e1f253b38734acc2e5bb17dabd65c9`, maybe I should wait some time ?

Alvin1Zhang, 2022-04-26:
Thanks for reporting, sorry the fix has not synced onto GitHub, will update once the fix is available. Thanks.

Alvin1Zhang, 2022-05-06:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/5c5e23db52471f6bbb2b9252b587b17a4e819e92, we are back porting the fix to release/4.3 and release/4.4, feel free to reopen if the issue still happens. Thanks.

closed 2022-05-06 by Alvin1Zhang

AxelLin, 2022-06-18:
Unfortunately, v4.3.3 does not include this fix.

AxelLin, 2022-07-13:
> Thanks for reporting, fix is available at [5c5e23d](https://github.com/espressif/esp-idf/commit/5c5e23db52471f6bbb2b9252b587b17a4e819e92), we are back porting the fix to release/4.3 and release/4.4, feel free to reopen if the issue still happens. Thanks.

v4.3 branch does not include this fix. @jack0c @mhdong

AxelLin, 2022-07-28:
> Thanks for reporting, fix is available at [5c5e23d](https://github.com/espressif/esp-idf/commit/5c5e23db52471f6bbb2b9252b587b17a4e819e92), we are back porting the fix to release/4.3 and release/4.4, feel free to reopen if the issue still happens. Thanks.

@Alvin1Zhang @freakyxue 

It seems the recent esp32-wifi-lib update in v4.3 branch missed this fix. https://github.com/espressif/esp32-wifi-lib/commit/c9f08e68e38e9412f3f943778af1ba49bb27067d
Could you check what's wrong. It's a long wait for the fix.