espressif/esp-idf #8434: How do we verify second stage bootloader in esp32 ? Any option need to configure ?

enthunilu, 2022-02-19:
## Environment

Development Kit: ESP32-Ethernet kit_A_V1.1
Kit version
Module or chip used: ESP32
IDF version :v4.2.1
Build System: Make|CMake|idf.py
Compiler version : xtensa-esp32-elf-gcc (crosstool NCG esp-2020r3)8.4.0
Operating System: Windows
(Windows only) environment type: Command Prompt
Using an IDE?: Yes (Eclipse)

## Problem Description
I was looking for app image verification and I found the documents for the same. 
I tried to find bootloader verification part, however could not find anything.
So, wanted to know do we verify second stage bootloader as well? If yes, how and from where we can enable it?

Thanks!!!

negativekelvin, 2022-02-19:
If you want the bootloader to be securely verified, use secure boot.  Otherwise it is verified by checksum.

0xjakob, 2022-02-21:
@enthunilu Please let us know if the concise summary of @negativekelvin is enough information.

enthunilu, 2022-02-21:
Hi @negativekelvin  and @0xjakob 

Thanks for your quick response. I'll go through it.

Thanks Again!!

closed 2022-02-21 by enthunilu