espressif/esp-idf #8436: BluFi cannot scan AP when wrong credential were given

innomatica, 2022-02-19:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version : v5.0-dev-1554-g20847eeb96
- Build System:  idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:    Linux
- Using an IDE?: No
- Power Supply:   USB

## Problem Description

I'm writing an Android app to use BluFi. It goes like this

1. Get Wi-Fi List via BluFi
2. User chooses AP and provides passcode
3. User checks connection state

When wrong credential is provided at the step 3, there is no way back to step 1, since the device is busy doing connection with a false credential. It says
```
W (1111737) wifi:sta_scan: STA is connecting, scan are not allowed!
```

I tried  
* `disconnect (0x4) from AP` 
* `set opmode to NULL` then back `set opmode to STA` 
but to no avail.

### Expected Behavior

When `disconnect from AP` is issued, the device should stop connecting to AP.

### Actual Behavior

There seems to be no way to stop the device when it is in connecting state.

### Steps to reproduce

0. Use BluFi protocol
1. Provide wrong Wi-Fi credential
2. Try to get Wi-Fi list

innomatica, 2022-02-19:
I found out that the device eventually stops connecting and responds to `get Wi-Fi list (0x09)` packet after it gets about 5 to 10 repeated packets of `disconnect from AP (0x04)`. 

From the app users point of view, it is not desirable but still workable at the moment. Let me know if I'm doing something wrong.

jack0c, 2022-05-23:
@innomatica Is the disconnect cmd transmitted by  BLE, this will be somehow not so fast. I think this part is not well considered in blufi, we will take a look to have a better solution.

Alvin1Zhang, 2023-04-25:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/0c8272b2e0d9c50eb2a573f1936d28fa53617c96, feel free to reopen. Thanks.

closed 2023-04-25 by Alvin1Zhang