espressif/esp-idf #8438: Wifi stops functioning on a specific router

nonameforgood, 2022-02-19:
## Environment

- Module or chip used: ESP32-WROOM-32E module
- IDF version: 4.3.1
- Build System: idf.py
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System: Windows
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Using the example http_server as-is,
the ESP32 stops responding to http and ping requests when connected to a Technicolor router model CGM4331COM.
After a boot, the problem can appear between 5 seconds to 2 minutes.

Using the same esp32, same firmware, connected to a different SSID on a different router (ASUS TM-AC1900),
the issue does not occur.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

Run http_server example on SSID from Technicolor router.

### Code to reproduce this issue

## Debug Logs

esp_wifi_statis_dump(WIFI_STATIS_ALL) when server responds:

wifi: BUFFER:
 wifi: t  max alloc total_alloc total_free fail_fc  fail_oom
 wifi: 1  32  0     34          34         0        0       
 wifi: 2  32  0     35          35         0        0       
 wifi: 3  32  0     38          38         0        0       
 wifi: 4  32  0     0           0          0        0       
wifi: 5  0   0     4           4          0        0       
wifi: 6  0   0     0           0          0        0       
wifi: 7  32  0     117         117        0        0       
wifi: 8  32  0     213         213        0        0       
wifi: 9  0   0     0           0          0        0       
wifi:
wifi: LMAC TX:
wifi: tx_all   lrc src age to 
wifi: 96       0   0   0   0  
wifi: 0        0   0   0   0  
wifi: 14       0   0   0   0  
wifi: 0        0   0   0   0  
wifi: 0        0   0   0   0  
wifi:
wifi: LMAC RX:
wifi: ampdu    mdpu     mgmt     ctrl     frag
wifi: 19       73       217      20       0   
wifi:
wifi: HW RX:
wifi: rx_suc   rx_end   full#1   full#2  
wifi: 14a      3399     a        0       
wifi:
wifi: Hardware:
wifi: rxbufbk  rxcckerr ofdmerr  rx_agc   rxsf     rx_abort rx_fcs   fifofull
wifi: 0        27be     27f8     0        1        701      25       0       
wifi: afull    rx_oth   tkiperr  samebm   ack_int  rts_int  cts_int  rifs_int
wifi: 0        388      0        0        41       c        2        0       
wifi: blockerr hoperr   txrts    txcts    track    trcts    trigger  txhung   panic   
wifi: 0        0        2        0        1f       c        b        0        0       
wifi:
wifi: DIAG:
wifi: diag0    diag1    diag2    diag3    diag4    diag5    diag6   
wifi: 200000   0        fa00000  210      72800001 a60836   0       
wifi: diag7    diag8    diag9    diag10   diag12   diagsel 
wifi: 10       0        8ac000   40       0        0       
wifi:
wifi: Power save:
wifi: tbtt     beacon   timset   dtimset  tbttdrm  actto    txwake   null1    null0    bcndly   sleep    wake     sleept   waket    totalt   percent 
wifi: 165      163      6        27       145      18       9        17       18       2216     154      154      13581347 3313465  16894812 80.39 %
wifi:
wifi: HMAC RX:
wifi: ifx all      data     ctl    mgmt   bcn/probe
wifi: sta 334      99       0      215    207  
wifi: ap  0        0        0      0      0    
wifi:
wifi: ifx auth     assoc    deauth action amsdu
wifi: sta 2        2        1      3      0    
wifi: ap  0        0        0      0      0    
wifi:
wifi: HMAC TX:
wifi: sta_tx   ap_tx    psq_tx psq_mc psq_uc ps_uc1 reorder oos   amsdu
wifi: 35       0        0      0      0      0       1    

esp_wifi_statis_dump(WIFI_STATIS_ALL) once server and ping strop responding:

wifi: BUFFER:
wifi: t  max alloc total_alloc total_free fail_fc  fail_oom
wifi: 1  32  0     55          55         0        0       
wifi: 2  32  0     35          35         0        0       
wifi: 3  32  0     50          50         0        0       
wifi: 4  32  0     0           0          0        0       
wifi: 5  0   0     4           4          0        0       
wifi: 6  0   0     0           0          0        0       
wifi: 7  32  0     27380       27380      0        0       
wifi: 8  32  0     480         480        0        0       
wifi: 9  0   0     0           0          0        0       
wifi:
wifi: LMAC TX:
wifi: tx_all   lrc src age to 
wifi: 106      0   0   0   0  
wifi: 0        0   0   0   0  
wifi: 34       0   0   0   0  
wifi: 0        0   0   0   0  
wifi: 0        0   0   0   0  
wifi:
wifi: LMAC RX:
wifi: ampdu    mdpu     mgmt     ctrl     frag
wifi: 5492     1480     486      28       0   
wifi:
wifi: HW RX:
wifi: rx_suc   rx_end   full#1   full#2  
wifi: 1d48     6967     a        0       
wifi:
wifi: Hardware:
wifi: rxbufbk  rxcckerr ofdmerr  rx_agc   rxsf     rx_abort rx_fcs   fifofull
wifi: 1        2ea7     2ea8     0        1        1199     36       0       
wifi: afull    rx_oth   tkiperr  samebm   ack_int  rts_int  cts_int  rifs_int
wifi: 0        9b5      0        83       4a       68c      2        0       
wifi: blockerr hoperr   txrts    txcts    track    trcts    trigger  txhung   panic   
wifi: 0        0        2        0        60d      68f      27       1        0       
wifi:
wifi: DIAG:
wifi: diag0    diag1    diag2    diag3    diag4    diag5    diag6   
wifi: 800002   0        fa00000  230      5a13017f 50015410 46c0002 
wifi: diag7    diag8    diag9    diag10   diag12   diagsel 
wifi: 45a00c   0        8ac000   40       0        0       
wifi:
wifi: Power save:
wifi: tbtt     beacon   timset   dtimset  tbttdrm  actto    txwake   null1    null0    bcndly   sleep    wake     sleept   waket    totalt   percent 
wifi: 436      430      10       49       337      22       9        21       22       2198     346      346      31879617 4881437  36761054 86.72 %
wifi:
wifi: HMAC RX:
wifi: ifx all      data     ctl    mgmt   bcn/probe
wifi: sta 28537    26701    0      482    474  
wifi: ap  0        0        0      0      0    
wifi:
wifi: ifx auth     assoc    deauth action amsdu
wifi: sta 2        2        1      3      0    
wifi: ap  0        0        0      0      0    
wifi:
wifi: HMAC TX:
wifi: sta_tx   ap_tx    psq_tx psq_mc psq_uc ps_uc1 reorder oos   amsdu
wifi: 55       0        0      0      0      0       1   

thanks!
Mike

MichelVachon, 2022-02-23:
I can provide more info, whatever is needed.

xueyunfei998, 2022-02-24:
hi @MichelVachon 

Can we have the corresponding sniffer log on channel.

Assuming that the channel of the router is 1,

use a linux laptop,You can use it to sniff the channel 1.

ifconfig wifi. down
iwconfig wifi. mode monitor
ifconfig wifi. up
iwconfig wifi. channel 1
wireshark

AxelLin, 2022-03-05:
@MichelVachon 
Any update regarding https://github.com/espressif/esp-idf/issues/8438#issuecomment-1049436780

nonameforgood, 2022-03-06:
Yes I will provide a log,
Will be from a Mac, not Linux, is that ok?

eritterlin, 2022-03-10:
@AxelLin I have this same problem! I have attached a MAC filtered packet capture of the issue.

In my case it took a fair amount of time to failure, but the interesting segment occurs around packet #13429. 
It looks like some sort of feedback loop where a particular MSDU is failing to be acknowledged. 

Not surprisingly, I lost all ability to communicate. My MQTT client began to fail, and never recovered. 

Decryption key is as follows mountjuliet:Powell-Home 
[FeedbackLoopNonIPLoggingSelection.zip](https://github.com/espressif/esp-idf/files/8224902/FeedbackLoopNonIPLoggingSelection.zip)

eritterlin, 2022-03-16:
@AxelLin Any thoughts on this?

AxelLin, 2022-03-17:
@xueyunfei998 
Any comments regarding https://github.com/espressif/esp-idf/issues/8438#issuecomment-1064213095

MaxwellAlan, 2022-03-17:
Hi @eritterlin @AxelLin 
We found that there is a compatibility issue about keyindex format in some router, please help try with this commit: 832123a5e1f39fc480941dfbc29372dcc171acdd
Looking forward to your reply.
Thanks.

eritterlin, 2022-03-21:
@MaxwellAlan 

I'm using IDF v4.4. If I understand correctly the commit is included in that release. So I don't believe that fixes the issue.

nonameforgood, 2022-03-21:
Seems to be fixed in idf 4.4, I can no longer repro with it.

I haven't tried commit 832123a directly.
Do you need that?

thanks, Mike

AxelLin, 2022-03-22:
> @MaxwellAlan
> 
> I'm using IDF v4.4. If I understand correctly the commit is included in that release. So I don't believe that fixes the issue.

@nonameforgood 
v4.4 includes 094b3dba33bff9233bc0a8f522617223a136f9aa, can you provide wireshark log for analysis?

xueyunfei998, 2022-03-22:
hi @nonameforgood 

Open the log switch of supplicant in menuconfig.

And set the log level to DEBUG in menuconfig.

Then reconnect to wifi, preferably with a wifi sniffer.

nonameforgood, 2022-03-23:
I attached a serial log.
At the end, notice the repeated msg about dup seq 336.
I turned off the esp because it was flooding the log.

I do not have a wifi packet capture this time.
I'll try to do one next time.

[serial capture.txt](https://github.com/espressif/esp-idf/files/8333103/serial.capture.txt)

nonameforgood, 2022-03-23:
I have a sniffer capture too, MAC filtered to esp32 + technicolor router

[wifi stops2.pcap.zip](https://github.com/espressif/esp-idf/files/8337022/wifi.stops2.pcap.zip)
.

xueyunfei998, 2022-03-24:
hi @nonameforgood

Summarize the following through your log:

1 Wifi can connect normally.
:45:23.726 -> V (39972397) wifi:sta recv dup seq=336 tid=0, discard
08:45:23.726 -> V (39972397) wifi:sta recv dup seq=336 tid=0, discard
08:45:23.726 -> V (39972407) wifi:sta recv dup seq=336 tid=0, discard
2 In the process of http communication, many packets with duplicate seqno were received, which led to packet loss in wifi layer.

Send the ssid and password of the router corresponding to the wifi grab packet, and the router model.
Because all the sniffer you send are encrypted packets.

eritterlin, 2022-03-24:
@AxelLin The wireshark log that I provided previously was with version 4.4

nonameforgood, 2022-03-24:
[ch1_esp.pcap.gz](https://github.com/espressif/esp-idf/files/8346243/ch1_esp.pcap.gz)

New capture including EAPOL packets.
PSK=85390b817465ac6298720880795613dc7cb2b4c4e410d680e43cab46448080a3
router:technicolor Model:CGM4331COM Hardware Revision:2.0

thanks
Mike

xueyunfei998, 2022-03-25:
hi @nonameforgood 

I need that ssid and the password of the router

Or you can set the router to open mode and grab a packet again.

negativekelvin, 2022-03-25:
@xueyunfei998 psk is derived from said and password and can be entered into Wireshark to decrypt

nonameforgood, 2022-03-25:
[m.pcap.zip](https://github.com/espressif/esp-idf/files/8353910/m.pcap.zip)

@[xueyunfei998](https://github.com/xueyunfei998)
new capture
ssid:tite-dent-helix-temp
pass:justforyou
router:technicolor Model:CGM4331COM Hardware Revision:2.0

xueyunfei998, 2022-03-28:
hi  @nonameforgood 

[image]

The analysis is as follows:

Since receiving the 241 qos null packet, tcp retransmission will occur later.

Qos null is found. This packet has the same tid value as tcp.

Package 241 Qos null  seqno = 294,Package 240 tcp seqno = 68

This is not normal, it should be Qos null  seqno = 68，Package 240 tcp seqno = 69

This may be a problem of compatibility between esp32 and router.

The package you are catching now is filtered, and some wifi management packages and BA and ack are not caught, resulting in incomplete analysis.

Can help to grab a complete package again, without any filtering, but also need the log corresponding to the package grab.

After grabbing the packet, please provide the mac address corresponding to the router and esp32.

nonameforgood, 2022-03-28:
@[xueyunfei998](https://github.com/xueyunfei998)

[m3.pcap.zip](https://github.com/espressif/esp-idf/files/8365040/m3.pcap.zip)

This is the same capture as m.pcap.zip but with less filtering.
Let me know if that is enough.

Also I don't have the log, let me know if you still need it.
That means a new capture of course :)

mike

xueyunfei998, 2022-03-29:
hi @nonameforgood 
Thanks for the sniffer.

Thanks for the grab bag.

However, it is better to provide the log corresponding to the grabbing package, which is more conducive to analysis.

Can you provide the unfiltered grab and log again?

xueyunfei998, 2022-03-29:
[image]

[image]

root cause:

39916th package seqno = 68, Then back to ba，It means that tid = 0, the packets before the flow sequence number 231 have been received, which leads to this problem.

The fundamental reason is that the 30829 qos null packet was received.

hi @nonameforgood 
You can turn off the ampdu option of tx rx of wifi in menuconfig and then test it again.

Alvin1Zhang, 2022-04-06:
Thanks for reporting, feel free to reopen.

negativekelvin, 2022-04-06:
@Alvin1Zhang I don't think this is solved, just confirmed the problem

MaxwellAlan, 2022-04-06:
@negativekelvin Yeah, since this issue is AP bug, if's difficult for our HW and SW to deal with that issue. So disable AMPDU is the easiest and safe way to workaroud for this issue.

nonameforgood, 2022-04-06:
I don't think closing this issue is the way to go.
There is a bug in ampdu, it might affect many other routers as well.
I think the bug should be looked into more.

negativekelvin, 2022-04-06:
@MaxwellAlan that's only ok for development. There is always a chance a user may have this troublesome router so should ampdu be disabled for all users?  Or choose to disable ampdu at runtime based on what indicator?

AxelLin, 2022-04-07:
> @negativekelvin Yeah, since this issue is AP bug, if's difficult for our HW and SW to deal with that issue. So disable AMPDU is the easiest and safe way to workaroud for this issue.

@MaxwellAlan 
Why other devices connected to the same Technicolor router like windows PCs, macs or phones do not have similar problems?

MaxwellAlan, 2022-04-07:
> @MaxwellAlan Why other devices connected to the same Technicolor router like windows PCs, macs or phones do not have similar problems?

@AxelLin 
Two posibility: 1. Other device may not trigger these wrong tid null data pkt. 2. Other device's hardware different from esp MAC.
Maybe we can try to dynamic disable AMPDU or just filter the wrong pkt when we can get back to shanghai office. :)

jack0c, 2022-05-20:
@xueyunfei998 @MaxwellAlan We can have a fix, please provide here.

MaxwellAlan, 2022-05-23:
Hi @nonameforgood @eritterlin 
Could you replace this wifi lib [gh_8438.tar.gz](https://github.com/espressif/esp-idf/files/8752923/gh_8438.tar.gz) with your $IDF_PATH/components/esp_wifi/lib/esp32 base on master commit:  https://github.com/espressif/esp-idf/commit/70109e23939e13229f36d9e8dca9797abfeeb8aa
Thanks.

nonameforgood, 2022-05-27:
I was trying to first repro with idf 4.4 as-is.
But I can't repro anymore, I've been trying for a few days now.
I don't understand, the router might have had a firmware upgrade maybe.

Next I will test your provided libs from the gz file.

But I want to clarify something first, you want me to take commit 70109e2 and overwrite the content with the gz file. True?

AxelLin, 2022-05-28:
> I was trying to first repro with idf 4.4 as-is. But I can't repro anymore, I've been trying for a few days now. 

1. make sure you have ampdu tx and rx enabled
2. If you cannot reporduce it with 4.4, could you check if v4.3.1 still has the issue?

> I don't understand, the router might have had a firmware upgrade maybe.

You should be able to easily verify the router firmware version and release date.

MaxwellAlan, 2022-06-01:
Hi @nonameforgood 
> But I want to clarify something first, you want me to take commit [70109e2](https://github.com/espressif/esp-idf/commit/70109e23939e13229f36d9e8dca9797abfeeb8aa) and overwrite the content with the gz file. True?
Yes. git checkout 70109e2 and git submodule update -f and then overwrite the lib with gz file.

AxelLin, 2022-06-15:
@nonameforgood  Any update?

MaxwellAlan, 2022-07-13:
Hi 
You can get this fix in master: 7e2ee82160dfde315eec5f97d889775c01ddf872
Thanks.

AxelLin, 2022-07-13:
@MaxwellAlan 
Just remind that this issue was reported on v4.3, please also fix v4.3 branch.

MaxwellAlan, 2022-07-31:
Hi @AxelLin  @nonameforgood 

Release/v4.3 can get fix in this commit: 11508b6fbe9c28845decff66603d0ce79b0c05d0

closed 2022-08-03 by Alvin1Zhang