espressif/esp-idf #8439: idf.py flash can't stop

m986883511, 2022-02-20:
when I execute it, it flash success, But I can't use Ctrl+C to kill it. It  flash again and again
`idf.py -p /dev/ttyUSB0 flash monitor`

`
W (295) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (309) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!
This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision 1, 2MB external flash
Minimum free heap size: 296120 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
Restarting now.
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6752
load:0x40078000,len:14796
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v5.0-dev-1599-gb66cc63c41 2nd stage bootloader
I (27) boot: compile time 07:09:32
I (27) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0782ch ( 30764) map
I (122) esp_image: segment 1: paddr=00017854 vaddr=3ffb0000 size=0243ch (  9276) load
I (126) esp_image: segment 2: paddr=00019c98 vaddr=40080000 size=06380h ( 25472) load
I (140) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=14978h ( 84344) map
I (171) esp_image: segment 4: paddr=000349a0 vaddr=40086380 size=04e24h ( 20004) load
I (180) esp_image: segment 5: paddr=000397cc vaddr=50000000 size=00010h (    16) load
I (186) boot: Loaded app from partition at offset 0x10000
I (186) boot: Disabling RNG early entropy source...
I (200) cpu_start: Pro cpu up.
I (200) cpu_start: Starting app cpu, entry point is 0x40081004
0x40081004: call_start_cpu1 at /opt/esp32/esp-idf/components/esp_system/port/cpu_start.c:152

I (187) cpu_start: App cpu up.
I (214) cpu_start: Pro cpu start user code
I (214) cpu_start: cpu freq: 160000000 Hz
I (214) cpu_start: Application information:
I (219) cpu_start: Project name:     hello_world
I (224) cpu_start: App version:      1
I (229) cpu_start: Compile time:     Feb 20 2022 07:09:24
I (235) cpu_start: ELF file SHA256:  fa55f0d1b0cc7408...
I (241) cpu_start: ESP-IDF:          v5.0-dev-1599-gb66cc63c41
I (248) heap_init: Initializing. RAM available for dynamic allocation:
I (255) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (261) heap_init: At 3FFB2D30 len 0002D2D0 (180 KiB): DRAM
I (267) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (273) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (280) heap_init: At 4008B1A4 len 00014E5C (83 KiB): IRAM
I (287) spi_flash: detected chip: generic
I (291) spi_flash: flash io: dio
W (295) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (309) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!
This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision 1, 2MB external flash
Minimum free heap size: 296120 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
`

AxelLin, 2022-02-20:
Ctrl+]

dobairoland, 2022-02-20:
> Ctrl+]

Yes, this is the way how to stop the monitor. Please note that you are actually not re-flashing over and over. Your application (probably the hello world) is restarting as expected. The monitor (serial terminal) just follows it. The exit sequence for the monitor is CTRL+]. Please see the documentation for more information: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-monitor.html

closed 2022-02-20 by dobairoland

m986883511, 2022-02-22:
> Yes, this is the way how to stop the monitor. Please note that you are actually not re-flashing over and over. Your application (probably the hello world) is restarting as expected. The monitor [...]

thank you

analiviagarbin, 2023-09-01:
i am having the same problem, i already used the CTRL+] but my esp32 does not stop from flashing again and again