espressif/esp-idf #8442: idf V4.4, ESP32 ECO3, high level IRQs

sicanins, 2022-02-20:
We are currently running on idf V4.3 and are checking how to migrate to V4.4.
There seems to be significant changes to the usage of high level IRQs, but the documentation is not very clear.

We are using a level5 IRQ to handle the UART fifo correctly. 
It seems that with V4.4, this should be changed to level4. 
But also it seems that the BT is now using that level4 IRQ and there is a new lib tha enables C functions to be called from level4 context. (With limitations, clear).

Now there is some questions:
a) do we need to enable CONFIG_BTDM_CTRL_HLI with V4.4? Why was it created at all? We did not find any info on that.
b) whats the recommended way now to implement the high level IRQ?
  - use the new library and call C-code
  - squeeze our code into the .S file and conditionally call it before the code that handles these BT irqs?
  - disable the CONFIG_BTDM_CTRL_HLI and use the level 4 IRQ like we did with the level 5 before?
c) for these 3 options, will the CONFIG_ESP32_ECO3_CACHE_LOCK_FIX work always?

It is important to know right now, because we need to extend the functionality of our high level IRQ. We dont want to do it all again soon.

Thanks!

blueMoods, 2022-02-21:
Hi @sicanins , for your questions.

a) `CONFIG_BTDM_CTRL_HLI` is used for fixing some Bluetooth bugs and improving it's performance. If you use Bluetooth in your project, suggest to enable this.
b)  
    * If you used Bluetooth,  suggest to use the way #2(squeeze our code into the .S file and conditionally call it before the code that handles these BT irqs) to  implement the high level IRQ. As there are many restrictions to call the C-code in level 4 interrupt, and it is not for customers. 
    * If you did not use Bluetooth, just disable `CONFIG_BT_ENABLED` and set `CONFIG_ESP_SYSTEM_CHECK_INT_LEVEL` to `Level 4 interrupt`. Then it is same as before.
    
c) `CONFIG_ESP32_ECO3_CACHE_LOCK_FIX` will works well when using the above options.

sicanins, 2022-02-21:
Thanks for the clear and fast answer!

closed 2022-02-21 by sicanins

borayildizme, 2022-05-19:
Hi @blueMoods ,
I work with @sicanins and tried to implement your suggested way#2 as we use the Bluetooth. It is clear that we will squeeze our UART code into the .S file and share the Level 4 interrupt with Bluetooth. However we have problem with registering Level 4 interrupt for UART. 

What is the proper way of registering to shared high level irq?

blueMoods, 2022-05-20:
you can  squeeze your code into the .S file and conditionally call it before the code that handles these BT irqs?, It is the same as you did, before.

Pay attention to that: Do not use C code. 

You can refer to components\esp_system\port\soc\esp32\highint_hdl.S

borayildizme, 2022-05-20:
actually my question is about how we allocate the level4 interrupt also for UART. so how we initialize it?

borayildizme, 2022-05-20:
before we were able to allocate level5 interrupt by following call:

`  ESP_ERROR_CHECK_WITHOUT_ABORT(
      esp_intr_alloc(ETS_UART1_INTR_SOURCE, ESP_INTR_FLAG_LEVEL5 | ESP_INTR_FLAG_IRAM, NULL, NULL, &_irq_handle) );`

Now it doesnt work for level4 interrupt. I guess this is a sharing issue.

sicanins, 2022-05-20:
Let me add here, that the "normal" allocation of a high level IRQ fails here. As far as we understand the timer and the BT are using the only two level IRQs available for level4. So we need to know how to "add" our IRQ to one of the slots...

blueMoods, 2022-05-20:
I understand, now.

There are some issue in `esp_intr_alloc` after adding the feature to config `CONFIG_ESP_SYSTEM_CHECK_INT_LEVEL`. We will fix it as soon as possible.

borayildizme, 2022-05-20:
are there any workaround can you suggest for the time being?

sicanins, 2022-05-22:
Hi, I dig a bit deeper.
we are using CONFIG_ESP_SYSTEM_CHECK_INT_LEVEL_5=y

Reading soc/esp32/include/soc/soc.h:
```
#if CONFIG_ESP_SYSTEM_CHECK_INT_LEVEL_5
//interrupt cpu using table, Please see the core-isa.h
/*************************************************************************************************************
 *      Intr num                Level           Type                    PRO CPU usage           APP CPU uasge
 *      0                       1               extern level            WMAC                    Reserved
 *      1                       1               extern level            BT/BLE Host HCI DMA     BT/BLE Host HCI DMA
 *      2                       1               extern level
 *      3                       1               extern level
 *      4                       1               extern level            WBB
 *      5                       1               extern level
 *      6                       1               timer                   FreeRTOS Tick(L1)       FreeRTOS Tick(L1)
 *      7                       1               software                BT/BLE VHCI             BT/BLE VHCI
 *      8                       1               extern level            BT/BLE BB(RX/TX)        BT/BLE BB(RX/TX)
 *      9                       1               extern level
 *      10                      1               extern edge
 *      11                      3               profiling
 *      12                      1               extern level
 *      13                      1               extern level
 *      14                      7               nmi                     Reserved                Reserved
 *      15                      3               timer                   FreeRTOS Tick(L3)       FreeRTOS Tick(L3)
 *      16                      5               timer                   Reserved                Reserved
 *      17                      1               extern level
 *      18                      1               extern level
 *      19                      2               extern level
 *      20                      2               extern level
 *      21                      2               extern level
 *      22                      3               extern edge
 *      23                      3               extern level
 *      24                      4               extern level
 *      25                      4               extern level            BT/BLE Controller       BT/BLE Controller
 *      26                      5               extern level            TG1_WDT & CACHEERR
 *      27                      3               extern level            Reserved                Reserved
 *      28                      4               extern edge
 *      29                      3               software                BT/BLE hli              BT/BLE hli
 *      30                      4               extern edge             Reserved                Reserved
 *      31                      5               extern level            IPC_ISR                 IPC_ISR
 *************************************************************************************************************
 */
```

Suggests, that the file hal/esp32/interrupt_descriptor_table.c can be modified.
From 
```//This is basically a software-readable version of the interrupt usage table in include/soc/soc.h
const static int_desc_t interrupt_descriptor_table [32]={
    { 1, INTTP_LEVEL, {INTDESC_RESVD,  INTDESC_RESVD } }, //0
    { 1, INTTP_LEVEL, {INTDESC_RESVD,  INTDESC_RESVD } }, //1
    { 1, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //2
    { 1, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //3
    { 1, INTTP_LEVEL, {INTDESC_RESVD,  INTDESC_NORMAL} }, //4
    { 1, INTTP_LEVEL, {INTDESC_RESVD,  INTDESC_RESVD } }, //5
    { 1, INTTP_NA,    {INT6RES,        INT6RES       } }, //6
    { 1, INTTP_NA,    {INTDESC_SPECIAL,INTDESC_SPECIAL}}, //7
    { 1, INTTP_LEVEL, {INTDESC_RESVD,  INTDESC_RESVD } }, //8
    { 1, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //9
    { 1, INTTP_EDGE , {INTDESC_NORMAL, INTDESC_NORMAL} }, //10
    { 3, INTTP_NA,    {INTDESC_SPECIAL,INTDESC_SPECIAL}}, //11
    { 1, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //12
    { 1, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //13
    { 7, INTTP_LEVEL, {INTDESC_RESVD,  INTDESC_RESVD } }, //14, NMI
    { 3, INTTP_NA,    {INT15RES,       INT15RES      } }, //15
    { 5, INTTP_NA,    {INTDESC_SPECIAL,INTDESC_SPECIAL} }, //16
    { 1, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //17
    { 1, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //18
    { 2, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //19
    { 2, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //20
[... 28 lines cut ...]
    { 1, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //12
    { 1, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //13
    { 7, INTTP_LEVEL, {INTDESC_RESVD,  INTDESC_RESVD } }, //14, NMI
    { 3, INTTP_NA,    {INT15RES,       INT15RES      } }, //15
    { 5, INTTP_NA,    {INTDESC_SPECIAL,INTDESC_SPECIAL} }, //16
    { 1, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //17
    { 1, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //18
    { 2, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //19
    { 2, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //20
    { 2, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_NORMAL} }, //21
    { 3, INTTP_EDGE,  {INTDESC_RESVD,  INTDESC_NORMAL} }, //22
[... cut ...]
    { 5, INTTP_LEVEL, {INTDESC_NORMAL, INTDESC_RESVD } }, //26
    { 3, INTTP_LEVEL, {INTDESC_RESVD,  INTDESC_RESVD } }, //27
    { 4, INTTP_EDGE,  {INTDESC_NORMAL, INTDESC_NORMAL} }, //28
    { 3, INTTP_NA,    {INTDESC_SPECIAL,INTDESC_SPECIAL}}, //29
    { 4, INTTP_EDGE,  {INTDESC_RESVD,  INTDESC_RESVD } }, //30
    { 5, INTTP_LEVEL, {INTDESC_RESVD,  INTDESC_RESVD } }, //31
};```

Can you please confirm this? I understand that this is only a hack, but we need to go on with development urgently.

Thanks!