espressif/esp-idf #8443: cannot read all PGN(ID) being sent via CAN

MatheusCaltran, 2022-02-20:
1 ESP32 send + TJA1050
2 ESP32 receiver , get only the PGN(ID) I want and assign it to the variables  + TJA1050

I'm using an ESP32 to simulate 6 PGN(ID) with random values ​​and send via can, everything working fine for now 
when (2)esp32 receives the information, esp32 can only read that it has received 2 or 3 PGN(ID)

[image]

Dazza0, 2022-02-21:
@MatheusCaltran What driver is this? It doesn't look like the TWAI driver in ESP-IDF.

MatheusCaltran, 2022-02-21:
so i changed the library and it worked

Dazza0, 2022-02-22:
@MatheusCaltran Again, I can't tell from your code snippet what library/driver you were using before, and what library/driver you are currently using. Judging from your code snippet you were/are using an Arduino library?

Dazza0, 2022-02-25:
@MatheusCaltran Closing this issue as it's unrelated to ESP-IDF.

closed 2022-02-25 by Dazza0