espressif/esp-idf #8444: compiled went error on "cdc_acm_host" demo

Rita858, 2022-02-21:
Hi all,
    I have face some trouble on compiling "[cdc_acm_host](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc/cdc_acm_host)" demo.
    branch of my sdk is master with commit f02169bc049.
    The error message after the compilation is completed is as follows：
    
[image]

   Looking forward to your reply. Thank you.
   Best regard.

Dazza0, 2022-02-21:
@Rita858 Did you call `idf.py set-target esp32s2/s3` before building?

Rita858, 2022-02-21:
I set esp32c3

> @Rita858 Did you call `idf.py set-target esp32s2/s3` before building?

Dazza0, 2022-02-21:
@Rita858 Are you still facing a compilation error after setting the target to ESP32-S2 or ESP32-S3?

Dazza0, 2022-02-25:
@Rita858 Closing this issue. Please feel free to reopen if necessary.

closed 2022-02-25 by Dazza0