espressif/esp-idf #8447: NVS Partition Generator incorrect or esp32 incorrect reading pre build nvs

igor-d-n, 2022-02-21:
## Environment

- Development Kit:     ESP32-D
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.4
- Build System:  idf.py
- Operating System:   Windows 10
- Using an IDE?: Yes Visual Studio with VisualMicro
- Power Supply:    USB

## Problem Description
I prepared NVS partition using manual from https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_partition_gen.html There was no errors. But after I uploaded partition to the esp32 board I saw that all keys exists but all values are empty.

## Output log

```
NVS key 'batch', type '33', len=-1515870811, val=
NVS key 'otaSalt', type '33', len=-1515870811, val=
NVS key 'authSalt', type '33', len=-1515870811, val=
NVS key 'actKeySalt', type '33', len=-1515870811, val=
NVS key 'actKeyPart', type '33', len=-1515870811, val=
NVS key 'test', type '20', val=-1515870811
```

### Code to show nvs data

```cpp
esp_err_t err;
const char nvsName[] = "pwds";

err = nvs_flash_init_partition(nvsName);
printf("\nnvs init pwds err=%d\n", err);

nvs_handle_t nvsHandle;
err = nvs_open("storage", NVS_READWRITE, &nvsHandle);
printf("\nnvs open pwds err=%d\n", err);

nvs_iterator_t it = nvs_entry_find(nvsName, "testPwds", NVS_TYPE_ANY);
while (it != NULL) {
nvs_entry_info_t info;
nvs_entry_info(it, &info);
it = nvs_entry_next(it);

if (info.type != 33) {
	int v;
	nvs_get_i32(nvsHandle, info.key, &v);
	printf("NVS key '%s', type '%d', val=%d\n", info.key, info.type, v);
} 
else {
	char buf[100];
	size_t len;
	err = nvs_get_str(nvsHandle, info.key, buf, &len);
	printf("NVS key '%s', type '%d', len=%d, val=%s\n", info.key, info.type, (int)len, buf);
}
};
```

## partitions
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1E0000,
app1,     app,  ota_1,   0x1F0000,0x1E0000,
nvs_key,  data, nvs_keys,0x3D0000,0x1000
pwds,     data, nvs,     0x3D1000,0x3000
spiffs,   data, spiffs,  0x3D4000,0x2C000,
```

## NVS csv file text
key,type,encoding,value
testPwds,namespace,,
batch,data,string,batch111
otaSalt,data,string,11111111111111111111111
authSalt,data,string,2222222222222222222222222
actKeySalt,data,string,3333333333333333333333333
actKeyPart,data,string,44444444444444444444444
test,data,i32,12324241

## How it was done
1. prepared it with command: 
```
PS C:\Users\igord\Documents\esp32-experiments\nvs> python C:/Users/igord/esp/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py generate .\nvs-pwds.csv nvs-pwds.bin 0x3000
C:\Users\igord\esp\esp-idf\components\nvs_flash\nvs_partition_generator\nvs_partition_gen.py:28: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  import distutils.dir_util
Creating NVS binary with version: V2 - Multipage Blob Support Enabled
Created NVS binary: ===> C:\Users\igord\Documents\esp32-experiments\nvs\nvs-pwds.bin
```
It looks ok. 
2. Then I uploaded it:
```
esptool.py -p COM4 -b 921600 --before default_reset --after no_reset --chip esp32 write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x3D1000 nvs-pwds.bin
esptool.py v3.2
Serial port COM4
Connecting...........
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:61:05:30:70:1c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x003d1000 to 0x003d3fff...
Compressed 12288 bytes to 444...
Wrote 12288 bytes (444 compressed) at 0x003d1000 in 0.2 seconds (effective 601.6 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
```

Strange that it creates keys but all data alway empty. It's writable but I need pre uploading nvs data with all values. There is no any advice in documentation and I think it must not work so)

0xjakob, 2022-02-22:
@igor-d-n Please check the return values of the function calls to IDF APIs. The you'll probably see a `ESP_ERR_NVS_NOT_FOUND`, because you use the handle associated with namespace "storage" to find entries of namespace "testPwds".

See the lines in your code vs. in the NVS csv file:
```
err = nvs_open("storage", NVS_READWRITE, &nvsHandle);
```
```
testPwds,namespace,,
```

igor-d-n, 2022-02-23:
@0xjakob thx, I'm blind=) Th was other mistake. If you use not default partition you must use function 
nvs_open_from_partition(nvsName, nvsNamespace, NVS_READONLY, &nvsHandle);

closed 2022-02-23 by igor-d-n

0xjakob, 2022-02-24:
@igor-d-n No problem. But there are two things here:
* NVS partition names: As you stated, if you're not using the default partition, you have to use different functions for initializing and opening handles where you specify the partition name.
* namespaces: You have to use the correct namespace when searching, writing, reading entries. Entries with the same name but different namespace are treated as distinct entries.