espressif/esp-idf #8448: Core 0 panic'ed on line 781 of llc.c

ndevev, 2022-02-22:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version v4
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.3.2
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

The board is connected to an AP and there is a permanent websocket connection with ping-pong, and an opened BT LE connection to a peripheral.
After some random time (sometimes several times within an hour, other times only after 1 day) restarts happened because the Core  0 panic'ed. 
There is no communication on the websocket (except the ping pong messages) and the BLE connection, when the crashes are happened.
The crashes not occurred if there is no BT and websocket connection, only the Wi-Fi is connected.

The crashes occurred with idf v4.3.2 and v4.3.1 and v3.40200.210118 and with the ESP32-Wrover-Kit v4 (ESP32-WROVER) and my custom board with ESP32-WROOM-32E.

## Debug Logs
With idf v4.3.2: IllegalInstruction!!!
```
I (4928302) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,0>, prof:1
I (4929342) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (4937322) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,0>, prof:1
I (4938352) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
ASSERT_PARAM(1 5), in llc.c at line 781
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008324c: 0000f01d 00004136 f01d0000
0x4008324c: coex_bt_callback at arch_main.c:?

Core  0 register dump:
PC      : 0x40083253  PS      : 0x00060e30  A0      : 0x80083631  A1      : 0x3ffe2910
0x40083253: r_assert at ??:?

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000005  A5      : 0x3ff9def5
A6      : 0x0000030d  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffe2880
A10     : 0x00000000  A11     : 0x3ffe28a3  A12     : 0x3ffe284f  A13     : 0x00000031
A14     : 0x00000000  A15     : 0x3ffe2854  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x40083539  LEND    : 0x40083541  LCOUNT  : 0x00000000
0x40083539: r_assert_param at ??:?

0x40083541: r_assert_param at ??:?

Backtrace:0x40083250:0x3ffe2910 0x4008362e:0x3ffe2930 0x400452c1:0x3ffe2950 0x400feb37:0x3ffe2970 0x40019d11:0x3ffe29b0 0x40055b4d:0x3ffe29d0 0x400f51d3:0x3ffe29f0 0x400f5824:0x3ffe2a10
0x40083250: r_assert at ??:?

0x4008362e: r_assert_param at ??:?

0x400feb37: r_lld_evt_deffered_elt_handler at ??:?

0x400f51d3: r_rw_schedule at ??:?

0x400f5824: btdm_controller_task at ??:?

ELF file SHA256: 073103dea3b4291d

Rebooting...
ets Jul 29 2019 12:21:46
```
```
I (1300251) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,0>, prof:1
I (1301271) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (1347251) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,0>, prof:1
I (1348271) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
ASSERT_PARAM(0 0), in llc.c at line 781
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008324c: 0000f01d 00004136 f01d0000
0x4008324c: coex_bt_callback at arch_main.c:?

Core  0 register dump:
PC      : 0x40083253  PS      : 0x00060a30  A0      : 0x80083631  A1      : 0x3ffe2000
0x40083253: r_assert at ??:?

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ff9def5
A6      : 0x0000030d  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffe1f70
A10     : 0x00000000  A11     : 0x3ffe1f93  A12     : 0x3ffe1f3f  A13     : 0x00000031
A14     : 0x00000000  A15     : 0x3ffe1f44  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x40083539  LEND    : 0x40083541  LCOUNT  : 0x00000000
0x40083539: r_assert_param at ??:?

0x40083541: r_assert_param at ??:?

Backtrace:0x40083250:0x3ffe2000 0x4008362e:0x3ffe2020 0x400452c1:0x3ffe2040 0x400feb37:0x3ffe2060 0x40019d11:0x3ffe20a0 0x40055b4d:0x3ffe20c0 0x400f51d3:0x3ffe20e0 0x400f5824:0x3ffe2100
0x40083250: r_assert at ??:?

0x4008362e: r_assert_param at ??:?

0x400feb37: r_lld_evt_deffered_elt_handler at ??:?

0x400f51d3: r_rw_schedule at ??:?

0x400f5824: btdm_controller_task at ??:?

ELF file SHA256: 073103dea3b4291d

Rebooting...
ets Jul 29 2019 12:21:46
```
With idf v4.3.1: Interrupt wdt timeout on CPU0!!!
```
I (1617465) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (1888515) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,0>, prof:1
I (1889435) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
ASSERT_PARAM(1 5), in llc.c at line 781
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x400835fc  PS      : 0x00060c34  A0      : 0x800452c4  A1      : 0x3ffe5810
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x00000000
A6      : 0x00000004  A7      : 0x3ffbdc28  A8      : 0x800835fc  A9      : 0x3ffe57f0
A10     : 0x00000000  A11     : 0x00000028  A12     : 0x00000014  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40083534  LEND    : 0x4008353b  LCOUNT  : 0x00000000

Backtrace:0x400835f9:0x3ffe5810 0x400452c1:0x3ffe5830 0x400febc7:0x3ffe5850 0x40019d11:0x3ffe5890 0x40055b4d:0x3ffe58b0 0x400f5e7f:0x3ffe58d0 0x400f64a5:0x3ffe58f0 0x4009410e:0x3ffe5920
Core  1 register dump:
PC      : 0x40197666  PS      : 0x00060c34  A0      : 0x800d807d  A1      : 0x3ffce7e0
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x80135fbe  A9      : 0x3ffce7b0
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000014  A13     : 0x000000a5
A14     : 0x000000a5  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x40197663:0x3ffce7e0 0x400d807a:0x3ffce800 0x40091f95:0x3ffce820 0x4009410e:0x3ffce840

ELF file SHA256: 046f76c8f4d5de51

Rebooting...
```
With idf v3.4: also Interrupt wdt timeout on CPU0!!!
```
ASSERT_PARAM(1 5), in llc.c at line 781
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x40083315  PS      : 0x00060434  A0      : 0x800452c4  A1      : 0x3ffb5de0
0x40083315: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x00000000
A6      : 0x00000004  A7      : 0x3ffbdc28  A8      : 0x80083310  A9      : 0x3ffb5dc0
A10     : 0x00000000  A11     : 0x00000028  A12     : 0x00000014  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40083248  LEND    : 0x4008324f  LCOUNT  : 0x00000000
0x40083248: r_assert_param at ??:?

0x4008324f: r_assert_param at ??:?

Backtrace:0x40083312:0x3ffb5de0 0x400452c1:0x3ffb5e00 0x400f9c67:0x3ffb5e20 0x40019d11:0x3ffb5e60 0x40055b4d:0x3ffb5e80 0x400f20ec:0x3ffb5ea0 0x400f26cd:0x3ffb5ec0 0x4008eb59:0x3ffb5ef0
0x40083312: r_assert_param at ??:?

0x400f9c67: r_lld_evt_deffered_elt_handler at ??:?

0x400f20ec: r_rw_schedule at ??:?

0x400f26cd: btdm_controller_task at ??:?

0x4008eb59: vPortTaskWrapper at C:/Users/nevi/.platformio/packages/framework-espidf/components/freertos/xtensa/port.c:143

Core  1 register dump:
PC      : 0x401981d6  PS      : 0x00060134  A0      : 0x800d49b5  A1      : 0x3ffbc660
0x401981d6: http_message_needs_eof at C:/Users/nevi/.platformio/packages/framework-espidf/components/nghttp/port/http_parser.c:2122

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001
A6      : 0x3ffb9fac  A7      : 0x00000000  A8      : 0x800d4906  A9      : 0x3ffbc630
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000014  A13     : 0x000000a5
A14     : 0x000000a5  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x401981d3:0x3ffbc660 0x400d49b2:0x3ffbc680 0x4008f4e9:0x3ffbc6a0 0x4008eb59:0x3ffbc6c0
0x401981d3: http_message_needs_eof at C:/Users/nevi/.platformio/packages/framework-espidf/components/nghttp/port/http_parser.c:2122

0x400d49b2: esp_vApplicationIdleHook at C:/Users/nevi/.platformio/packages/framework-espidf/components/esp_common/src/freertos_hooks.c:63

0x4008f4e9: prvIdleTask at C:/Users/nevi/.platformio/packages/framework-espidf/components/freertos/tasks.c:3386 (discriminator 1)

0x4008eb59: vPortTaskWrapper at C:/Users/nevi/.platformio/packages/framework-espidf/components/freertos/xtensa/port.c:143

ELF file SHA256: 464e9d1fab5f9bdd

Rebooting...
ets Jun  8 2016 00:22:57
```

Alvin1Zhang, 2022-02-24:
@ndevev Thanks for reporting, we have a fix on master branch https://github.com/espressif/esp-idf/commit/b66cc63c413a3b012f8cce9213dcb7ca281501d2, we have back ported the fixes to other branches. Currently the fix on release/4.3 is being synchronised to GitHub, will update once the fix on release/4.3 is available, thanks.

ndevev, 2022-02-24:
> @ndevev Thanks for reporting, we have a fix on master branch [b66cc63](https://github.com/espressif/esp-idf/commit/b66cc63c413a3b012f8cce9213dcb7ca281501d2), we have back ported the fixes to other branches. Currently the fix on release/4.3 is being synchronised to GitHub, will update once the fix on release/4.3 is available, thanks.

This is about ble adv_underrun issue. But there is no ble advertising when the crashes happens in my case.
@Alvin1Zhang Why do you think this solves this problem?

wmy-espressif, 2022-02-24:
Hi @ndevev , the commit provided is a combination of two fixes, one is about ADV_UNDERRUN, and the other is about LLCP transaction collision. If you look at the commit messages of the git submodule: `esp-idf/components/bt/controller/lib_esp32`, you the commit message says:
```
    Update esp32 bt-lib (038aea7)
    
      - LLCP collision fix for ch map update
      - Fix ble ADV_UNDERRUN error lead to RX disable
```

The LLCP collision fix should resolve the issue you met.

ndevev, 2022-02-25:
I checked it with IDF v4.4, which I think contains that fix ( [b2ef617](https://github.com/espressif/esp-idf/commit/b2ef617a34aa793a8922de6b4508f836b7a81904) ).
And it doesn't fix my issue, but I have more specific error description in the panic log and the core dump:
```
I (3061025) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (3331055) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,0>, prof:1
I (3331975) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
ASSERT_PARAM(1 5), in llc.c at line 781
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008f880: f01d092d 00004136 f01d0000
0x4008f880: btdm_sleep_check_duration at D:/prog/platformio/esp-idf/components/bt/controller/esp32/bt.c:1091

Core  0 register dump:
PC      : 0x4008f887  PS      : 0x00060630  A0      : 0x80083c21  A1      : 0x3ffe3cb0
0x4008f887: r_assert at D:/prog/platformio/esp-idf/components/bt/controller/esp32/bt.c:2019

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000005  A5      : 0x3ff9def5
A6      : 0x0000030d  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffe3c20
A10     : 0x00000000  A11     : 0x3ffe3c43  A12     : 0x3ffe3bef  A13     : 0x00000031
A14     : 0x00000000  A15     : 0x3ffe3bf4  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x40083b29  LEND    : 0x40083b31  LCOUNT  : 0x00000000
0x40083b29: r_assert_param at ??:?

0x40083b31: r_assert_param at ??:?

Backtrace:0x4008f884:0x3ffe3cb00x40083c1e:0x3ffe3cd0 0x400452c1:0x3ffe3cf0 0x400fdb9b:0x3ffe3d10 0x40019d11:0x3ffe3d50 0x40055b4d:0x3ffe3d70 0x400f426f:0x3ffe3d90 0x400f48bc:0x3ffe3db0
0x4008f884: r_assert at D:/prog/platformio/esp-idf/components/bt/controller/esp32/bt.c:2018

0x40083c1e: r_assert_param at ??:?

0x400fdb9b: r_lld_evt_deffered_elt_handler at ??:?

0x400f426f: r_rw_schedule at ??:?

0x400f48bc: btdm_controller_task at ??:?

ELF file SHA256: 94bfbd4db998f7df

I (2417) esp_core_dump_flash: Save core dump to flash...
I (2424) esp_core_dump_flash: Erase flash 24576 bytes @ 0x3e0000
I (2769) esp_core_dump_flash: Write end offset 0x5504, check sum length 4
I (2769) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
ets Jul 29 2019 12:21:46
```

```
Crashed task handle: 0x3ffb5c04, name: 'btController', GDB name: 'process 1073437700'

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x40115883
epc2           0x0
epc3           0x4000bff0
epc4           0x400941b5
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x60e20
eps4           0x60e23
eps5           0x0
eps6           0x0
pc             0x4008f887          0x4008f887 <r_assert+3>
lbeg           0x40083b29          1074281257
lend           0x40083b31          1074281265
lcount         0x0                 0
sar            0x4                 4
ps             0x60420             394272
threadptr      <unavailable>
br             <unavailable>
[... 33 lines cut ...]
#2  0x400452c4 in ?? ()
#3  0x400fdb9e in r_lld_evt_deffered_elt_handler () at D:/prog/platformio/esp-idf/components/bt/controller/esp32/bt.c:1888
#4  0x40019d14 in ?? ()
#5  0x40055b50 in ?? ()
#6  0x400f4272 in r_rw_schedule () at D:/prog/platformio/esp-idf/components/bt/controller/esp32/bt.c:1888
#7  0x400f48bf in btdm_controller_task () at D:/prog/platformio/esp-idf/components/bt/controller/esp32/bt.c:1888

======================== THREADS INFO =========================
  Id   Target Id          Frame 
* 1    process 1073437700 r_assert (condition=0x0, param0=1, param1=5, file=0x3ff9def5 <error: Cannot access memory at address 0x3ff9def5>, line=781) at D:/prog/platformio/esp-idf/components/bt/controller/esp32/bt.c:2019
  2    process 1073469004 0x401976aa in esp_pm_impl_waiti () at D:/prog/platformio/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
  3    process 1073467104 0x4000bff0 in ?? ()
  4    process 1073469364 0x4008c668 in esp_crosscore_int_send_yield (core_id=0) at D:/prog/platformio/esp-idf/components/esp_system/crosscore_int.c:144
  5    process 1073578384 0x4008c668 in esp_crosscore_int_send_yield (core_id=0) at D:/prog/platformio/esp-idf/components/esp_system/crosscore_int.c:144
  6    process 1073651544 0x4008c668 in esp_crosscore_int_send_yield (core_id=0) at D:/prog/platformio/esp-idf/components/esp_system/crosscore_int.c:144
  7    process 1073567980 0x4000bff0 in ?? ()
  8    process 1073455228 0x4008c668 in esp_crosscore_int_send_yield (core_id=0) at D:/prog/platformio/esp-idf/components/esp_system/crosscore_int.c:144
  9    process 1073456808 0x400918da in xQueueSemaphoreTake (xQueue=0x3ffba244, xTicksToWait=<optimized out>) at D:/prog/platformio/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:39
  10   process 1073572632 0x4008c668 in esp_crosscore_int_send_yield (core_id=0) at D:/prog/platformio/esp-idf/components/esp_system/crosscore_int.c:144
  11   process 1073559424 0x4008c668 in esp_crosscore_int_send_yield (core_id=0) at D:/prog/platformio/esp-idf/components/esp_system/crosscore_int.c:144
  12   process 1073642988 0x4008c668 in esp_crosscore_int_send_yield (core_id=0) at D:/prog/platformio/esp-idf/components/esp_system/crosscore_int.c:144
  13   process 1073461268 0x4000bff0 in ?? ()
  14   process 1073588764 0x4008c668 in esp_crosscore_int_send_yield (core_id=0) at D:/prog/platformio/esp-idf/components/esp_system/crosscore_int.c:144

```

wmy-espressif, 2022-02-25:
Hi @ndevev , I would like to make sure the correct libbtdm_app.a is used: have you used `git submodule update` when yo checkout the specific commit? And there will be loggings of Bluetooth lib file version when Bluetooth is initialized, please check if your loggings contains:
`BTDM_INIT: BT controller compile version [038aea7]`

ndevev, 2022-02-25:
Oh, sorry, I didn't use submodule update and the BT controller was an older one [6a07b06].
Now I have the latest (I see in the log: BTDM_INIT: BT controller compile version [038aea7]) and started to test it.

AxelLin, 2022-03-02:
v4.3 fix: 2d657035476603a828e8bdd6d6f9dd1f7d103a00

Weijian-Espressif, 2022-03-30:
@ndevev  If there are other question, please close this issue

closed 2022-04-05 by ndevev

bbsan2k, 2022-04-20:
I'm sorry I have to come up with this one again, but I'm still facing the same issue on 4.4 branch:

`Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400901a0: f01d092d 00004136 f01d0000
0x400901a0: btdm_sleep_check_duration at /Users/bbsan/ESP/esp-idf/components/bt/controller/esp32/bt.c:1091

Core  0 register dump:
PC      : 0x400901a7  PS      : 0x00060034  A0      : 0x80083e71  A1      : 0x3ffc2f4c
0x400901a7: r_assert at /Users/bbsan/ESP/esp-idf/components/bt/controller/esp32/bt.c:2019

A2      : 0x00000000  A3      : 0x00000400  A4      : 0x00000000  A5      : 0x3f412694
A6      : 0x0000017d  A7      : 0x00000000  A8      : 0x00000072  A9      : 0x0000006c
A10     : 0x00000003  A11     : 0x3ffbeaa0  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x00000000  A15     : 0x3ffc30cc  SAR     : 0x0000001e  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x40083d79  LEND    : 0x40083d81  LCOUNT  : 0x00000000
0x40083d79: r_assert_param at ??:?

0x40083d81: r_assert_param at ??:?

Backtrace:0x400901a4:0x3ffc2f4c |<-CORRUPTED
0x400901a4: r_assert at /Users/bbsan/ESP/esp-idf/components/bt/controller/esp32/bt.c:2018

ELF file SHA256: d4cba1b92736a028`

My idf is at this commit: 9b75e5664e60886db6a97194177ae174b5bdba9d

AxelLin, 2022-04-27:
@bbsan2k 
Please create a new issue if you still hit the issue with recent esp-idf update. (please check if you still hit the issue after v4.4.1.)

bbsan2k, 2022-04-28:
> @bbsan2k
> Please create a new issue if you still hit the issue with recent esp-idf update. (please check if you still hit the issue after v4.4.1.)

I just checked and it also triggers with 4.4.1.
During debugging I got an assert in some „rwbt.c“ file. Is this part of the compiled Bluetooth binary driver?