espressif/esp-idf #8450: HTTP OTA update feature (not with DEFINES on compile time)

gonzabrusco, 2022-02-22:
**Is your feature request related to a problem? Please describe.**
Kind of. I'm developing using the Arduino framework based on IDF. I would like to use HTTP or HTTPS without cert verification but the HttpsOTAUpdateClass (based on esp_https_ota.c) only works with HTTPS with cert verification. You allowed users to avoid this cert check by providing a define "CONFIG_ESP_HTTPS_OTA_ALLOW_HTTP" but that define is not visible from Arduino stack (IDF comes precompiled).

By the way, log messages are enables too (I see them on the Serial ouput which is not nice if you are using that UART port for something else). There's no way to disable them. For example:

```
E (119505) esp-tls: Failed to open new connection
E (119506) TRANSPORT_BASE: Failed to open a new connection
E (119513) HTTP_CLIENT: Connection failed, sock < 0
E (119515) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (119521) esp_https_ota: Failed to establish HTTP connection
>W                LB111E (132588) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (132589) esp-tls: Failed to open new connection
E (132589) TRANSPORT_BASE: Failed to open a new connection
E (132598) HTTP_CLIENT: Connection failed, sock < 0
E (132598) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (132605) esp_https_ota: Failed to establish HTTP connection
```

**Describe the solution you'd like**
I would like to be hable to set the behaviour setting the right parameters on the passing esp_http_client_config_t 

- if plain HTTP must be used (most inscure) (http url)
- if HTTPS without cert verification must be used (insecure I know) (https url with cert == NULL)
- if HTTPS with cert verification must be used (secure) (https url with cert != NULL)

If you allow those configurations on esp_http_client(), it would provide much more functionality for Arduino users. 

**Describe alternatives you've considered**

Handle everything myself manually with WiFiClientSecure/WiFiClient  but I liked the idea of using a IDF standard function for this.

mahavirj, 2022-02-23:
FYI, this config was renamed in https://github.com/espressif/esp-idf/commit/fbc9c729b296edd4b1bfcb18b58dc2a99913874b. Looping in @me-no-dev for Arduino support.

gonzabrusco, 2022-02-23:
I tried defining CONFIG_ESP_HTTPS_OTA_ALLOW_HTTP and CONFIG_OTA_ALLOW_HTTP but it fails with error:

`esp_https_ota: No option for server verification is enabled in esp_http_client config.`

It seems that when using Arduino core the IDF comes precompiled?