espressif/esp-idf #8451: lwip local port forwarding not working in lwip - corrected files attached

mtnbkr88, 2022-02-22:
I needed the ability to port forward back onto the same network but the current lwip files as of ESP-IDF 4.4.0 do not support this. The attached files add this capability to lwip. I built an ESP32LocalPortForwarder app that provides a web interface to manage port forward rules using these updated lwip files (https://github.com/mtnbkr88/ESP32LocalPortForwarder). Copy the attached files (removing .txt from the name) to the these locations to enable local port forwarding:

ESP-IDF\components\lwip\lwip\src\core\ipv4\ip4.c
ESP-IDF\components\lwip\lwip\src\core\ipv4\ip4_napt.c
ESP-IDF\components\lwip\lwip\src\include\lwip\ip4_napt.h

I also had to modify opt.h and enable the below because menuconfig does not support enabling all these options:
#define IP_FORWARD 1
#define IP_NAPT 1
#define IP_FORWARD_ALLOW_TX_ON_RX_NETIF 1

Please add these changed files to future versions of lwip so local port forwarding will be available for anybody if needed.

[ip4_napt.h.txt](https://github.com/espressif/esp-idf/files/8120624/ip4_napt.h.txt)
[ip4.c.txt](https://github.com/espressif/esp-idf/files/8120625/ip4.c.txt)
[ip4_napt.c.txt](https://github.com/espressif/esp-idf/files/8120626/ip4_napt.c.txt)

0xjakob, 2022-02-23:
@mtnbkr88 Would you mind creating a pull request for this? Then we can easily have discussions on part of the code and the contribution will be attributed to you.

mtnbkr88, 2022-02-23:
I'm new to the git pull push stuff. Please tell me what to pull and how to do a pull request for this specific issue. 

In the files I changed some debug stuff to give better info for debug. The real changes are as follows:
1. Added one more field to napt_table in ip4-napt.c so it tracks src address/port to the to port on the port forwarder and the from port on the port forwarder to the dest address/port.
2. In ip4.c added function ip4_forward_local which is called from ip4_input when it's determined the incoming packet is for this local interface. ip4_forward_local calls ip_napt_forward_local.
3. In ip4_napt.c added function ip_napt_forward_local which checks napt_table to see if port forwarding info on the same network going either direction is present. If yes, the source and dest address/port are updated to forward the packet back out the same netif. If the info is not found, portmap_table is checked to see if a port forwarding rule for this local port is there. If a rule is found, complete port forwarding info is added to the napt_table and the source and dest address/port are updated to forward the packet out the same netif. If a rule is not found in portmap_table, the function returns processing back to ip4_forward_local then back to ip4_input.
4. You already saw my comment about setting IP_FORWARD, IP_NAPT and IP_FORWARD_ALLOW_TX_ON_RX_NETIF in opt.h. 

I'm happy to answer any questions you have but I will be away on a camping trip from Thursday 2/24 through Sunday 2/27 so my responses may be delayed a few days.

Thank you.

xueyunfei998, 2022-02-23:
hi @mtnbkr88 

If you just want to use the data forwarding function, you can turn on the IP_FAWARD function in menuconfig, but there is a bug in this function, and I will provide you with a patch.

xueyunfei998, 2022-02-23:
[0001-bugfix-for-Packet-forwarding-failed.txt](https://github.com/espressif/esp-idf/files/8122481/0001-bugfix-for-Packet-forwarding-failed.txt)

mtnbkr88, 2022-02-23:
As currently written in ESP-IDF 4.4.0, lwip cannot possibly forward back onto the same network (what I need) without errors. I will explain. If A is the source, B the port forwarder and C the destination, as currently written ip4_forward only modifies the destination in the packet. So a packet will be forwarded with source as A and destination as C. This is okay if the packet is going to a different network than the source. If the destination is on the same network (netif) as the source then when C receives the packet it will return directly to A. Then A will drop the packet because it does not have any known connections with C. A had a connection open to B, not C. (I verified this behavior with wireshark.) My update to lwip fixes this. If the packet is forwarded to the same network as the source, my update will have destination C return to B which will then return to A. All traffic is as expected, all connections are properly handled.

0xjakob, 2022-02-24:
> I'm new to the git pull push stuff. Please tell me what to pull and how to do a pull request for this specific issue.

You first need to fork the repository, then upload your code into a branch there, then you can create a pull request in IDF and specify the branch from your fork. See more in [how to create a pull reaquest](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request).

closed 2022-03-09 by espressif-bot