espressif/esp-idf #8453: WPA-enterprise PEAP TLS Authentication Protocol Error

jeanbie0, 2022-02-23:
## Environment

- Development Kit:  Custom board
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.3
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: WSL
- Using an IDE?: Yes: VSCode
- Power Supply:  Battery

## Problem Description
I try to connect ESP-32 to a WPA-enterprise network (peap mschapv2). It working at my office but not at my customer.

The problem is linked to the way Feedim implement the message exchange fragmentation.

According to the specs [RFC 2716](https://datatracker.ietf.org/doc/html/rfc2716) once the server send its last fragment (More fragment bit false), the client must sens a response containing a TLS message (message length present and higher than 0)

In my case the ESP32 ack the last message and didn't response.

```
No.     Time           Source                Destination           Protocol Length Info
   2984 0.000000       Cisco_e9:92:44        Espressi_6f:69:68     TLSv1.2  1071   Server Hello

Frame 2984: 1071 bytes on wire (8568 bits), 1071 bytes captured (8568 bits)
Radiotap Header v0, Length 114
802.11 radio information
IEEE 802.11 QoS Data, Flags: ......F.C
Logical-Link Control
802.1X Authentication
Extensible Authentication Protocol
    Code: Request (1)
    Id: 182
    Length: 915
    Type: Protected EAP (EAP-PEAP) (25)
    EAP-TLS Flags: 0x01
        0... .... = Length Included: False
        .0.. .... = More Fragments: False
        ..0. .... = Start: False
        .... .001 = Version: 1
    [5 EAP-TLS Fragments (4917 bytes): #2949(1002), #2954(1002), #2970(1002), #2980(1002), #2984(909)]
        [Frame: 2949, payload: 0-1001 (1002 bytes)]
        [Frame: 2954, payload: 1002-2003 (1002 bytes)]
        [Frame: 2970, payload: 2004-3005 (1002 bytes)]
        [Frame: 2980, payload: 3006-4007 (1002 bytes)]
        [Frame: 2984, payload: 4008-4916 (909 bytes)]
        [Fragment Count: 5]
        [Reassembled EAP-TLS Length: 4917]
    Transport Layer Security
        TLSv1.2 Record Layer: Handshake Protocol: Server Hello
            Content Type: Handshake (22)
            Version: TLS 1.2 (0x0303)
            Length: 89
            Handshake Protocol: Server Hello
                Handshake Type: Server Hello (2)
                Length: 85
                Version: TLS 1.2 (0x0303)
                Random: 98ae9f01c65a1a3f1e9165076eddce798169890c50a9c68408ba687ebca6d98f
                Session ID Length: 32
                Session ID: 7fbd5283438b3e1893a0addc44f390a37d174730537b2d8f42120caa9cf92cb2
                Cipher Suite: TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 (0xc030)
                Compression Method: null (0)
                Extensions Length: 13
                Extension: renegotiation_info (len=1)
                Extension: ec_point_formats (len=4)
                [JA3S Fullstring: 771,49200,65281-11]
                [JA3S: 0debd3853f330c574b05e0b6d882dc27]

No.     Time           Source                Destination           Protocol Length Info
   2996 0.024215       Espressi_6f:69:68     Cisco_e9:92:44        EAP      162    Response, Protected EAP (EAP-PEAP)

Frame 2996: 162 bytes on wire (1296 bits), 162 bytes captured (1296 bits)
Radiotap Header v0, Length 116
802.11 radio information
IEEE 802.11 Data, Flags: .......TC
Logical-Link Control
802.1X Authentication
Extensible Authentication Protocol
    Code: Response (2)
    Id: 182. <===== C’est une réponse au dernier message du serveur
    Length: 6
    Type: Protected EAP (EAP-PEAP) (25)
    EAP-TLS Flags: 0x01
        0... .... = Length Included: False. <==== Ce message est vide - length = 0
        .0.. .... = More Fragments: False
        ..0. .... = Start: False
        .... .001 = Version: 1
```

### Questions
I saw in the release notes:
> Fixed enterprise connection issue with windows radius server (1) 
> Fixed interoperability issue with Windows 2008 radius server. (1)
I didn't find any more details about this changelog in commits. Is it related ? I can't go easily to the customers office so I want to be sure that error is resolved when I'll go there.

Do you have any related issues? I didn't find anything.

Thank you

### Code to reproduce this issue

```cpp
esp_wifi_set_config
esp_wifi_sta_wpa2_ent_set_identity
esp_wifi_sta_wpa2_ent_set_username
esp_wifi_sta_wpa2_ent_set_password
esp_wifi_sta_wpa2_ent_enable
esp_wifi_start
```

negativekelvin, 2022-02-23:
Did you check https://github.com/espressif/esp-idf/issues/7422 and https://github.com/espressif/esp-idf/issues/7422

kapilkedawat, 2022-02-28:
Hi @jeanbie0 ,

ESP client sends the response after processing is done. Are you seeing the there was no response from esp32 at all? 
In that case, are you seeing any error in the logs? 

Also, are you using the latest v4.3?

jeanbie0, 2022-02-28:
Update to 4.3.2 resolve my problem. Radius server use SHA1 so it was linked to https://github.com/espressif/esp-idf/issues/7384 and https://github.com/espressif/esp-idf/issues/7422.

Thank you

sagb2015, 2022-04-25:
@jeanbie0 Thanks for the update. Feel free to open if you find any issues.

closed 2022-04-25 by sagb2015