espressif/esp-idf #8454: Modbus sending an additional bit

silardg, 2022-02-23:
## Environment

- Module or chip used:  ESP32-WROOM-32|ESP32-WROOM-32D
- IDF version
v2.12-rc3-375-g5ba18a3

## Problem Description

I'm trying to setup a modbus master on the esp32 to control a waveshare relay board

I'm suppose to send the next command:
01 03 02 00 01 79 84

`mb_param_request_t request = {.slave_addr = 1, .command = 3, .reg_start = 2, .reg_size = 1};`
But on the oscilloscope I'm getting this:
01 03 00 02 00 01 37 202

The last two is the crc so they are of course different, but still there is another byte.

My setup looks like this
```
mb_communication_info_t comm;
        comm.port     = 1;                // TODO(silardg): maybe add checking for multiple modbuses on here
        comm.mode     = MB_MODE_RTU;      // RTU mode always
        comm.baudrate = 9600;             // baudrate
        comm.parity   = MB_PARITY_NONE;   // no parity always

        // init the master
        esp_err_t err = mbc_master_init(MB_PORT_SERIAL_MASTER, &this->master_handler);
        if (err != ESP_OK) {
            error(TAG, "mb controller initialization fail");
        }

        // setup the master with the config
        err = mbc_master_setup((void *)&comm);
        if (err != ESP_OK) {
            error(TAG, "mb controller setup fail");
        }

        // Set UART pin numbers
        err = uart_set_pin(comm.port, pin_tx, pin_rx, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
        if (err != ESP_OK) {
            error(TAG, "Couldn't set uart pins");
        }

        // start it up
        err = mbc_master_start();
        if (err != ESP_OK) {
            error(TAG, "Couldnt't start mb master");
        }

        // Set driver mode to Half Duplex
        err = uart_set_mode(comm.port, UART_MODE_RS485_HALF_DUPLEX);//UART_MODE_RS485_COLLISION_DETECT);
        if (err != ESP_OK) {
            error(TAG, "Couldn't set uart to RS485 half duplex");
        }
```

Reading looks like this:
```
esp_err_t err = ESP_FAIL;

    void *temp_data = malloc(sizeof(uint8_t));

    // use the mbc api
    err           = mbc_master_send_request(request, temp_data);

    if (err == ESP_OK) {
        // convert void pointer to uint16_t
        uint8_t value = *(uint8_t *)(temp_data);
        free(temp_data);
        // set the result to the value
        *result        = value;
    } else {
        return err;
    }
    return err;
```

alisitsyn, 2022-02-24:
Hi @silardg ,

1. Which interface you are using? Did you try to check your connection with the [uart rs 485 echo example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo_rs485)
2. The correct command to read holding register from address 0x0002 is `01 03 00 02 00 01 25 CA`.
3. The data field you decode on the oscilloscope is correct `01 03 00 02 00 01`  the CRC field `37 202` is decoded incorrectly. Try to decode it again and send the diagram here.

silardg, 2022-02-24:
1. Tested, works fine.
2. 3. I'm using `err           = mbc_master_send_request(request, temp_data);` and that is generating the CRC which is 37 202, thus incorrect?
4. Correct
5. v4.3.1

alisitsyn, 2022-02-24:
@silardg ,

2. The CRC is correct 37 dec = 0x25, 202 dec =0xCA, means the stack generates the CRC correctly!

silardg, 2022-02-24:
I'm trying to communicate with this device: https://www.waveshare.com/wiki/Modbus_RTU_Relay

If you look at reading software version
01 03 02 00 C8 F0 B8

This is an example
Why is there another 0 inserted between the 03 and 02 on the ESP?

alisitsyn, 2022-02-24:
The command `01 03 00 02 00 01 25 CA` is the request from master device to read holding register from address 0x0002, The `01 03 02 00 01 79 84` command is also correct but it looks like response from slave with the register data = 0x0001 read from register 0x0002.

alisitsyn, 2022-02-24:
In order to read `Read device address` with software revision you need to configure the request as below:
```
mb_param_request_t request = {.slave_addr = 1, .command = 3, .reg_start = 0x4000, .reg_size = 1};
```
Then the request will be `01 03 40 00 00 01 91 CA` and response could be `01 03 02 00 01 79 84 `

silardg, 2022-02-24:
Got it to work, but it only responds with something when using 0 as address, thus broadcast. Will close for now. Thanks!

closed 2022-02-24 by silardg

alisitsyn, 2022-02-24:
You just need to set the slave address of your device and will be able to address this device in the command without broadcast address.