espressif/esp-idf #8456: Problem during directory creation using FatFs

simeondmr, 2022-02-23:
Hello,

I'm using this example: https://github.com/espressif/espidf/blob/master/examples/storage/ext_flash_fatfs/main/ext_flash_fatfs_example_main.c

with the external flash: W25QXX

I have only one problem, I have problems during directory creation.

for example, this code:

`FRESULT res = f_mkdir("/extflash/test2");`

return always `FR_NO_PATH`

what could be the problem?

Thanks

Simeon

igrr, 2022-02-23:
@simeondmr please note that FatFS library isn't aware of the mount path ('/extflash'). This full path can be handled by VFS (virtual filesystem) component if you call C standard library functions and POSIX directory handling functions. For more information on how VFS handles paths, see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/vfs.html#paths.
To create a directory using POSIX functions, call [mkdir](https://pubs.opengroup.org/onlinepubs/009695299/functions/mkdir.html).

If you wish to call the underlying FatFS functions directly (which is not recommend) you need to use FatFS internal drive number as a prefix, instead. In the typical case this will be '0:', so you can call `f_mkdir("0:test2")`.

simeondmr, 2022-02-24:
Hello @igrr, thanks, I solved it!
You can close the issue.

closed 2022-02-24 by simeondmr