espressif/esp-idf #8457: idf.py monitor stopped working

0xFEEDC0DE64, 2022-02-24:
Since my latest rebase to the newest master esp-idf, the log output stopped showing in idf.py

```
$ idf.py -p /dev/ttyUSB0 -b 921600 monitor -E
Executing action: monitor
Running idf_monitor in directory /home/feedc0de/fronius-wattpilot-firmware
Executing "/home/feedc0de/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/feedc0de/fronius-wattpilot-firmware/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 921600 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 3 /home/feedc0de/fronius-wattpilot-firmware/build_hw4/wattpilot_hw4.elf --encrypted -m '/home/feedc0de/.espressif/python_env/idf5.0_py3.10_env/bin/python' '/home/feedc0de/fronius-wattpilot-firmware/esp-idf/tools/idf.py' '-p' '/dev/ttyUSB0' '-b' '921600'"...
--- idf_monitor on /dev/ttyUSB0 921600 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
```

I tried screen too, and there I still the the log output from the bootloader and the app successfully.

dobairoland, 2022-02-24:
@0xFEEDC0DE64 There was a recent fix in the master branch. The baud rate set with `idf.py -b <baud> monitor` wasn't actually set for the monitor. Is your console application really uses 921600? You can check this in menuconfig. Monitor now follows the console baud rate by default. So you don't have to set from command line.

dobairoland, 2022-02-24:
Here are more information about the changes: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/tools.html

0xFEEDC0DE64, 2022-02-24:
i used the high baud rate to improve flashing times, usually I prepend a encrypted-app-flash before the monitor, can i specify different baud rates for flashing and text interface?

dobairoland, 2022-02-24:
I think all of these should work for you:
```sh
idf.py -b 921600 flash
idf.py flash -b 921600
idf.py monitor
idf.py monitor -b 115200
idf.py -b 921600 flash monitor -b 115200
```

The following will work only if https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp-console-uart-baudrate is set to 921600.
```
idf.py -b 921600 monitor
```

closed 2022-02-24 by 0xFEEDC0DE64