espressif/esp-idf #8458: Watchdog trigger missing from wifi internal stack task [IDF v4.4]

Arsanios, 2022-02-24:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|
- Kit version WroverKit
- Module or chip used:  [ESP32-WROOM-32]
- IDF version: v4.4-2-g780b9b6d78
- Build System:         idf.py
- Compiler version : xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt | PowerShell].
- Using an IDE?: VScode for coding - without ESP plugin
- Power Supply:         [USB]

## Problem Description

Hi,
I upgraded the esp-idf to the version mentioned above coming from 4.3.2 and after compiling and running my application, i found that the application got panicked due to reset watchdog missing during the WIFI start after a stop

### Expected Behavior
The WIFI should start in AP mode

### Actual Behavior

```
I (17237) wifi:wifi driver task: 3ffd65b8, prio:23, stack:3072, core=0
I (17237) wifi:wifi firmware version: 7679c42
I (17237) wifi:wifi certification version: v7.0
I (17247) wifi:config NVS flash: disabled
I (17247) wifi:config nano formating: enabled
I (17257) wifi:Init data frame dynamic rx buffer num: 8
I (17257) wifi:Init management frame dynamic rx buffer num: 8
I (17267) wifi:Init management short buffer num: 32
I (17267) wifi:Init dynamic tx buffer num: 8
I (17277) wifi:Init static rx buffer size: 1600
I (17277) wifi:Init static rx buffer num: 4
I (17277) wifi:Init dynamic rx buffer num: 8
I (17287) wifi_init: tcpip mbox: 6
I (17287) wifi_init: udp mbox: 6
I (17297) wifi_init: tcp mbox: 8
I (17297) wifi_init: tcp tx win: 8192
I (17297) wifi_init: tcp rx win: 8192
I (17307) wifi_init: tcp mss: 1436
I (17307) wifi_init: WiFi IRAM OP enabled
I (17317) wifi_init: WiFi RX IRAM OP enabled
I (17317) wifi_init: LWIP IRAM OP enabled
E (27227) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (27227) task_wdt:  - IDLE (CPU 0)
E (27227) task_wdt: Tasks currently running:
E (27227) task_wdt: CPU 0: wifi
E (27227) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x40141D8E:0x3FFBE7100x40082D6E:0x3FFBE730 0x4019885A:0x3FFD6480 0x40191B0B:0x3FFD64A0 0x401815D5:0x3FFD64C0 0x40181D23:0x3FFD64E0 0x40180620:0x3FFD6500 0x4009B425:0x3FFD6520
0x40141d8e: task_wdt_isr at C:/Projects/Haier/esp/esp-idf/components/esp_system/task_wdt.c:183 (discriminator 3)

0x40082d6e: _xt_lowint1 at C:/Projects/Haier/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x4019885a: hal_init at ??:?

0x40191b0b: ic_enable at ??:?

0x401815d5: wifi_hw_start at ??:?

0x40181d23: wifi_start_process at ??:?

0x40180620: ieee80211_ioctl_process at ??:?

0x4009b425: ppTask at ??:?
```

Thanks

MaxwellAlan, 2022-03-01:
Hi @Arsanios 
According your description, we can't reproduce this issue and it's difficult for us to debug, can you upload a demo code and sdkconfig to reproduce this issue.
Thanks.

MaxwellAlan, 2022-03-01:
Hi @Arsanios 
Check hal_init in wifi lib, when MAC works in abnormally may triger task watchdog. Can you provide more information abort what have done before wifi stop and start.

Arsanios, 2022-03-01:
> Hi @Arsanios Check hal_init in wifi lib, when MAC works in abnormally may triger task watchdog. Can you provide more information abort what have done before wifi stop and start.

The procedure is to call this function (StartAP) then at a certain time we should stop the wifi and then recall the StartAP again.
The problem happens when the StartAP is called for the second time. If i comment the first call the second one doesn't make the watchdog problem.

Here is the code of StartAP using NETIF

```
void WIFI__StartAP(char* pc_SSID, char* pc_password)
{
  size_t u8_ipv4_dec[4];

  memset((uint8_t*)gx_WIFI_Manager.ap_ssid, 0, 64);
  memset((uint8_t*)gx_WIFI_Manager.ap_pwd, 0, 64);
  
  strcpy(gx_WIFI_Manager.ap_ssid, pc_SSID);
  strcpy(gx_WIFI_Manager.ap_pwd, pc_password);
  

  // set default STA configuration to enable scan of networks during pairing mode
  memset((uint8_t*)gx_WIFI_Manager.sta_ssid, 0, 64);
  memset((uint8_t*)gx_WIFI_Manager.sta_pwd, 0, 64);
  strcpy((char*)gx_WIFI_Manager.wifi_config.sta.ssid, gx_WIFI_Manager.sta_ssid);
  strcpy((char*)gx_WIFI_Manager.wifi_config.sta.password, gx_WIFI_Manager.sta_pwd);

  //  WIFI_FAST_SCAN = 0,                   /**< Do fast scan, scan will end after find SSID match AP */
  //  WIFI_ALL_CHANNEL_SCAN,                /**< All channel scan, scan will end after scan all the channel */
  gx_WIFI_Manager.wifi_config.sta.scan_method = WIFI_ALL_CHANNEL_SCAN;

  //  WIFI_CONNECT_AP_BY_SIGNAL = 0,        /**< Sort match AP in scan list by RSSI */
  //  WIFI_CONNECT_AP_BY_SECURITY,          /**< Sort match AP in scan list by security mode */
  gx_WIFI_Manager.wifi_config.sta.sort_method = WIFI_CONNECT_AP_BY_SIGNAL;
  gx_WIFI_Manager.wifi_config.sta.threshold.rssi = -127;

  //  WIFI_AUTH_OPEN = 0,         /**< authenticate mode : open */
  //  WIFI_AUTH_WEP,              /**< authenticate mode : WEP */
  //  WIFI_AUTH_WPA_PSK,          /**< authenticate mode : WPA_PSK */
  //  WIFI_AUTH_WPA2_PSK,         /**< authenticate mode : WPA2_PSK */
  //  WIFI_AUTH_WPA_WPA2_PSK,     /**< authenticate mode : WPA_WPA2_PSK */
  //  WIFI_AUTH_WPA2_ENTERPRISE,  /**< authenticate mode : WPA2_ENTERPRISE */
  gx_WIFI_Manager.wifi_config.sta.threshold.authmode = WIFI_AUTH_OPEN;

  // set SSID to create
  strcpy((char*)gx_WIFI_Manager.wifi_config.ap.ssid, gx_WIFI_Manager.ap_ssid);
  // set password if required
  strcpy((char*)gx_WIFI_Manager.wifi_config.ap.password, gx_WIFI_Manager.ap_pwd);

  // calculate ssid length
  gx_WIFI_Manager.wifi_config.ap.ssid_len = strlen((char*)gx_WIFI_Manager.wifi_config.ap.ssid);

  // set configured max connections
  gx_WIFI_Manager.wifi_config.ap.max_connection = gx_NVS_ATCmd_CWSAP_Params.max_conn;

  gx_WIFI_Manager.wifi_config.ap.channel = gx_NVS_ATCmd_CWSAP_Params.channel;

  // auto configure security mode if password is set
  if ((strlen((char*)gx_WIFI_Manager.wifi_config.ap.password) == 0) ||
      (gx_NVS_ATCmd_CWSAP_Params.ecn == 0))
  {
    gx_WIFI_Manager.wifi_config.ap.authmode = WIFI_AUTH_OPEN;
  }
  else
  {
    gx_WIFI_Manager.wifi_config.ap.authmode = WIFI_AUTH_WPA_WPA2_PSK;
  }

  // clear flag to wait for AP creation
  gx_WIFI_Manager.ap_set_on = false;

  if(ap_netif_handle == NULL)
  {
    ap_netif_handle = esp_netif_create_default_wifi_ap();
    if(ap_netif_handle == NULL)
    {
      ESP_LOGE(__func__, "esp_netif_create_default_wifi_ap failed.");
    }
  }

  if(instance_any_id == NULL)
  {
    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        ESP_EVENT_ANY_ID,
                                                        &netif_event_handler,
                                                        NULL,
                                                        &instance_any_id));
  }

  ESP_ERROR_CHECK_WITHOUT_ABORT(esp_netif_dhcps_stop(ap_netif_handle));
  // init wifi_ap_tcpip_adapter_ip_info
  memset((uint8_t *)&gx_WIFI_Manager.wifi_ap_esp_netif_ip_info, 0, sizeof(esp_netif_ip_info_t));

  Utilities__IPStrToIPV4Dec(AP_IP, u8_ipv4_dec);
  IP4_ADDR(&gx_WIFI_Manager.wifi_ap_esp_netif_ip_info.ip, u8_ipv4_dec[0], u8_ipv4_dec[1], u8_ipv4_dec[2], u8_ipv4_dec[3]);

  Utilities__IPStrToIPV4Dec(AP_GATEWAY, u8_ipv4_dec);
  IP4_ADDR(&gx_WIFI_Manager.wifi_ap_esp_netif_ip_info.gw, u8_ipv4_dec[0], u8_ipv4_dec[1], u8_ipv4_dec[2], u8_ipv4_dec[3]);

  Utilities__IPStrToIPV4Dec(AP_NETMASK, u8_ipv4_dec);
  IP4_ADDR(&gx_WIFI_Manager.wifi_ap_esp_netif_ip_info.netmask, u8_ipv4_dec[0], u8_ipv4_dec[1], u8_ipv4_dec[2], u8_ipv4_dec[3]);

  if(ap_netif_handle != NULL)
  {
    // confirm TCP/IP settings
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_netif_set_ip_info(ap_netif_handle, &gx_WIFI_Manager.wifi_ap_esp_netif_ip_info));
    // start AP DHCP
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_netif_dhcps_start(ap_netif_handle));
  }

  wifi_init_config_t cfg_ap = WIFI_INIT_CONFIG_DEFAULT();

  printf( "%s, esp_wifi_init(&cfg_ap)\n",__func__);

  ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_init(&cfg_ap));

  ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_mode(WIFI_MODE_APSTA));
  ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_config(ESP_IF_WIFI_STA, &gx_WIFI_Manager.wifi_config));
  ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_config(ESP_IF_WIFI_AP, &gx_WIFI_Manager.wifi_config));
  ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_storage(WIFI_STORAGE_RAM));
  ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_start());
}
```

Here is also the sdkconfig in txt format
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8160761/sdkconfig.txt)

Arsanios, 2022-03-01:
In addition, here you can find the wifi stop function:

```
void WIFI__Stop(void)
{
  NetMonitor__ResetAllTimeouts();

  memset((uint8_t*)gx_WIFI_Manager.ap_ssid, 0, 64);
  memset((uint8_t*)gx_WIFI_Manager.ap_pwd, 0, 64);
  memset((uint8_t*)gx_WIFI_Manager.sta_ssid, 0, 64);
  memset((uint8_t*)gx_WIFI_Manager.sta_pwd, 0, 64);
  
  HTTP_Manager__StopServer();

  if (gx_WIFI_Manager.sta_scan_status == true)
  {
	ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_scan_stop());
	gx_WIFI_Manager.sta_scan_status = false;
  }

  ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_disconnect());
  ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_stop());

  if(ap_netif_handle != NULL)
  {
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_netif_dhcps_stop(ap_netif_handle));
  }

  if(instance_got_ip != NULL)
  {
    ESP_ERROR_CHECK(esp_event_handler_instance_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, instance_got_ip));
    instance_got_ip = NULL;
  }

  if(instance_any_id != NULL)
  {
    ESP_ERROR_CHECK(esp_event_handler_instance_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, instance_any_id));
    instance_any_id = NULL;
  }

  if(ap_netif_handle != NULL){
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_clear_default_wifi_driver_and_handlers((void*)ap_netif_handle));
  }

  ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_deinit());

  
}
```

MaxwellAlan, 2022-03-02:
> The procedure is to call this function (StartAP) then at a certain time

Hi @Arsanios 
Try the main produce of your wifi stop and wifi start, works well then. Can you just try only using your wifi start and stop without at a certain time to check the stop and start work well?
And can you provide more details about a certain time? What really happen at that time?
From your sdkconfig, the app also using Bluetooth right?

Alvin1Zhang, 2022-04-15:
@Arsanios Thanks for reporting, would you please help share if any updates for the issue? Thanks.

Arsanios, 2022-04-19:
@Alvin1Zhang I found a workaround, which is to not deinit the netif when we are not using it. It is working. Thanks.

closed 2022-04-20 by Alvin1Zhang