espressif/esp-idf #8459: Watchdog trigger missing from wifi internal stack task [IDF v4.4]

Arsanios, 2022-02-24:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|
- Kit version WroverKit
- Module or chip used:  [ESP32-WROOM-32]
- IDF version: v4.4-2-g780b9b6d78
- Build System:         idf.py
- Compiler version : xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt | PowerShell].
- Using an IDE?: VScode for coding - without ESP plugin
- Power Supply:         [USB]

## Problem Description

Hi,
I upgraded the esp-idf to the version mentioned above coming from 4.3.2 and after compiling and running my application, i found that the application got panicked due to reset watchdog missing during the WIFI start after a stop

### Expected Behavior
The WIFI should start in AP mode

### Actual Behavior

```
I (17237) wifi:wifi driver task: 3ffd65b8, prio:23, stack:3072, core=0
I (17237) wifi:wifi firmware version: 7679c42
I (17237) wifi:wifi certification version: v7.0
I (17247) wifi:config NVS flash: disabled
I (17247) wifi:config nano formating: enabled
I (17257) wifi:Init data frame dynamic rx buffer num: 8
I (17257) wifi:Init management frame dynamic rx buffer num: 8
I (17267) wifi:Init management short buffer num: 32
I (17267) wifi:Init dynamic tx buffer num: 8
I (17277) wifi:Init static rx buffer size: 1600
I (17277) wifi:Init static rx buffer num: 4
I (17277) wifi:Init dynamic rx buffer num: 8
I (17287) wifi_init: tcpip mbox: 6
I (17287) wifi_init: udp mbox: 6
I (17297) wifi_init: tcp mbox: 8
I (17297) wifi_init: tcp tx win: 8192
I (17297) wifi_init: tcp rx win: 8192
I (17307) wifi_init: tcp mss: 1436
I (17307) wifi_init: WiFi IRAM OP enabled
I (17317) wifi_init: WiFi RX IRAM OP enabled
I (17317) wifi_init: LWIP IRAM OP enabled
E (27227) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (27227) task_wdt:  - IDLE (CPU 0)
E (27227) task_wdt: Tasks currently running:
E (27227) task_wdt: CPU 0: wifi
E (27227) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x40141D8E:0x3FFBE7100x40082D6E:0x3FFBE730 0x4019885A:0x3FFD6480 0x40191B0B:0x3FFD64A0 0x401815D5:0x3FFD64C0 0x40181D23:0x3FFD64E0 0x40180620:0x3FFD6500 0x4009B425:0x3FFD6520
0x40141d8e: task_wdt_isr at C:/Projects/Haier/esp/esp-idf/components/esp_system/task_wdt.c:183 (discriminator 3)

0x40082d6e: _xt_lowint1 at C:/Projects/Haier/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x4019885a: hal_init at ??:?

0x40191b0b: ic_enable at ??:?

0x401815d5: wifi_hw_start at ??:?

0x40181d23: wifi_start_process at ??:?

0x40180620: ieee80211_ioctl_process at ??:?

0x4009b425: ppTask at ??:?
```

Thanks

KaeLL, 2022-02-24:
Posted twice #8458

Arsanios, 2022-02-24:
Ok thanks, i will close this and let the other one open

closed 2022-02-24 by Arsanios