espressif/esp-idf #8461: twai_alert_and_recovery example unclear

franz-ms-muc, 2022-02-24:
## Problem Description

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/twai/twai_alert_and_recovery

it perfectly does the Recovery, but after it was done it exit. 

wouldnt it be better to show how to CONTINUE the Communication ?

in my case a additional  
`ESP_ERROR_CHECK(twai_start());`
`twai_reconfigure_alerts(TWAI_ALERT_ABOVE_ERR_WARN | TWAI_ALERT_ERR_PASS | TWAI_ALERT_BUS_OFF, NULL);`

was needed, but is it the really right way to do so ? 

also some specs require the following behaviour: 

Busoff, wait 25ms, Recover
Busoff, wait 25ms, Recover
Busoff, wait 25ms, Recover
Busoff, wait 25ms, Recover
Busoff, wait 200ms, Recover
Busoff, wait 25ms, Recover

Thanks, 
Franz

Dazza0, 2022-02-28:
@franz-ms-muc Thanks for the suggestion. I'll update the alert and recovery example to demonstrate communication post recovery.

Regarding the spec, I'm not sure what specification that is, and where the fixed 25ms comes from. The CAN2.0 spec is quite clear about bus off recovery as quoted below:

> An node which is ’bus off’ is permitted to become ’error active’ (no longer ’bus off’) with its error counters both set to 0 after 128  occurrence of 11 consecutive ’recessive’ bits have been monitored on the bus.

franz-ms-muc, 2022-02-28:
That spec is from a Bosch Sensor.

Holen Sie sich Outlook für Android<https://aka.ms/AAb9ysg>

[https://www.meisterschulen-mchn.de/images/meisterschulen/allgemein/4-logos/logofc.jpg]

Meisterschulen am Ostbahnhof
Mühldorfstr. 6
81671 München

Tel.: 089 416002 - 0
Fax: 089 416002 - 111

Internet: www.meisterschulen-münchen.de<http://www.meisterschulen-mchn.de>

Dazza0, 2022-02-28:
@franz-ms-muc Do you happen to have a link to the spec in question?

franz-ms-muc, 2022-02-28:
re64819-b_2019-03.pdf
re64819-b_2019-03.pdf
https://www.boschrexroth.com/documents/12605/25209273/re64819-b_2019-03.pdf/7a3956db-a683-5cde-5438-63abbe9b2bd5?version=1.0&download=true

they tell about a Random time to avoid deadlock. 

the Spec in Question i will search and send you once i find.

Alvin1Zhang, 2022-08-04:
Thanks for reporting, any further updates for the issue? Thanks.

diplfranzhoepfinger, 2022-08-31:
> @franz-ms-muc Do you happen to have a link to the spec in question?

can i send you by E-Mail ?

iggardo, 2023-11-04:
> in my case a additional `ESP_ERROR_CHECK(twai_start());` `twai_reconfigure_alerts(TWAI_ALERT_ABOVE_ERR_WARN | TWAI_ALERT_ERR_PASS | TWAI_ALERT_BUS_OFF, NULL);`
> 
> was needed

You save my day.
Thanks a lot!

closed 2025-08-19 by espressif-bot