espressif/esp-idf #8463: not found function esp_wifi_get_pmf_config_internal in esp-idf

greentd, 2022-02-25:
not found function esp_wifi_get_pmf_config_internal in esp-idf 

esp_hostap.c

#ifdef CONFIG_IEEE80211W

    esp_wifi_get_pmf_config_internal(&pmf_cfg, WIFI_IF_AP);

    if (pmf_cfg.required) {
	pairwise_cipher = WIFI_CIPHER_TYPE_CCMP;
    }
#endif /* CONFIG_IEEE80211W */

MaxwellAlan, 2022-02-25:
Hi @greentd 
Can you share more detail about idf commit and which example or uploading your demo code to reproduce this issue.
And maybe you should using this command : git submodule update -f to update your submodule components.
Looking forward your reply.
Thanks.

AxelLin, 2022-02-25:
@greentd 
It's in closed source library libnet80211.a, implementation source code is not available.

closed 2022-03-17 by Alvin1Zhang