espressif/esp-idf #8464: Installing Python environment and packages error   ,   virtualenv.py: error: no such option: --seeder

cgrrty, 2022-02-25:
`pi@ubuntu:~/workspace/esp/esp-idf-v4.4$ ./install.sh 
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.6.9
"python3" has been detected
Installing ESP-IDF tools
Selected targets are: esp32s3, esp32, esp32c3, esp32s2
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2021r2-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2021r2-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2021r2-8.4.0 (already installed)
Skipping riscv32-esp-elf@esp-2021r2-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20211220 (already installed)
Installing Python environment and packages
Creating a new Python environment in /home/pi/.espressif/python_env/idf4.4_py3.6_env
Usage: virtualenv.py [OPTIONS] DEST_DIR

virtualenv.py: error: no such option: --seeder
Traceback (most recent call last):
  File "/home/pi/workspace/esp/esp-idf-v4.4/tools/idf_tools.py", line 1828, in <module>
    main(sys.argv[1:])
  File "/home/pi/workspace/esp/esp-idf-v4.4/tools/idf_tools.py", line 1824, in main
    action_func(args)
  File "/home/pi/workspace/esp/esp-idf-v4.4/tools/idf_tools.py", line 1525, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'virtualenv', '--seeder', 'pip', '/home/pi/.espressif/python_env/idf4.4_py3.6_env']' returned non-zero exit status 2.
`

cgrrty, 2022-02-25:
```
pi@ubuntu:~/workspace/esp/esp-idf-v4.4$ 
pi@ubuntu:~/workspace/esp/esp-idf-v4.4$ /usr/bin/python -m virtualenv --version
virtualenv 20.13.2 from /usr/local/lib/python2.7/dist-packages/virtualenv/__init__.pyc
pi@ubuntu:~/workspace/esp/esp-idf-v4.4$ pacman -Qs virtualenv
XIO:  fatal IO error 22 (Invalid argument) on X server ":0"
      after 1136 requests (1136 known processed) with 0 events remaining.
pi@ubuntu:~/workspace/esp/esp-idf-v4.4$ /usr/bin/python -m pip freeze
asn1crypto==0.24.0
attrs==17.4.0
Automat==0.6.0
chardet==3.0.4
click==6.7
colorama==0.3.7
configparser==3.5.0
constantly==15.1.0
contextlib2==0.6.0.post1
cryptography==2.1.4
distlib==0.3.4
enum34==1.1.6
filelock==3.2.1
future==0.15.2
hyperlink==17.3.1
idna==2.6
importlib-metadata==2.1.3
importlib-resources==3.3.1
incremental==16.10.1
ipaddress==1.0.17
keyring==10.6.0
keyrings.alt==3.0
PAM==0.4.2
pathlib2==2.3.7.post1
platformdirs==2.0.2
pyasn1==0.4.2
pyasn1-modules==0.2.1
pycrypto==2.6.1
pygobject==3.26.1
pykerberos==1.1.14
pyOpenSSL==17.5.0
pyparted==3.11.1
pyroute2==0.4.21
pyserial==3.4
python-apt==1.6.5+ubuntu0.5
python-debian==0.1.32
pyxdg==0.25
PyYAML==3.12
scandir==1.10.0
SecretStorage==2.3.1
service-identity==16.0.0
singledispatch==3.7.0
six==1.11.0
Twisted==17.9.0
typing==3.10.0.0
virtualenv==20.13.2
wifibroadcast==21.3.127.7722
zipp==1.2.0
zope.interface==4.3.2
pi@ubuntu:~/workspace/esp/esp-idf-v4.4$ 

```

cgrrty, 2022-02-25:
sudo pip3 install virtualenv --upgrade

0xjakob, 2022-02-25:
> `sudo pip3 install virtualenv --upgrade`

@cgrrty Did this help?

dobairoland, 2022-02-25:
```
virtualenv 20.13.2 from /usr/local/lib/python2.7/dist-packages/virtualenv/__init__.pyc
```
Your virtualenv is broken. It seems to be linked to Python 2.7.

Please remove all versions of virtualenv from your computer. Don't use `sudo pip3` for installing packages. It will make things worse.

After you've removed them, the ESP-IDF install script will be able to install a suitable virtualenv locally (without `sudo`).

```
pip3 uninstall virtualenv
sudo pip3 uninstall virtualenv
```
You might need to run the above commands several times if you have multiple versions.
If you have installed virtualenv with your package manager (eg. `apt-get`) then remove it as well because `sudo pip3` broke it as well.

dobairoland, 2022-02-25:
I see you opened https://github.com/espressif/esp-idf/issues/8465. Does that mean you already solved this issue?

maBarabas, 2022-03-03:
`python -m pip install --upgrade virtualenv` fixed this for me.

```
# before
$ python -m virtualenv --version
16.7.10

# after
$ python -m virtualenv --version
virtualenv 20.13.2 from /home/barabas/.local/lib/python3.8/site-packages/virtualenv/__init__.py
```

dobairoland, 2022-03-03:
I assume this is no longer an issue.

closed 2022-03-03 by dobairoland