espressif/esp-idf #8465: PermissionError: [Errno 13] Permission denied:

cgrrty, 2022-02-25:
```
pi@ubuntu:~/workspace/esp/hello_world$ idf.py build-system-targets esp32
Executing action: build-system-targets
Traceback (most recent call last):
  File "/home/pi/workspace/esp/esp-idf-v4.4/tools/idf.py", line 812, in <module>
    main()
  File "/home/pi/workspace/esp/esp-idf-v4.4/tools/idf.py", line 730, in main
    cli(sys.argv[1:], prog_name=PROG, complete_var='_IDF.PY_COMPLETE')
  File "/home/pi/.espressif/python_env/idf4.4_py3.6_env/lib/python3.6/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.espressif/python_env/idf4.4_py3.6_env/lib/python3.6/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/home/pi/.espressif/python_env/idf4.4_py3.6_env/lib/python3.6/site-packages/click/core.py", line 1691, in invoke
    return _process_result(rv)
  File "/home/pi/.espressif/python_env/idf4.4_py3.6_env/lib/python3.6/site-packages/click/core.py", line 1628, in _process_result
    value = ctx.invoke(self._result_callback, value, **ctx.params)
  File "/home/pi/.espressif/python_env/idf4.4_py3.6_env/lib/python3.6/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/pi/workspace/esp/esp-idf-v4.4/tools/idf.py", line 636, in execute_tasks
    task(ctx, global_args, task.action_args)
  File "/home/pi/workspace/esp/esp-idf-v4.4/tools/idf.py", line 217, in __call__
    self.callback(self.name, context, global_args, **action_args)
  File "/home/pi/workspace/esp/esp-idf-v4.4/tools/idf_py_actions/core_ext.py", line 48, in list_build_system_targets
    build_target('help', ctx, args)
  File "/home/pi/workspace/esp/esp-idf-v4.4/tools/idf_py_actions/core_ext.py", line 28, in build_target
    ensure_build_directory(args, ctx.info_name)
  File "/home/pi/workspace/esp/esp-idf-v4.4/tools/idf_py_actions/tools.py", line 191, in ensure_build_directory
    os.makedirs(build_dir)
  File "/usr/lib/python3.6/os.py", line 220, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/home/pi/workspace/esp/hello_world/build'
```

Alvin1Zhang, 2022-02-25:
@cgrrty Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

dobairoland, 2022-02-25:
This usually happens if you previously run `idf.py` with `sudo`. You don't have access right to the build directory. Please try to remove it manually, probably with `sudo rm -r build` and after that don't use `sudo` to run `idf.py` commands.

CNSNGopenSourceProjects, 2022-03-26:
I had a similar problem when I tried to create a custom partition.
When if fails, this errno13 shows up addressing the workplace without any reference to missing partition table file.

It looks like the error handling is showing a generic message instead of report properly the missing file.

dobairoland, 2022-03-28:
@CNSNGopenSourceProjects I think your issue is a different one. Please open a new issue for it and provide more information. The error message could be the same because it comes from the operating system and not ESP-IDF.

I think the current issue can be closed. Nothing to be done for it.

closed 2022-03-28 by dobairoland

CNSNGopenSourceProjects, 2022-03-28:
Hi,

I discovered the problem.
IDF dislikes UTF-8 characters on path names.
I changed it and the erro disappeared.
Thank you.