espressif/esp-idf #8466: espBlufi 扫描不到wifi列表

ouwen0116, 2022-02-25:
在esp32S3设备没有连到wifi时，用espBlufi配网APP，连上蓝牙后点扫描，没有反应，也显示不了wifi列表。log提示wifi:sta_scan: STA is connecting, scan are not allowed!
但是在esp32S3连到wifi时，这时候再用espBlufi配网app，连上蓝牙后点扫描就会显示wifi列表。
请问这个问题该怎么解决？

ouwen0116, 2022-02-25:
[image]
断开网络后，再点扫描无法显示wifi列表。

ouwen0116, 2022-02-28:
@Alvin1Zhang

CYCHEN97, 2022-02-28:
@0xjakob
@esp-lis
@ESP-YJM
@negativekelvin
@alisitsyn

jack0c, 2022-05-23:
@ouwen0116 这种情况是ESP32S3正在连接AP，此时会拒绝扫描命令。要等到连接成功或连接失败并放弃连接以后，才能再接收扫描命令。一般可以做以下事情：1. 给出提醒，设备忙；2. 通过后台等待的方式，发起新一轮的扫描

jack0c, 2022-05-23:
#8436

Alvin1Zhang, 2023-04-25:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/0c8272b2e0d9c50eb2a573f1936d28fa53617c96, feel free to reopen. Thanks.

closed 2023-04-25 by Alvin1Zhang