espressif/esp-idf #8468: Regression in 4.4 Release: SIG_SUPPLICANT_DEL_TASK race condition in esp_common.c (IDFGH-5688) #7409

cmsedore, 2022-02-25:
This bug seems to have reappeared in 4.4 Release.

components/wpa_supplicant/esp_supplicant/src/esp_common.c

```
int esp_supplicant_post_evt(uint32_t evt_id, uint32_t data)
{
	supplicant_event_t *evt = os_zalloc(sizeof(supplicant_event_t));
	if (evt == NULL) {
		return -1;
	}
	evt->id = evt_id;
	evt->data = data;

	SUPPLICANT_API_LOCK();
	if (xQueueSend(s_supplicant_evt_queue, &evt, 10 / portTICK_PERIOD_MS ) != pdPASS) {
		SUPPLICANT_API_UNLOCK();
		os_free(evt);
		return -1;
	}
	SUPPLICANT_API_UNLOCK();
	return 0;
}
```

_Originally posted by @cmsedore in https://github.com/espressif/esp-idf/issues/7409#issuecomment-1051337832_

AxelLin, 2022-02-26:
v4.3 branch also needs fix.

AxelLin, 2022-03-31:
@kapilkedawat 

The fix in master was commited 3 months ago.
But the backport to stable branches is so slow?

AxelLin, 2022-04-21:
v4.4.1 does not include the fix.

Alvin1Zhang, 2022-04-26:
Fix on release/4.3 is available https://github.com/espressif/esp-idf/commit/3477c342e29e6697b71ed5a7a0eba14fafbd35fc, and will be included in the next bug fix release of release/4.3. 

Fix on release/4.4 is now being synchronised to GitHub, and will be included in the next bug fix release of release/4.4. Thanks.

Alvin1Zhang, 2022-05-06:
Fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/36823b45ccbf6bafb76e59d0986efbcdf7078f40.

Will update once the next bug fix releases of both release/4.4 and release/4.3 happen, feel free to reopen.

closed 2022-05-06 by Alvin1Zhang