espressif/esp-idf #8469: promiscuous mode doesn't capture own control packets

moefear85, 2022-02-25:
** Description of the Problem **

I noticed that in promiscuous mode, even when setting the promiscuous filters as such:

> wifi_promiscuous_filter_t wifi_filter={.filter_mask=WIFI_PROMIS_FILTER_MASK_ALL};
> wifi_promiscuous_filter_t wifi_ctrl_filter={.filter_mask=WIFI_PROMIS_CTRL_FILTER_MASK_ALL};

Assume we have 3 ESPs, called **A**, **B**, and **C**. Then esp **A** can capture various control and management packets from **other** nodes but not **itself**. In other words, if the promiscuous esp **A** sends data to another esp **B** using esp_wifi_80211_tx, then when the other esp **B** responds, the promiscuous esp **A** won't show any ACK packets. If I use a 3rd esp **C** in promiscuous mode, that esp **C** can capture all data and control/management packets between esp **A** and **B**, including the ACK packets, but esp **C** still wouldn't be able to capture any ACK or other control packets that are sent back to esp **C** if esp **C** initiated the communication using esp_wifi_80211_tx.

IE, Any promiscuous esp can capture control/ACK packets, even if these are intended to the promiscuous esp and set the promiscuous esp's mac in the destination field, but not if if those control/ACKs are natural responses to any transmissions by the promiscuous esp using esp_wifi_80211_tx. 

**Describtion of the desired solution**

It would be nice if any esp sending data using esp_wifi_80211_tx, is still able to recieve any control/ACK frames sent back to it as a response, otherwise it breaks/complicates the application I am trying to design. I want reliable communication when using esp_wifi_80211_tx, so I need to know if ACKs have been recieved or not. Without them, I have to encode extra ACK frames as separate packets, which adds 4x the overhead, thus greatly reducing speed and unnecessarily complicating the code. Additionally, esp_wifi_80211_tx is non-blocking, meaning its return values are not related to the transmission itself. It would be nice if additionally/instead the esp_wifi_80211_tx returns whether the packet was successfully delivered or not (after a configurable timeout, which can be set to 0 to force non-blocking behavior).

This is related to an additional problem/feature (depending on perspective), that esp_wifi_80211_tx will attempt to keep resending the packet until an ACK is recieved or a timeout occurs. This feature is nice, but restrictive unless the coder can configure how many retries, and whether to retry at all. Sometimes the coder wants to send a packet to a specific address, but does not care if that address is currently online or not, and doesn't want the esp to slow down with retries because of that. Currently, one is forced to accept a severe slowdown in the offline case.

MaxwellAlan, 2022-03-30:
hi @moefear85 
Thanks for your report. We will fix the sniffer issue when we can go back to office, since shanghai is suffering COVID-19 now.

Thanks.

moefear85, 2022-03-30:
ok no problem. btw i'm sending everything to broadcast currently as an workaround. But i realized a different issue. Under moderate streaming load, if a packet fails to arrive, i enter a resend loop on a set of packets without any delay in between, and always check the return value. despite always reporting esp-ok, it turns out the first packet in the resend burst is always skipped, and only the rest get send, which causes my algorithm to repeat the loop indefinately, since skip keeps happening on the first resent packet. it skips it silently. on other sporadic occasions, it suddenly stops sending alltogether without any error message. it often tesumes transmission again sporadically if i wait a while.

MaxwellAlan, 2022-03-31:
@moefear85 
Will also look into this strange issue when we get back to office.

jack0c, 2022-05-20:
@MaxwellAlan we can provide a patch to test the issue.

moefear85, 2022-05-21:
Update on the skipping issue:

The moment I attach an antenna to the sender, the problem disappears. I was getting pretty frustrated thinking I had a bug somewhere. By chance I connected an antenna midway while the problem was happening, and it suddenly disappeared. The first packet is then always caught. I'm not using any power save features nor reducing the tx power in anyway. Quite the opposite I have it set at max. This shocked me. So it seems an electrical/physical (not software) issue, in that after a pause in transmissions, the next packet to get sent will somehow get sent at significantly reduced power or gets garbled or dropped, while all other packets in the burst will arrive. But if an antenna is connected, then even after a pause, the first packet still gets through. This might also explain why overnight all of a sudden transmissions would stop happening, since my app is coded such that if any packet is missed, it waits then retries in bursts. So if the first packet of every burst is consistently not recieved, then the whole stream remains stalled. But I wasn't using antennas. I haven't been able to test that specific application again with antennas. I hope to soon and I'll post results.

MaxwellAlan, 2022-05-23:
Hi @moefear85 
Could you please replace this patch lib [gh_8469_38065a73.tar.gz](https://github.com/espressif/esp-idf/files/8753138/gh_8469_38065a73.tar.gz) to your $IDF_PATH/components/esp_wifi/lib/esp32 to retest the sniffer ack issue, and check the firmware version is 38065a7 in boot log.
Thanks.

moefear85, 2022-05-26:
@MaxwellAlan
Ok I'll give it a try, I just downloaded it. But I need some time. I just detected a data corruption issue in usb-cdc that exists both in release/v4.4 & latest master. I'll open an issue later, I'm currently trying to localize the source & fix it, because this is important to me. I'm also currently setting up a v4.3.1 environment, to test if the issue was present there too, back when I first started using usb features, because I never detected it back then.

moefear85, 2022-06-07:
yes it works thx!

closed 2022-07-10 by moefear85