espressif/esp-idf #8470: ./install.sh fails to install, error "Can not create virtual environment"

kylemallory, 2022-02-25:
Installing latest clone of 'esp-idf', running ./install.sh fails with error:

```
Extracting /home/kylemallory/.espressif/dist/openocd-esp32-linux-amd64-0.11.0-esp32-20211220.tar.gz to /home/kylemallory/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220
Installing Python environment and packages
ERROR: This script was called from a virtual environment, can not create a virtual environment again
```

This is on a fairly fresh install of Pop_OS (basically Ubuntu).  Looking into the code for ./tools/idf_tools.py, at approx line 1577 (inside "action_install_python_env()") the test to determine if a virtual-environment is installed is bad logic.

Changed the "or" from:

`is_virtualenv = hasattr(sys, 'real_prefix') or (hasattr(sys, 'base_prefix') and sys.base_prefix != sys.prefix)`

to an "and":

`is_virtualenv = hasattr(sys, 'real_prefix') and (hasattr(sys, 'base_prefix') and sys.base_prefix != sys.prefix)`

fixes the problem.

dobairoland, 2022-02-27:
Hi @kylemallory,

```
ERROR: This script was called from a virtual environment, can not create a virtual environment again
```
The error means that you've opened the terminal and run `. export.sh` which activated the virtual environment. And now you have run `install.sh` and it cannot install a new virtual environment if one is already activated.

The real solution to this problem is to open a new terminal window (clean shell) and run the `install.sh` there.

Please note that if you've added the `. export.sh` into one of the scripts processed by every shell (for example `.bashrc`) then you can get the same error. The solution for it is the same.

0xjakob, 2022-02-28:
@kylemallory Do you think it would be helpful do have a note about this problem in the [IDF setup instructions]( https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#step-3-set-up-the-tools)? Or did you look at any other place?

closed 2022-03-03 by dobairoland

QGB, 2022-09-26:
ERROR: This script was called from a virtual environment, can not create a virtual environment again

jcwren, 2022-11-12:
> Hi @kylemallory,
>
> ```
> ERROR: This script was called from a virtual environment, can not create a virtual environment again
> ```
>
> The error means that you've opened the terminal and run `. export.sh` [...]

I don't believe this is entirely correct. I don't speak Python, but the script appears to find ANY virtual environment that's in the path, regardless if it's esp-idf or not. I ran into this exact problem, had a completely clean environment as far as esp-idf goes, and it blew it's brains out as above due to finding a virtual environment for a statistics package that someone at work wrote that was in my path.

AshUK, 2022-11-21:
Removing the following from my .bash_profile fixed this on macOS 13

```
export IDF_PYTHON_ENV_PATH=$HOME/.espressif/python_env/idf4.3_py3.9_env

export PATH=$HOME/.espressif/python_env/idf4.4_py3.9_env/bin:$PATH
```

AkashJohnSubash, 2023-03-21:
> I don't believe this is entirely correct. I don't speak Python, but the script appears to find ANY virtual environment that's in the path, regardless if it's esp-idf or not. I ran into this exact [...]

Faced the same issue as @jcwren . Had to remove another unrelated virtual environment from my `zsh` profile before I got  `install.sh` to work. Steps are below:

1. `nano ~/.zshrc`
2.  removed exported virtual environment paths from $PATH
3.  start fresh terminal , re-run `install.sh`

brant-ci, 2023-03-27:
I'm seeing this issue under Windows. Fresh git copy of everything, cleaned .espressif folder, new powershell window, new cmd.exe window, rebooted the machine, all suggested solutions do not work.

The install script is clearly picking up *any* python environment in PATH. I believe I'm getting this error because platformio installs its own python virtual environment. I cannot remove this environment, nor should I have to.

I edited idf_tools.py as suggested in the first comment in this thread and it works fine.

joshua-beck-0908, 2023-06-02:
> > > Hi @kylemallory,
> > > ```
> > > ERROR: This script was called from a virtual environment, can not create a virtual environment again
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > The error means that you've opened the terminal and run `. export.sh` which activated the virtual environment. And now you have run `install.sh` and it cannot install a new virtual environment if one is already activated.
> > > The real solution to this problem is to open a new terminal window (clean shell) and run the `install.sh` there.
> > > Please note that if you've added the `. export.sh` into one of the scripts processed by every shell (for example `.bashrc`) then you can get the same error. The solution for it is the same.
> > 
> > 
> > I don't believe this is entirely correct. I don't speak Python, but the script appears to find ANY virtual environment that's in the path, regardless if it's esp-idf or not. I ran into this exact problem, had a completely clean environment as far as esp-idf goes, and it blew it's brains out as above due to finding a virtual environment for a statistics package that someone at work wrote that was in my path.
> 
> Faced the same issue as @jcwren, on a mac m1 . Had to remove another unrelated virtual environment from my `zsh` profile before I got `install.sh` to work. Steps are below:
> 
>     1. `nano ~/.zshrc`
> 
>     2. removed exported virtual environment paths from $PATH
> 
>     3. start fresh terminal , re-run `install.sh`

Yeah, I had the same issue here from an unrelated PlatformIO virtual environment in the path. Removed it and it worked.
Thanks, this helps a lot.