espressif/esp-idf #8473: ESP32-S3 RMT unexpected behavior [bug]

remsdev, 2022-02-26:
## Environment

- Development Kit:      [ESP32-S3-DevKitC-1]
- Kit version : [v1]
- Module or chip used:  [ESP32-S3-WROOM-1]
- IDF version : V4.4
- Build System:   [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     [Windows]
- environment type: [ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply: [USB]

## Problem Description

Trying to use the RMT peripheral to generate clock bursts. The RMT peripheral of the ESP32-S3 sends wrong signals. The signals period is much too long (about 400us, 12.5ns expected) and the global shape is not as expected. Using the same code on ESP32 works as expected. Could someone confirm it is a bug and help to fix it? Thank you!

Scope analog traces:

### Expected Behavior
Test 1, working on ESP 32:

[image]

Test 2, working on ESP 32:

[image]

### Actual Behavior
Test 1, failing on ESP32-S3 (expecting 12.5ns instead of 410us):

[image]

Test 2, failing on ESP32-S3 (expecting 12.5ns instead of 407us and 8 pulses data bursts as for ESP32)

[image]

### Steps to reproduce

1. menuconfig default settings
2. Use provided code

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

//*****

#include "driver/rmt.h"

#define SCK GPIO_NUM_16

rmt_config_t SCK_config;
rmt_item32_t SCK_items[24];

void app_main(void){
	//*************RMT setup**********//
	//*************SCK setup**********//
	SCK_config.rmt_mode = RMT_MODE_TX;
	SCK_config.channel = RMT_CHANNEL_0;
	SCK_config.gpio_num = SCK;
	SCK_config.mem_block_num = 1;
	SCK_config.tx_config.loop_en = 1;
	SCK_config.tx_config.carrier_en = 0;
	SCK_config.tx_config.idle_output_en = 1;
	SCK_config.tx_config.idle_level = RMT_IDLE_LEVEL_LOW;
	SCK_config.tx_config.carrier_level = RMT_CARRIER_LEVEL_HIGH;
	SCK_config.clk_div = 1; // 80MHz

	for(int i=0; i<24; i++){
		SCK_items[i].duration0=1;
		SCK_items[i].level0=1;
		SCK_items[i].duration1=1;
		SCK_items[i].level1=0;
	}
	
	//*************Test 2 configuration, uncomment****************
	
	// SCK_items[6].duration0=1;
	// SCK_items[6].level0=1;
	// SCK_items[6].duration1=40;
	// SCK_items[6].level1=0;
	// SCK_items[14].duration0=1;
	// SCK_items[14].level0=1;
	// SCK_items[14].duration1=40;
	// SCK_items[14].level1=0;
	// SCK_items[22].duration0=1;
	// SCK_items[22].level0=1;
	// SCK_items[22].duration1=90;
	// SCK_items[22].level1=0;
	
	//************Test 2 configutation end****************

	
	rmt_driver_install(SCK_config.channel, 0, 0); 
  
	rmt_config(&SCK_config);

	rmt_write_items(SCK_config.channel, SCK_items, 24, 0);
}

## Debug Logs from idf.py monitor:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1648
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f58
entry 0x403b6248
I (24) boot: ESP-IDF v4.4 2nd stage bootloader
I (25) boot: compile time 20:40:30
I (25) boot: chip revision: 0
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=084a8h ( 33960) map
I (98) esp_image: segment 1: paddr=000184d0 vaddr=3fc918d0 size=02694h (  9876) load
I (102) esp_image: segment 2: paddr=0001ab6c vaddr=40374000 size=054ach ( 21676) load
I (113) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1853ch ( 99644) map
I (135) esp_image: segment 4: paddr=00038564 vaddr=403794ac size=0841ch ( 33820) load
I (143) esp_image: segment 5: paddr=00040988 vaddr=50000000 size=00010h (    16) load
I (148) boot: Loaded app from partition at offset 0x10000
I (148) boot: Disabling RNG early entropy source...
I (163) cpu_start: Pro cpu up.
I (164) cpu_start: Starting app cpu, entry point is 0x40375194
0x40375194: call_start_cpu1 ****/Espressif/frameworks/esp-idf-v4.4/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (178) cpu_start: Pro cpu start user code
I (178) cpu_start: cpu freq: 160000000
I (178) cpu_start: Application information:
I (180) cpu_start: Project name:     hello_world
I (186) cpu_start: App version:      1
I (190) cpu_start: Compile time:     Feb 26 2022 20:40:20
I (196) cpu_start: ELF file SHA256:  ae8190444409106f...
I (202) cpu_start: ESP-IDF:          v4.4
I (207) heap_init: Initializing. RAM available for dynamic allocation:
I (214) heap_init: At 3FC94A30 len 0004B5D0 (301 KiB): D/IRAM
I (220) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (227) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (233) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (240) spi_flash: detected chip: generic
I (244) spi_flash: flash io: dio
W (248) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (262) sleep: Configure to isolate all GPIO pins in sleep state
I (268) sleep: Enable automatic switching of GPIO sleep configuration
I (276) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

weixiongmei, 2022-02-27:
probably is because of the power management is enabled, because i had the same issue before...

remsdev, 2022-02-27:
Thank you for your answer. I checked in menuconfig -> Component config -> Power management:
-Support for power management is unchecked
-Power down CPU in light speed is checked (I tried unchecked as well, still not working...)

It really seems to be a bug in the driver, if I change: SCK_config.clk_div = 1; to SCK_config.clk_div = 2; I get the correct new theoretical frequency (25ns for each sample) but 48 samples instead of 24...

remsdev, 2022-02-27:
I confirm for bugs in the driver for ESP32-S3, I tried to output a second RMT signal on GPIO17 with RMT_CHANNEL_1 and I have both signals alternatively outputed on GPIO16 instead of each signal on separate pins (same code working perfectly on ESP32).

Without a fix this very useful driver seems unusable...

suda-morris, 2022-02-27:
Hi @remsdev  Thanks for your detailed report. The clock part of the RMT module on ESP32S3 has been redesigned. The TRM provides a formula:
[image]
In your case, this can't be satisfied.
I simply altered your code by increase the `clk_div`,
```c
void app_main(void)
{
    rmt_config_t SCK_config = {
        .rmt_mode = RMT_MODE_TX,
        .channel = 0,
        .gpio_num = 16,
        .mem_block_num = 1,
        .tx_config.loop_en = 1,
        .tx_config.idle_output_en = 1,
        .tx_config.idle_level = 0,
        .clk_div = 10, // resolution = 80MHz/10=8MHz, 1 tick = 125ns
    };

    rmt_item32_t SCK_items = {
        .duration0 = 1,
        .level0 = 1,
        .duration1 = 0,
        .level1 = 0,
    };

    rmt_driver_install(0, 0, 0);
    rmt_config(&SCK_config);
    rmt_write_items(SCK_config.channel, &SCK_items, 1, 0);
}
```
so will generate a square wave with a correct frequency:
[image]

remsdev, 2022-02-27:
Hi @suda-morris, thank you for your reply. It is very clear now, if I respect the new frequency limitation it works fine. 😃 For the problem of the 2 signals on the same pin it seems that it was because of a bad variable name... 
Everything is working for me now, thank you for your help!

closed 2022-02-27 by remsdev