espressif/esp-idf #8474: esp32c3 turn on the apsta mode, if the sta is disconnected, the ap will sometimes disappear

shun-dgut, 2022-02-27:
## Environment

- Chip:    esp32c3
- IDF version (run ``git describe --tags`` to find it):
    v4.3.1-448-g48997e82c0

## Problem Description

esp32c3 turn on the apsta mode, if the sta is disconnected, the ap will sometimes disappear

## LOG

```cpp
wifi_config_t wifi_config = {
        .sta = {
            .ssid = "wcs",
            .password = "12345678",
	    .threshold.authmode = WIFI_AUTH_WPA2_PSK,
            .pmf_cfg = {
                .capable = true,
                .required = false
            },
        },

        
    };
    wifi_config_t wifi_config1 ={
        .ap = {
                .ssid = "123",
                .password = "123",
                .max_connection = 1,
        },
    };
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA) );
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_config1) );
    ESP_ERROR_CHECK(esp_wifi_start() );

```

nishanth-radja, 2022-03-03:
@shun-dgut can we have the logs for the same when the issue happens, Pls enable wpa_supplicant logs,embed_TLS logs and Log level to debug using the menuconfig.
On what channel is the in-STA has connected to ex-AP. Can you check if the AP's channel has changed to CH-1 after disconnection.
Is there a particular scenario in which this issue is seen? Any traffic is being run ?
Do you see any crash on the DUT?

Alvin1Zhang, 2022-04-14:
@shun-dgut Thanks for reporting, would you please help share if any updates for the issue? Thanks.

jack0c, 2022-04-24:
The code here is not correct:
 ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
 ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_config1) );
when run second, station config missed.

   wifi_config_t wifi_config1 ={
        .ap = {
                .ssid = "123",
                .password = "123",
                .max_connection = 1,
        },
    };
Also only config very limited members and set password to "123" will also not work well.

@shun-dgut Please provide the log here.

nishanth-radja, 2022-05-23:
Issue can be closed as there seems to be a correction in the code used by user and user has not replied after trying out the correction suggested.

closed 2022-05-23 by Alvin1Zhang