espressif/esp-idf #8477: master branch build issues

chegewara, 2022-02-27:
I dont know what is broken in current master, but i have problems to build project which i could build week or two ago.

Here you can see few errors:
1. 
```
esp32-owb/owb_gpio.c:85:5: error: implicit declaration of function 'ets_delay_us'; did you mean '_us_delay'? [-Werror=implicit-function-declaration]
     ets_delay_us(time_us);
```
2. whole bunch of those, and yes, i do have `#include "freertos/FreeRTOS.h"`:
```
../main/bme688.c:228:63: error: 'portTICK_RATE_MS' undeclared (first use in this function); did you mean 'portTICK_PERIOD_MS'?
     esp_err_t err = i2c_master_cmd_begin(I2C_BUS, cmd, 1000 / portTICK_RATE_MS);
                                                               ^~~~~~~~~~~~~~~~
                                                               portTICK_PERIOD_MS
```
3. also whole bunch of those:
```
/home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/include/freertos/semphr.h:319:92: note: in definition of macro 'xSemaphoreTake'
 #define xSemaphoreTake( xSemaphore, xBlockTime )    xQueueSemaphoreTake( ( xSemaphore ), ( xBlockTime ) )
```

board: esp32
esp-idf: master branch

chegewara, 2022-02-28:
To confirm error 1 i made simple component in hello_world example, with one function:
```
void test_delay()
{
    ets_delay_us(1000);
}
```
and here is error:
```
../components/test/test.c: In function 'test_delay':
../components/test/test.c:6:5: error: implicit declaration of function 'ets_delay_us'; did you mean 'test_delay'? [-Werror=implicit-function-declaration]
     ets_delay_us(1000);
     ^~~~~~~~~~~~
     test_delay
cc1: some warnings being treated as errors
```

suda-morris, 2022-02-28:
I think this is an intentional breaking change when recently we cleaned up the FreeRTOS component dependency.  
Now the `configENABLE_BACKWARD_COMPATIBILITY` is default to `0`, so `#define portTICK_RATE_MS              portTICK_PERIOD_MS` won't take effect. 

`ets_sys.h` won't be exposed by `freertos` component anymore, we recommend using ROM APIs listed in the `esp_rom_xxx.h`, e.g. ets_delay_us -> esp_rom_delay_us

chegewara, 2022-02-28:
It is very inconvenient, especially to maintain components compatible with v4.x and v5.x, but the answer is acceptable.

Thanks

closed 2022-02-28 by chegewara

AxelLin, 2022-03-03:
@suda-morris 
I also hit compile error on master like below.

fatal error: driver/gpio.h: No such file or directory
 #include "driver/gpio.h"
          ^~~~~~~~~~~~~~~
Any document for such breaking change?

igrr, 2022-03-03:
@AxelLin I think it's this one: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/build-system.html#dependency-on-driver-component-shall-be-explicit