espressif/esp-idf #8478: DPP Enrollee Example cannot configure Bootstrapping key

zachdeibert, 2022-02-27:
## Environment

- Development Kit: `none`
- Module or chip used:  `ESP32-WROOM-32D`
- IDF version: `v5.0-dev-1730-g229ed08484`
- Build System: `idf.py`
- Compiler version: `xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0`
- Operating System: `Linux`
- Using an IDE?: `No`
- Power Supply: `USB`

## Problem Description

The [DPP Enrollee Example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/wifi_easy_connect/dpp-enrollee) has a config setting `ESP_DPP_BOOTSTRAPPING_KEY` which is described as `Private key string for DPP Bootstrapping in PEM format.`.  One such key would be similar to the following:

```
-----BEGIN EC PRIVATE KEY-----
MHcCAQEEIK7RQHOGfGjDlCIuqxrIWIpNN+h24aPD1Qr+P8gLDbwxoAoGCCqGSM49
AwEHoUQDQgAEYnNGCHd4ThO5azrtrek7gOrydPe7XYYvhoVOlA+O5GGxmezPLQqn
U/jqX6zMRJdm3q7TSt602dnUh3tJnar3dg==
-----END EC PRIVATE KEY-----
```

By inspection of the [IDF code](https://github.com/espressif/esp-idf/blob/229ed084847455da5aeda1dfd17ad3b6ccadc9b0/components/wpa_supplicant/src/common/dpp.c#L6111), it seems that it actually wants this encoded in hex, similar to the following:

```
2D2D2D2D2D424547494E2045432050524956415445204B45592D2D2D2D2D0A4D48634341514545494B375251484F4766476A446C434975717872495749704E4E2B6832346150443151722B5038674C446277786F416F4743437147534D34390A417745486F55514451674145596E4E474348643454684F35617A727472656B37674F72796450653758595976686F564F6C412B4F354747786D657A504C51716E0A552F6A7158367A4D524A646D337137545374363032646E556833744A6E61723364673D3D0A2D2D2D2D2D454E442045432050524956415445204B45592D2D2D2D2D0A
```

Additionally, going further down into the [code](https://github.com/espressif/mbedtls/blob/8b0e35f2ad477fcc2a267cf434528024b8499085/include/mbedtls/pk.h#L666-L669), it seems it also wants it to be NULL-terminated:

```
2D2D2D2D2D424547494E2045432050524956415445204B45592D2D2D2D2D0A4D48634341514545494B375251484F4766476A446C434975717872495749704E4E2B6832346150443151722B5038674C446277786F416F4743437147534D34390A417745486F55514451674145596E4E474348643454684F35617A727472656B37674F72796450653758595976686F564F6C412B4F354747786D657A504C51716E0A552F6A7158367A4D524A646D337137545374363032646E556833744A6E61723364673D3D0A2D2D2D2D2D454E442045432050524956415445204B45592D2D2D2D2D0A00
```

Placing this into the config for the project causes it to fail at runtime, however.

### Expected Behavior

The example code should run and generate a QR code similar to the example's [README](https://github.com/espressif/esp-idf/blob/master/examples/wifi/wifi_easy_connect/dpp-enrollee/README.md).

### Actual Behavior

The example hangs before printing the DPP QR code.

### Steps to reproduce

1. Set the `ESP_DPP_BOOTSTRAPPING_KEY` to the above NULL-terminated, hex-encoded key (`idf.py menuconfig`)
2. Build the project
3. Flash it onto a board
4. Observe monitor log

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6780
ho 0 tail 12 room 4
load:0x40078000,len:14796
load:0x40080400,len:3792
entry 0x40080694
I (28) boot: ESP-IDF v5.0-dev-1730-g229ed08484 2nd stage bootloader
I (28) boot: compile time 15:47:37
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=16810h ( 92176) map
I (146) esp_image: segment 1: paddr=00026838 vaddr=3ffb0000 size=03930h ( 14640) load
I (152) esp_image: segment 2: paddr=0002a170 vaddr=40080000 size=05ea8h ( 24232) load
I (162) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=82404h (533508) map
I (355) esp_image: segment 4: paddr=000b242c vaddr=40085ea8 size=0e438h ( 58424) load
I (379) esp_image: segment 5: paddr=000c086c vaddr=50000000 size=00010h (    16) load
I (390) boot: Loaded app from partition at offset 0x10000
I (390) boot: Disabling RNG early entropy source...
I (402) cpu_start: Pro cpu up.
I (402) cpu_start: Starting app cpu, entry point is 0x40081194
0x40081194: call_start_cpu1 at /mnt/d/github.com/zachdeibert/esp-idf/components/esp_system/port/cpu_start.c:152

I (0) cpu_start: App cpu up.
I (416) cpu_start: Pro cpu start user code
I (416) cpu_start: cpu freq: 160000000 Hz
I (416) cpu_start: Application information:
I (421) cpu_start: Project name:     dpp-enrollee
I (426) cpu_start: App version:      v5.0-dev-1730-g229ed08484
I (433) cpu_start: Compile time:     Feb 27 2022 15:51:33
I (439) cpu_start: ELF file SHA256:  7c023e7b8d1f124d...
I (445) cpu_start: ESP-IDF:          v5.0-dev-1730-g229ed08484
I (452) heap_init: Initializing. RAM available for dynamic allocation:
I (459) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (465) heap_init: At 3FFB7870 len 00028790 (161 KiB): DRAM
I (471) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (477) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (484) heap_init: At 400942E0 len 0000BD20 (47 KiB): IRAM
I (491) spi_flash: detected chip: generic
I (495) spi_flash: flash io: dio
W (498) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (513) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (572) wifi:wifi driver task: 3ffc0024, prio:23, stack:6656, core=0
I (572) system_api: Base MAC address is not set
I (572) system_api: read default base MAC address from EFUSE
I (582) wifi:wifi firmware version: c1cbbc6
I (582) wifi:wifi certification version: v7.0
I (582) wifi:config NVS flash: enabled
I (582) wifi:config nano formating: disabled
I (592) wifi:Init data frame dynamic rx buffer num: 32
I (592) wifi:Init management frame dynamic rx buffer num: 32
I (602) wifi:Init management short buffer num: 32
I (602) wifi:Init dynamic tx buffer num: 32
I (612) wifi:Init static rx buffer size: 1600
I (612) wifi:Init static rx buffer num: 10
I (612) wifi:Init dynamic rx buffer num: 32
I (622) wifi_init: rx ba win: 6
I (622) wifi_init: tcpip mbox: 32
I (632) wifi_init: udp mbox: 6
I (632) wifi_init: tcp mbox: 6
I (632) wifi_init: tcp tx win: 5744
I (642) wifi_init: tcp rx win: 5744
I (642) wifi_init: tcp mss: 1440
I (652) wifi_init: WiFi IRAM OP enabled
I (652) wifi_init: WiFi RX IRAM OP enabled
I (662) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (762) wifi:mode : sta (ac:67:b2:39:4e:28)
I (762) wifi:enable tsf
I (762) wifi dpp-enrollee: Started listening for DPP Authentication
```

## Other items if possible

[sdkconfig](https://github.com/espressif/esp-idf/files/8149489/sdkconfig.txt)
[dpp-enrollee.elf.gz](https://github.com/espressif/esp-idf/files/8149494/dpp-enrollee.elf.gz)

Punforgiven, 2022-03-02:
I've been able to fix this issue in my own testing. The commit https://github.com/espressif/esp-idf/pull/8479/commits/1997757e3c7b4e62c2ce828a814460ed0ca78692 submitted by @zachdeibert  only seems to fix part of the issue. 
I had to change the "dpp_bootstrap_gen" function of the "dpp.c" module to the following to get it to work with a string representation of a EC key (there is likely some redundancy here, and i also added a modified version of the "get_param" function. This can likely be incorporated in a better way):

```
char * get_param_no_ws(const char *cmd, const char *param)
{
	const char *pos;
	char *val;
	size_t len;

	pos = os_strstr(cmd, param);
	if (!pos)
		return NULL;

	pos += os_strlen(param);
	
	len = os_strlen(pos);
	
	val = os_malloc(len + 1);
	if (!val)
		return NULL;
	os_memcpy(val, pos, len);
	val[len] = '\0';
	return val;
}

int dpp_bootstrap_gen(struct dpp_global *dpp, const char *cmd)
{
	char *chan = NULL, *mac = NULL, *info = NULL, *pk = NULL, *curve = NULL;
	char *key = NULL;
	u8 *privkey = NULL;
	size_t privkey_len = 0;
	size_t len;
	int ret = -1;
	struct dpp_bootstrap_info *bi;

	bi = os_zalloc(sizeof(*bi));
	if (!bi)
		goto fail;

	if (os_strstr(cmd, "type=qrcode"))
		bi->type = DPP_BOOTSTRAP_QR_CODE;
	else if (os_strstr(cmd, "type=pkex"))
		bi->type = DPP_BOOTSTRAP_PKEX;
	else if (os_strstr(cmd, "type=nfc-uri"))
		bi->type = DPP_BOOTSTRAP_NFC_URI;
	else
		goto fail;

	chan = get_param(cmd, " chan=");
	mac = get_param(cmd, " mac=");
	info = get_param(cmd, " info=");
	curve = get_param(cmd, " curve=");

// Start of changes
	key = get_param_no_ws(cmd, " key=");

	if (key) {
		privkey_len = os_strlen(key) + 1;
		privkey = os_malloc(privkey_len);
		if (!privkey)
			goto fail;
		os_memcpy(privkey, key, privkey_len);
	}
// End of changes
	wpa_hexdump(MSG_DEBUG, "private key", privkey, privkey_len);

	pk = dpp_keygen(bi, curve, privkey, privkey_len);
	if (!pk)
		goto fail;
       ...
```

nachiketkukade, 2022-03-02:
@zachdeibert , there has been a mistake in the documentation. The Bootstrapping key actually needs to be in RAW hex format, which would be a 64 hex digits (32 bytes) of private key. Internally we're adding prefix and postfix to the key which is fixed and then generate the public key. Can you pls try with the raw private key? If you're using openssl, `openssl pkey -in [key file] -text` will display the raw key separately. Thanks.

nachiketkukade, 2022-03-03:
@Punforgiven , I've mentioned the quick way of making this work in my previous comment. If the present input method is not intuitive even after updating the documentation, we'll modify it. Thanks for your input.

nachiketkukade, 2022-03-03:
@zachdeibert , thanks. Yes, I'll update the doc. I'll respond on the PR separately.

closed 2022-03-03 by nachiketkukade