espressif/esp-idf #8481: esp32c3: BLE provisioning : STA disconnection on wifi_prov_mgr_deinit()

HiFiPhile, 2022-02-28:
## Environment

- Development Kit:      [ESP32-C3-DevKitM-1]
- Kit version : [v1]
- IDF version (run ``git describe --tags`` to find it):
```
 v5.0-dev-1554-g20847eeb96
 f31694c9f1746ba189a4bcae2e34db15135ddb22 components/asio/asio (asio-1-12-0-201-gf31694c9)
 d037ec89546fad14b5c4d5456c2e23a71e554966 components/bootloader/subproject/components/micro-ecc/micro-ecc (v1.0)
 c0a5619d46e56cbf4602327c925fd14f04d125eb components/bt/controller/lib_esp32 (c0a5619)
 a3f7d7ffa2b261b1415042d8b7cd457cc2b4b1de components/bt/controller/lib_esp32c3_family (heads/master)
 554cd264cd028886411b183885bb21b47b8901a6 components/bt/host/nimble/nimble (heads/nimble-1.3.0-idf)
 eeecc49ce8af123cf8ad40efdb9673e37b56230f components/cmock/CMock (v2.5.2-2-geeecc49)
 3aa11612c143c9734d72022720f33e12506f7a2c components/coap/libcoap (v4.2.1-540-g3aa1161)
 fced1a554f1b870f5a4e336431ab9915c4d90326 components/esp_phy/lib (heads/master)
 66cc3ae82adce95f38d1f6738b00fca5bf09fda1 components/esp_wifi/lib (remotes/origin/feature/ftm_support_stage2-29-g66cc3ae)
 845252b0b2b768c7bea96f11af5439fbd12c73f5 components/esptool_py/esptool (v3.2-37-g845252b)
 5545eb14d1117a7865a05a6b4a0f19c62760f8c7 components/ieee802154/lib (heads/master)
 d348621ca93571343a56862df7de4ff3bc9b5667 components/json/cJSON (v1.7.15)
 2749568fe15df2003f6c3f37f0dfd44f8f01fcd6 components/lwip/lwip (remotes/origin/master-224-g2749568f)
 8b0e35f2ad477fcc2a267cf434528024b8499085 components/mbedtls/mbedtls (mbedtls-2.13.1-8771-g8b0e35f2a)
 985078affa8a2d2b56b87c8e6455252850f895c6 components/mqtt/esp-mqtt (ESP-MQTT_FOR_IDF_3.1-293-g985078a)
 901353a8c17d5c3040b3850d571550b0472b921b components/openthread/lib (heads/master)
 e736d2488d9d4f787650d81f2b84f59a194f2b41 components/openthread/openthread (thread-reference-20200818-1424-ge736d2488)
 dac1a65feac4ad72f612aab99f487056fbcf5c1a components/protobuf-c/protobuf-c (v1.3.0)
 f5e26c4e933189593a71c6b82cda381a7b21e41c components/spiffs/spiffs (0.2-221-gf5e26c4)
 c1986586098643edfa6b75a776299c576ad0ce8b components/tinyusb/tinyusb (remotes/origin/esp-based_on_9e992698)
 7d2bf62b7e6afaf38153041a9d53c21aeeca9a25 components/unity/unity (v2.4.3-51-g7d2bf62)
 7e8e249990ec491ec15990cf95b6d871a66cf64a examples/build_system/cmake/import_lib/main/lib/tinyxml2 (2.0.2-659-g7e8e249)
 077758db50d4b55addab56e723a5a379f09077d0 examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib (077758d)
```
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
(crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

When BLE provisioning is successfully finished, STA become disconnected once wifi_prov_mgr_deinit() is called. If I use wifi_prov_mgr_disable_auto_stop() then STA connection is kept.

```
I (77961) app: Connected with IP Address:192.168.1.195
I (77961) esp_netif_handlers: sta ip: 192.168.1.195, mask: 255.255.255.0, gw: 192.168.1.1
I (77961) wifi_prov_mgr: STA Got IP
I (77961) app: Provisioning successful
I (77971) app: Hello World!
I (78971) app: Hello World!
I (79971) app: Hello World!
I (80971) app: Hello World!
I (81971) app: Hello World!
I (82891) NimBLE: GAP procedure initiated: stop advertising.

I (82901) NimBLE: GAP procedure initiated: stop advertising.

I (82921) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

I (82971) app: Hello World!
E (83041) protocomm_nimble: Error setting advertisement data; rc = 30
W (83051) Timer: Timer not stopped
W (83061) Timer: Timer not stopped
I (83131) wifi_prov_mgr: Provisioning stopped
W (83131) BTDM_INIT: esp_bt_mem_release not implemented, return OK
I (83131) wifi_prov_scheme_ble: BTDM memory released
I (83971) app: Hello World!
I (84971) app: Hello World!
I (85971) app: Hello World!
I (86971) app: Hello World!
[... 217 lines cut ...]
I (92931) wifi:state: auth -> assoc (0)
I (92951) wifi:state: assoc -> run (10)
I (92961) wifi:connected with Mengsk, aid = 3, channel 1, BW20, bssid = 00:25:9c:13:9e:88
I (92961) wifi:security: WPA2-PSK, phy: bgn, rssi: -28
I (92971) wifi:pm start, type: 1

I (92971) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
I (92981) app: Hello World!
I (92991) wifi:BcnInt:102400, DTIM:2
I (93981) app: Hello World!
I (94461) app: Connected with IP Address:192.168.1.195
I (94461) esp_netif_handlers: sta ip: 192.168.1.195, mask: 255.255.255.0, gw: 192.168.1.1
I (94981) app: Hello World!
I (95981) app: Hello World!
I (96981) app: Hello World!
I (97981) app: Hello World!
I (98981) app: Hello World!
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)

[wifi_prov_mgr.zip](https://github.com/espressif/esp-idf/files/8156057/wifi_prov_mgr.zip)

HiFiPhile, 2022-02-28:
[full log](https://github.com/espressif/esp-idf/files/8157446/full.txt)

This beacon timeout WIFI_EVENT:21 is suspicious:
```
I (103238) app: Hello World!
I (104238) app: Hello World!
I (105238) app: Hello World!
I (106238) app: Hello World!
I (106658) wifi:bcn_timout,ap_probe_send_start
D (106658) event: running post WIFI_EVENT:21 with handler 0x42008042 and context 0x3fca0a80 on loop 0x3fc9fad8
0x42008042: event_handler at E:\mcu\esp32c3\wifi_prov_mgr\build/../main/app_main.c:48

I (107238) app: Hello World!
I (108238) app: Hello World!
I (109158) wifi:ap_probe_send over, resett wifi status to disassoc
I (109158) wifi:state: run -> init (c800)
D (109158) wifi:connect status 5 -> 3
D (109158) wifi:send diassoc for beacon timeout(200)
D (109158) wifi:sta leave
I (109168) wifi:pm stop, total sleep time: 6875891 us / 13992877 us

D (109168) wifi:stop CSA timer
D (109178) wifi:remove 00:25:9c:13:9e:88 from rc list
I (109178) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
D (109188) wifi:filter: set rx policy=8
D (109188) wifi:Send disconnect event, reason=200, AP number=0
D (109198) wifi:connect status 3 -> 0
D (109198) event: running post WIFI_EVENT:5 with handler 0x42008042 and context 0x3fca0a80 on loop 0x3fc9fad8
0x42008042: event_handler at E:\mcu\esp32c3\wifi_prov_mgr\build/../main/app_main.c:48

I (109208) app: Disconnected. Connecting to the AP again...
```
```

Weijian-Espressif, 2022-03-30:
@wmy-espressif  please take a look

BetterJincheng, 2022-04-18:
@HiFiPhile We have located the issue somewhere in libphy.a. The attachment is the fixed one, please unzip and replace `components/esp_phy/lib/esp32c3/libbtbb.a` with the attachment then test again on branch `release/v4.4` . We wait for your test result.
Thanks
[libphy.a.zip](https://github.com/espressif/esp-idf/files/8503214/libphy.a.zip)

HiFiPhile, 2022-04-18:
@BetterJincheng I've tested with the same revision of issue `v5.0-dev-1554-g20847eeb96`. Now there is no more disconnection after the provisioning.

HiFiPhile, 2022-04-25:
I saw there is a new error in log though provisioning seems working, should I care about it ?

```
D:\mcu\esp-idf>git describe
v5.0-dev-2563-g45c1d1cba2

E (17820) NimBLE: exp_time:0.now:1740.ticks:30000.active:1.Need reset.
```
Full log:
```
I (1610) phy_init: phy_version 909,aa05aec,Apr 16 2022,13:42:08
I (1700) wifi:mode : sta (84:f7:03:42:97:44)
I (1700) wifi:enable tsf
W (1710) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (1710) BTDM_INIT: BT controller compile version [d913766]
I (1710) BTDM_INIT: Bluetooth MAC: 84:f7:03:42:97:46

I (1720) protocomm_nimble: BLE Host Task Started
I (1730) wifi_prov_mgr: Provisioning started with service name : Linkify_000001 
I (1730) main: Provisioning started
I (1740) NimBLE: GAP procedure initiated: stop advertising.

I (1740) main: Provisioning params: name - Linkify_000001 pin - abcd1234
I (1750) NimBLE: GAP procedure initiated: advertise; 
I (1760) NimBLE: disc_mode=2
I (1760) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (1770) NimBLE: 

E (17820) NimBLE: exp_time:0.now:1740.ticks:30000.active:1.Need reset.

I (65760) main: Received Wi-Fi credentials
        SSID     : xxxxxxxx
        Password : xxxxxxxx
I (69680) wifi:new:<13,0>, old:<1,0>, ap:<255,255>, sta:<13,0>, prof:1
I (70870) wifi:state: init -> auth (b0)
I (70870) wifi:state: auth -> assoc (0)
I (70880) wifi:state: assoc -> run (10)
I (71080) wifi:connected with xxxxxxxx, aid = 1, channel 13, BW20, bssid = a0:04:60:aa:30:50
I (71080) wifi:security: WPA2-PSK, phy: bgn, rssi: -44
I (71090) wifi:pm start, type: 2
```

rahult-github, 2022-04-25:
@HiFiPhile  , if the query is regarding below print, then this is fixed in latest commits. This is wrongly coming up as an Error print and hence can be ignored. 

E (17820) NimBLE: exp_time:0.now:1740.ticks:30000.active:1.Need reset.

HiFiPhile, 2022-04-25:
> @HiFiPhile , if the query is regarding below print, then this is fixed in latest commits. This is wrongly coming up as an Error print and hence can be ignored.

I'm using the latest master branch `v5.0-dev-2563-g45c1d1cba2` as above. You mean the bug fix commit isn't published yet ?

rahult-github, 2022-04-26:
Hi @HiFiPhile ,

`You mean the bug fix commit isn't published yet ?`  

Yes, it will get synced within few days.

Alvin1Zhang, 2022-06-06:
Thanks for reporting, sorry for the very slow turnaround, fix is available at https://github.com/espressif/esp-idf/commit/56914d3c811ed6ed4adbc0ee2e6e82c3a95d87c4.

parksj10, 2023-05-08:
@HiFiPhile, random question, what are you guys using for monitoring USB power consumption?

HiFiPhile, 2023-05-08:
@parksj10 
You can try a Nordic PPK2.

parksj10, 2023-05-08:
> @parksj10 You can try a Nordic PPK2.

Thanks so much!