espressif/esp-idf #8485: About DMA read in ESP32-S3.

zh11311, 2022-03-01:
I adapt the example of DMA read. In esp32-s3, six channels of ADC1 and one channel of ADC2 are used, and the values of six channels of ADC1 can be read normally.
    However, if I use ADC2 of 8 channels and 1 channel of ADC1, I can only read the value of the first channel of ADC1, and this value is always 0.
    The information of the monitor:
    "Unit: 1,_Channel: 0, Value: 0
     Unit: 1,_Channel: 0, Value: 0
     Unit: 1,_Channel: 0, Value: 0
     Unit: 1,_Channel: 0, Value: 0
     Unit: 1,_Channel: 0, Value: 0
     Unit: 1,_Channel: 0, Value: 0
     Unit: 1,_Channel: 0, Value: 0
     ......

    why?

ginkgm, 2022-03-02:
Hi @zh11311 ,
Could you provide more information, especially the IDF version and your code, for us to better check this issue?

zh11311, 2022-03-02:
Hi，idf version is 4.4 or 5.0，The chip is ESP32-S3.

The source is following below：
#include <string.h>
#include <stdio.h>
#include "sdkconfig.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "driver/adc.h"

#define TIMES              256
#define GET_UNIT(x)        ((x>>3) & 0x1)

#if CONFIG_IDF_TARGET_ESP32
#define ADC_RESULT_BYTE     2
#define ADC_CONV_LIMIT_EN   1                       //For ESP32, this should always be set to 1
#define ADC_CONV_MODE       ADC_CONV_SINGLE_UNIT_1  //ESP32 only supports ADC1 DMA mode
#define ADC_OUTPUT_TYPE     ADC_DIGI_OUTPUT_FORMAT_TYPE1
#elif CONFIG_IDF_TARGET_ESP32S2
#define ADC_RESULT_BYTE     2
#define ADC_CONV_LIMIT_EN   0
#define ADC_CONV_MODE       ADC_CONV_BOTH_UNIT
#define ADC_OUTPUT_TYPE     ADC_DIGI_OUTPUT_FORMAT_TYPE2
#elif CONFIG_IDF_TARGET_ESP32C3 || CONFIG_IDF_TARGET_ESP32H2
#define ADC_RESULT_BYTE     4
#define ADC_CONV_LIMIT_EN   0
#define ADC_CONV_MODE       ADC_CONV_ALTER_UNIT     //ESP32C3 only supports alter mode
#define ADC_OUTPUT_TYPE     ADC_DIGI_OUTPUT_FORMAT_TYPE2
#elif CONFIG_IDF_TARGET_ESP32S3
#define ADC_RESULT_BYTE     4
#define ADC_CONV_LIMIT_EN   0
#define ADC_CONV_MODE       ADC_CONV_BOTH_UNIT
#define ADC_OUTPUT_TYPE     ADC_DIGI_OUTPUT_FORMAT_TYPE2
#endif

static uint16_t adc1_chan_mask = BIT(0) | BIT(1) | BIT(2) | BIT(3) | BIT(4) | BIT5(5) | BIT(6) | BIT(7);
static uint16_t adc2_chan_mask = BIT(0);
static adc_channel_t channel[9] = {ADC1_CHANNEL_0, ADC1_CHANNEL_1, ADC1_CHANNEL_2, ADC1_CHANNEL_3, ADC1_CHANNEL_4, ADC1_CHANNEL_5, ADC1_CHANNEL_6, ADC1_CHANNEL_7, (ADC2_CHANNEL_0 | 1 << 3)};

static const char *TAG = "ADC DMA";

static void continuous_adc_init(uint16_t adc1_chan_mask, uint16_t adc2_chan_mask, adc_channel_t *channel, uint8_t channel_num)
{
    adc_digi_init_config_t adc_dma_config = {
        .max_store_buf_size = 1024,
        .conv_num_each_intr = TIMES,
        .adc1_chan_mask = adc1_chan_mask,
        .adc2_chan_mask = adc2_chan_mask,
    };
    ESP_ERROR_CHECK(adc_digi_initialize(&adc_dma_config));

    adc_digi_configuration_t dig_cfg = {
        .conv_limit_en = ADC_CONV_LIMIT_EN,
        .conv_limit_num = 250,
        .sample_freq_hz = 10 * 1000,
        .conv_mode = ADC_CONV_MODE,
        .format = ADC_OUTPUT_TYPE,
    };

    adc_digi_pattern_config_t adc_pattern[SOC_ADC_PATT_LEN_MAX] = {0};
    dig_cfg.pattern_num = channel_num;
    for (int i = 0; i < channel_num; i++) {
        uint8_t unit = GET_UNIT(channel[i]);
        uint8_t ch = channel[i] & 0x7;
        adc_pattern[i].atten = ADC_ATTEN_DB_0;
        adc_pattern[i].channel = ch;
        adc_pattern[i].unit = unit;
        adc_pattern[i].bit_width = SOC_ADC_DIGI_MAX_BITWIDTH;
[... 66 lines cut ...]
                else if (ADC_CONV_MODE == ADC_CONV_SINGLE_UNIT_2) {
                    ESP_LOGI(TAG, "Unit: %d, Channel: %d, Value: %x", 2, p->type1.channel, p->type1.data);
                } else if (ADC_CONV_MODE == ADC_CONV_SINGLE_UNIT_1) {
                    ESP_LOGI(TAG, "Unit: %d, Channel: %d, Value: %x", 1, p->type1.channel, p->type1.data);
                }
    #endif  //#if CONFIG_IDF_TARGET_ESP32S2
    #endif
            }
            //See `note 1`
            vTaskDelay(1);
        } else if (ret == ESP_ERR_TIMEOUT) {
            /**
             * ``ESP_ERR_TIMEOUT``: If ADC conversion is not finished until Timeout, you'll get this return error.
             * Here we set Timeout ``portMAX_DELAY``, so you'll never reach this branch.
             */
            ESP_LOGW(TAG, "No data, increase timeout or reduce conv_num_each_intr");
            vTaskDelay(1000);
        }

    }

    adc_digi_stop();
    ret = adc_digi_deinitialize();
    assert(ret == ESP_OK);
}

zh11311, 2022-03-03:
> I am asking Espressif for the same problem. Actually any small change irrelevant to ADC in the example codes, for example, adding one more ESP_LOGI statement and/or adding one or more character in other ESP_LOGI string, will just kill ADC DMA.
> 
> They have reproduced the issue and I was told they are working with this.

@eos1d3 Thank you to tell me these information.  Hope Espressif can solve this problem as soon as possible.

zh11311, 2022-03-03:
@ginkgm Could you have any suggestions?

Evaneee, 2022-03-09:
for esp32s3 i have same problem with @zh11311 

 and for esp32s2, i have same problem with @eos1d3 

hope can be solved

ginkgm, 2022-03-09:
Hi @zh11311, @eos1d3 , @Evaneee ,

We have a related fix, which is 748037708f8dc708dfea145807ebf792b20d9494 on master, and 121ddb87e5130314e4fcc5e9cb260a81b7d30d36 on release/v4.4

Could you please have a try and see what happend?

Evaneee, 2022-03-09:
@ginkgm ths
i v do a test result is 
[image]
============================================================================
============================================================================
[image]

eos1d3, 2022-03-09:
@zh11311 @ginkgm @zh11311 

For my old tests with original R4.4 and current test with the new fix, I never find problem with ESP32S2.

And it is great to tell all those problem for ESP32S3 just gone. With the fix, I have tested the followings with ESP32S3:

1.  Without changing any code, the original example works.
2.  Use `ADC_CONV_SINGLE_UNIT_1` instead of `ADC_CONV_BOTH_UNIT` and use only `ADC1_CHANNEL_2`, `ADC1_CHANNEL_3`. It can work with ADC1 only and only these two channels are printed. It never worked before.
3. Connect 3.3V and GND respectively to each of the two channels, results are correct.
4. Add one more `ADC1_CHANNEL_4`, and use 3.3V and GND respectively to all the three channels, all ADC readings are correct.

So the problem is solved. @ginkgm Please fix another serious problem for current ADC DMA implementation as it costs too much RAM and CPU with extremely slow response when real time ADC reading is required. See https://github.com/espressif/esp-idf/issues/8383

Can ADC DMA work in another real time mode so that:
1.  only last 8/16/...256 readings of a channel are saved in an array for user program? There is no need to keep old ADC readings, only the latest values.
2. OR the 8/16/...256 readings of a channel are added so that next ADC read from user program can get the sum (or average) directly. This is similar to STM32 ADC oversampling in ADC DMA mode. But it works in all hardware without using interrupt.

And thank for the good work!

spkurtzweil, 2022-04-22:
I am having the exact same issue with a project I'm working on. I'm new to github; is the fix mentioned earlier part of the vscode extension release or will I need to wait for a while to get the fix?

eos1d3, 2022-04-22:
VS Code has nothing related to the issue. The ADC driver in older IDF 5.0 and 4.4 have bugs. And I believe recent updates have patched them. The IDF 5.0 I used a while ago fixed the issue. And I expect IDF 4.4.1 should fix it too.

ESP IDF keeps changing even they are released. Just use the latest ones and try. If you do not know how to git pull, just removed the IDF and reinstall again.

felixcollins, 2022-05-12:
This looks like is perhaps a caching issue?

closed 2022-10-14 by zh11311

breeeak, 2022-10-25:
I recently came across the same problem. I uesed two new identical ESP32-S3 chips, but one of the ADC2 was not working and the other worked well. 
I have tried the official sample code `dma_read` for IDF version v4.4 and [continuous_read](https://github.com/espressif/esp-idf/tree/release/v5.0/examples/peripherals/adc/continuous_read) v5.0 but they had the same issue. 
Is this a hardware problem or  something else?
Can anyone help me?  thx

one ESP32-S3 dma adc2 not wrok:
12:59:04.702 -> [0;32mI (14627) ADC DMA: Unit: 1,_Channel: 0, Value: 0[0m
12:59:04.702 -> [0;32mI (14637) ADC DMA: Unit: 1,_Channel: 3, Value: 3e4[0m
12:59:04.702 -> [0;32mI (14637) ADC DMA: Unit: 1,_Channel: 0, Value: 0[0m
12:59:04.702 -> [0;32mI (14647) ADC DMA: Unit: 1,_Channel: 2, Value: 3c5[0m
12:59:04.702 -> [0;32mI (14647) ADC DMA: Unit: 1,_Channel: 0, Value: 0[0m
12:59:04.702 -> [0;32mI (14657) ADC DMA: Unit: 1,_Channel: 3, Value: 3e5[0m
12:59:04.702 -> [0;32mI (14657) ADC DMA: Unit: 1,_Channel: 0, Value: 0[0m
12:59:04.749 -> [0;32mI (14667) ADC DMA: Unit: 1,_Channel: 2, Value: 3ce[0m
12:59:04.749 -> [0;32mI (14677) ADC DMA: Unit: 1,_Channel: 0, Value: 0[0m
12:59:04.749 -> [0;32mI (14677) ADC DMA: Unit: 1,_Channel: 3, Value: 3f3[0m
12:59:04.749 -> [0;32mI (14687) ADC DMA: Unit: 1,_Channel: 0, Value: 0[0m
12:59:04.749 -> [0;32mI (14687) ADC DMA: Unit: 1,_Channel: 2, Value: 3d3[0m
12:59:04.749 -> [0;32mI (14697) ADC DMA: Unit: 1,_Channel: 0, Value: 0[0m

one ESP32-S3 dma adc2 wroked well:
13:15:30.229 -> [0;32mI (13274) ADC DMA: Unit: 2,_Channel: 0, Value: 2b5[0m
13:15:30.229 -> [0;32mI (13284) ADC DMA: Unit: 1,_Channel: 3, Value: fff[0m
13:15:30.229 -> [0;32mI (13284) ADC DMA: Unit: 2,_Channel: 0, Value: 217[0m
13:15:30.229 -> [0;32mI (13294) ADC DMA: Unit: 1,_Channel: 2, Value: a9d[0m
13:15:30.229 -> [0;32mI (13304) ADC DMA: Unit: 2,_Channel: 0, Value: 25d[0m
13:15:30.229 -> [0;32mI (13304) ADC DMA: Unit: 1,_Channel: 3, Value: fff[0m
13:15:30.276 -> [0;32mI (13314) ADC DMA: Unit: 2,_Channel: 0, Value: 2ba[0m
13:15:30.276 -> [0;32mI (13314) ADC DMA: Unit: 1,_Channel: 2, Value: aad[0m
13:15:30.276 -> [0;32mI (13324) ADC DMA: Unit: 2,_Channel: 0, Value: 2c8[0m
13:15:30.276 -> [0;32mI (13324) ADC DMA: Unit: 1,_Channel: 3, Value: fff[0m
13:15:30.276 -> [0;32mI (13334) ADC DMA: Unit: 2,_Channel: 0, Value: 25a[0m